lwfinger/rtw89 #88: unable to run driver on hp pavillion

diatel, 2021-12-03:
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace$ git clone https://github.com/lwfinger/rtw89.git -b v7
Cloning into 'rtw89'...
remote: Enumerating objects: 704, done.
remote: Counting objects: 100% (704/704), done.
remote: Compressing objects: 100% (450/450), done.
remote: Total 704 (delta 507), reused 435 (delta 249), pack-reused 0
Receiving objects: 100% (704/704), 1.85 MiB | 10.41 MiB/s, done.
Resolving deltas: 100% (507/507), done.
```
`maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace$ cd rtw89/`
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ make
make -C /lib/modules/5.11.0-41-generic/build M=/home/maximdiatel/workspace/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-41-generic'
  CC [M]  /home/maximdiatel/workspace/rtw89/core.o
  CC [M]  /home/maximdiatel/workspace/rtw89/debug.o
  CC [M]  /home/maximdiatel/workspace/rtw89/mac80211.o
  CC [M]  /home/maximdiatel/workspace/rtw89/mac.o
  CC [M]  /home/maximdiatel/workspace/rtw89/phy.o
  CC [M]  /home/maximdiatel/workspace/rtw89/fw.o
  CC [M]  /home/maximdiatel/workspace/rtw89/rtw8852a.o
  CC [M]  /home/maximdiatel/workspace/rtw89/rtw8852a_table.o
  CC [M]  /home/maximdiatel/workspace/rtw89/rtw8852a_rfk.o
  CC [M]  /home/maximdiatel/workspace/rtw89/rtw8852a_rfk_table.o
  CC [M]  /home/maximdiatel/workspace/rtw89/cam.o
  CC [M]  /home/maximdiatel/workspace/rtw89/efuse.o
  CC [M]  /home/maximdiatel/workspace/rtw89/regd.o
  CC [M]  /home/maximdiatel/workspace/rtw89/coex.o
  CC [M]  /home/maximdiatel/workspace/rtw89/ps.o
  CC [M]  /home/maximdiatel/workspace/rtw89/sar.o
  CC [M]  /home/maximdiatel/workspace/rtw89/ser.o
  LD [M]  /home/maximdiatel/workspace/rtw89/rtw89core.o
  CC [M]  /home/maximdiatel/workspace/rtw89/pci.o
  LD [M]  /home/maximdiatel/workspace/rtw89/rtw89pci.o
  MODPOST /home/maximdiatel/workspace/rtw89/Module.symvers
  CC [M]  /home/maximdiatel/workspace/rtw89/rtw89core.mod.o
  LD [M]  /home/maximdiatel/workspace/rtw89/rtw89core.ko
  BTF [M] /home/maximdiatel/workspace/rtw89/rtw89core.ko
Skipping BTF generation for /home/maximdiatel/workspace/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/maximdiatel/workspace/rtw89/rtw89pci.mod.o
  LD [M]  /home/maximdiatel/workspace/rtw89/rtw89pci.ko
  BTF [M] /home/maximdiatel/workspace/rtw89/rtw89pci.ko
Skipping BTF generation for /home/maximdiatel/workspace/rtw89/rtw89pci.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-41-generic'
```
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ sudo make sign-install
make -C /lib/modules/5.11.0-41-generic/build M=/home/maximdiatel/workspace/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-41-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-41-generic'
kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der rtw89core.ko
kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der rtw89pci.ko
Install rtw89 SUCCESS
```
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ hexdump -Cv rtw89pci.ko | tail -n 5
001779a0  4e 94 f2 ce c1 c6 80 cc  85 47 50 32 a2 cc d9 48  |N........GP2...H|
001779b0  d7 f4 59 86 00 00 02 00  00 00 00 00 00 00 01 bc  |..Y.............|
001779c0  7e 4d 6f 64 75 6c 65 20  73 69 67 6e 61 74 75 72  |~Module signatur|
001779d0  65 20 61 70 70 65 6e 64  65 64 7e 0a              |e appended~.|
001779dc
```
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ uname -r
5.11.0-41-generic
```
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
```
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ lspci -v | grep -i network
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
```
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ sudo modprobe rtw89pci
modprobe: ERROR: could not insert 'rtw89pci': Exec format error
```
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ dmesg
[ 1906.768803] rtw89core: disagrees about version of symbol module_layout
```

VanZwen, 2021-12-03:
I just came across the same issue on my pavillion, so i guess this is reproducable

diatel, 2021-12-03:
I would say it worked previously. i use combined connection wired ws wireless so i can't say when driver becomed unusable.
Also i would say previously i used `sudo make install` without signing driver and it worked somehow on my system. 
when i faced an issue i started to read all topics about this and similar issues and found that probably it can be related to the security boot.
so i decided to pass steps https://ubuntu.com/blog/how-to-sign-things-for-secure-boot from this manual to sign kernel objects.
spoiler it not helped and i tried both ways `make install` and `make sign-install` result actually the same.
yes i did 
```sudo make uninstall
sudo make clean``` each try to be sure i start from the clean page.

lwfinger, 2021-12-03:
When you see the "rtw89core: disagrees about version of symbol module_layout" message, it means eithor that the version of the kernel used to build the modules is different that the one running, or the gcc version is different. You can check the first by comparing the output os the 'uname -r' command with the "Entering directory" message from the build. In @diatel's instance, the latter is "5.11.0-41-generic".

diatel, 2021-12-03:
@lwfinger yes
```
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-41-generic'
```
```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ uname -r
5.11.0-41-generic
```
seems the same.
i do not know how to check gcc, as it's ubuntu's kernel. I do not know which gcc was used to compile kernel for my distro

```
maximdiatel@HP-Pavilion-15-ec2xxx:~/workspace/rtw89$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

lwfinger, 2021-12-03:
Do a 'dmesg | less'. The gcc version is in the line that starts "Linux version". It should be very early in the log. It is the second line of mine.

diatel, 2021-12-03:
```
[    0.000000] Linux version 5.11.0-41-generic (buildd@lgw01-amd64-005) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 (Ubuntu 5.11.0-41.45~20.04.1-generic 5.11.22)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.11.0-41-generic root=UUID=87752284-6de2-45a7-b01b-1c1a1248af74 ro quiet splash vt.handoff=7
```

lwfinger, 2021-12-03:
Then, I have no idea what is going wrong. Please do the following in the rtw89 directory:

sudo make uninstall
make clean
make
sudo make sign-install
sudo modprobe -v rtw89pci

diatel, 2021-12-06:
seems it was something with bad ubuntu update. (it is LTS version and it's close to impossible but i have no idea what's else)
i tried different things and finally decided to just reinstall ubuntu. 
i wrote this message using wifi driver. Still can't understand what's was wrong. looks like my laptop actually used old kernel and old libs/headers but also had new kernel and new libs installed in a wrong way so `uname -r` get new version instead actual.
again - i have no idea how it can be. It's first time i even faced such situation.

closed 2021-12-06 by diatel