lwfinger/rtw89 #89: RTL8832AU test result: [ERR]HCI FC pre init -22

s-2, 2021-12-03:
Hi,
thanks for you efforts in merging support for the USB variants of these chips!
I successfully built and installed the driver on ubuntu with 5.4.0-91 (adding a few more VID/PIDs).

When plugging the USB device (e.g. Fenvi FU-AX1800P), this happens:
```
[  615.327133] usb 1-6: new high-speed USB device number 7 using xhci_hcd
[  615.475728] usb 1-6: New USB device found, idVendor=0bda, idProduct=885c, bcdDevice= 0.00
[  615.475733] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  615.475737] usb 1-6: Product: 802.11ax WLAN Adapter
[  615.475739] usb 1-6: Manufacturer: Realtek
[  615.475742] usb 1-6: SerialNumber: 00e04c000001
[  616.042592] rtwifi_usb 1-6:1.0: 
               usb endpoint descriptor (0):
[  616.042593] rtwifi_usb 1-6:1.0: bLength=7
[  616.042594] rtwifi_usb 1-6:1.0: bDescriptorType=5
[  616.042595] rtwifi_usb 1-6:1.0: bEndpointAddress=84
[  616.042595] rtwifi_usb 1-6:1.0: wMaxPacketSize=512
[  616.042596] rtwifi_usb 1-6:1.0: bInterval=0
[  616.042597] rtwifi_usb 1-6:1.0: USB: dir in endpoint num 4

// more endpoints...

[  616.042624] rtwifi_usb 1-6:1.0: 
               usb endpoint descriptor (7):
[  616.042624] rtwifi_usb 1-6:1.0: bLength=7
[  616.042625] rtwifi_usb 1-6:1.0: bDescriptorType=5
[  616.042625] rtwifi_usb 1-6:1.0: bEndpointAddress=c
[  616.042626] rtwifi_usb 1-6:1.0: wMaxPacketSize=512
[  616.042627] rtwifi_usb 1-6:1.0: bInterval=0
[  616.042627] rtwifi_usb 1-6:1.0: USB: out endpoint num 12
[  616.042628] rtwifi_usb 1-6:1.0: USB_SPEED_HIGH
[  616.042629] rtwifi_usb 1-6:1.0: USB: bulkout_size: 512
[  616.043090] rtwifi_usb 1-6:1.0: Firmware version 0.13.30.0, cmd version 0, type 1
[  616.043093] rtwifi_usb 1-6:1.0: Firmware version 0.13.30.0, cmd version 0, type 3
[  616.048020] rtw89_usb_ops_mac_pre_init enter
[  616.051036] rtwifi_usb 1-6:1.0: [ERR]HCI FC pre init -22
[  616.051038] rtwifi_usb 1-6:1.0: failed to setup chip information
[  616.051104] BUG: unable to handle page fault for address: 0000000fffffffe8
[  616.051955] #PF: supervisor write access in kernel mode
[  616.052808] #PF: error_code(0x0002) - not-present page
[  616.053637] PGD 0 P4D 0 
[  616.054443] Oops: 0002 [#1] SMP PTI
[  616.055232] CPU: 7 PID: 4159 Comm: systemd-udevd Tainted: P           OE     5.4.0-91-generic #102-Ubuntu
[  616.056021] Hardware name: Micro-Star International Co., Ltd. GS73VR 6RF/MS-17B1, BIOS E17B1IMS.10C 11/28/2017
[  616.056809] RIP: 0010:skb_dequeue+0x50/0x70
[  616.057577] Code: 74 41 4c 39 e3 74 3c 8b 43 10 83 e8 01 89 43 10 49 8b 14 24 49 8b 44 24 08 49 c7 04 24 00 00 00 00 49 c7 44 24 08 00 00 00 00 <48> 89 42 08 48 89 10 4c 89 ef e8 81 76 21 00 4c 89 e0 5b 41 5c 41
[  616.059137] RSP: 0018:ffffb52a04a139d0 EFLAGS: 00010097
[  616.059892] RAX: ffff8dacc173b948 RBX: ffff8dacbe05d770 RCX: 000000000000d3ef
[  616.060641] RDX: 0000000fffffffe0 RSI: 0000000000000286 RDI: ffff8dacbe05d784
[  616.061375] RBP: ffffb52a04a139e8 R08: 0000000000000001 R09: 0000000000004000
[  616.062094] R10: ffff8dad2c221858 R11: ffff8dad2ebeaf38 R12: ffff8dacc173b940
[  616.062813] R13: ffff8dacbe05d784 R14: ffff8dacbe0587a0 R15: ffff8dacbe059e80
[  616.063520] FS:  00007fdd4fcb6880(0000) GS:ffff8dad2ebc0000(0000) knlGS:0000000000000000
[  616.064216] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  616.064905] CR2: 0000000fffffffe8 CR3: 0000000449320004 CR4: 00000000003606e0
[  616.065610] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  616.066305] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  616.066999] Call Trace:
[  616.067702]  skb_queue_purge+0x1c/0x30
[  616.068430]  rtw_usb_deinit_tx+0x2b/0x60 [rtw89core]
[  616.069143]  rtw_usb_probe.cold+0x3bf/0x5be [rtw89core]
[  616.069819]  usb_probe_interface+0xed/0x290
[  616.070380]  really_probe+0x2b0/0x3d0
[  616.070924]  driver_probe_device+0xbc/0x100
[  616.071476]  device_driver_attach+0x5d/0x70
[  616.072021]  __driver_attach+0x8f/0x150
[  616.072572]  ? device_driver_attach+0x70/0x70
[  616.073111]  bus_for_each_dev+0x7e/0xc0
[  616.073634]  driver_attach+0x1e/0x20
[... 344 lines cut ...]
[  189.814678] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d c3 f5 0c 00 f7 d8 64 89 01 48
[  189.814681] RSP: 002b:00007ffe4f95c108 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  189.814685] RAX: ffffffffffffffda RBX: 0000558ebe274cd0 RCX: 00007f8d1b8ec89d
[  189.814687] RDX: 0000000000000000 RSI: 00007f8d1b7c9ded RDI: 0000000000000006
[  189.814690] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000000000000000
[  189.814692] R10: 0000000000000006 R11: 0000000000000246 R12: 00007f8d1b7c9ded
[  189.814695] R13: 0000000000000000 R14: 0000558ebde7ff30 R15: 0000558ebe274cd0
[  189.814699] Modules linked in: option usb_wwan usbserial rtw89core(OE+) uas usb_storage xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c bpfilter br_netfilter bridge stp llc rfcomm aufs overlay cmac algif_hash algif_skcipher af_alg bnep vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) binfmt_misc nls_iso8859_1 uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common btrtl btbcm rtsx_usb_ms memstick videodev btintel bluetooth mc ecdh_generic ecc snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi ath10k_pci ath10k_core snd_hda_intel snd_intel_dspcfg snd_hda_codec mei_hdcp ath snd_hda_core mac80211 snd_hwdep snd_pcm mei_me intel_rapl_msr snd_seq_midi snd_seq_midi_event snd_rawmidi intel_rapl_common snd_seq x86_pkg_temp_thermal intel_powerclamp coretemp cfg80211 libarc4 mei kvm_intel snd_seq_device snd_timer snd kvm msi_wmi
[  189.814760]  soundcore sparse_keymap intel_pch_thermal nvidia_uvm(POE) rapl intel_cstate input_leds joydev mac_hid acpi_pad mxm_wmi intel_wmi_thunderbolt serio_raw sch_fq_codel msr parport_pc ppdev lp parport ip_tables x_tables autofs4 dm_mirror dm_region_hash dm_log hid_gt683r hid_generic rtsx_usb_sdmmc usbhid hid rtsx_usb nvidia_drm(POE) nvidia_modeset(POE) crct10dif_pclmul crc32_pclmul ghash_clmulni_intel nvidia(POE) i915 aesni_intel crypto_simd i2c_algo_bit cryptd glue_helper drm_kms_helper nvme syscopyarea sysfillrect psmouse sysimgblt fb_sys_fops i2c_i801 drm alx nvme_core ahci ipmi_devintf mdio libahci ipmi_msghandler wmi video
[  189.814808] CR2: 0000000fffffffe8
[  189.814812] ---[ end trace 4ecc688daff23d4a ]---
[  189.814818] RIP: 0010:skb_dequeue+0x50/0x70
[  189.814823] Code: 74 41 4c 39 e3 74 3c 8b 43 10 83 e8 01 89 43 10 49 8b 14 24 49 8b 44 24 08 49 c7 04 24 00 00 00 00 49 c7 44 24 08 00 00 00 00 <48> 89 42 08 48 89 10 4c 89 ef e8 81 76 21 00 4c 89 e0 5b 41 5c 41
[  189.814825] RSP: 0018:ffffb44142eff9d0 EFLAGS: 00010097
[  189.814829] RAX: ffff9f9b7a18bd08 RBX: ffff9f9b69b9d770 RCX: 000000008040003c
[  189.814831] RDX: 0000000fffffffe0 RSI: 0000000000000286 RDI: ffff9f9b69b9d784
[  189.814834] RBP: ffffb44142eff9e8 R08: 0000000000000000 R09: ffffffffc2246800
[  189.814836] R10: ffff9f9b7a18b700 R11: 0000000000000001 R12: ffff9f9b7a18bd00
[  189.814839] R13: ffff9f9b69b9d784 R14: ffff9f9b69b987a0 R15: ffff9f9b69b99e80
[  189.814842] FS:  00007f8d1b36a880(0000) GS:ffff9f9beebc0000(0000) knlGS:0000000000000000
[  189.814845] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  189.814848] CR2: 0000000fffffffe8 CR3: 000000044bb86001 CR4: 00000000003606e0
[  189.814851] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  189.814853] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
```

bjlockie, 2021-12-04:
I use lubuntu:
5.13.0-22-generic #22-Ubuntu
I think the generic kernel is the same as a vanilla one but I'm not sure.

I think the DLink appears as a storage device because they include the Windows driver onboard.

I noticed the log says "Product: 802.11ac WLAN Adapter", it should say "Product: 802.11ax WLAN Adapter".

I think the new Product IDs should be defines.
Maybe like (but I am not familiar with other USB drivers or the kernel coding style):
`/* USB Vendor/Product IDs */
#define RTW_USB_VENDOR_ID_REALTEK               0x0BDA
#define RTW_USB_PRODUCT_ID_REALTEK              0x885A

# D-Link DWA-X1850, monobloc, SMD metal + Airgain stacked PCB antenna
#define VENDOR_ID_DLINK                 0x2001
#define PRODUCT_ID_DLINK_DWA_X1850      0x3321

# ASUS USB-AX56, flippable antennas
#define VENDOR_ID_ASUS                  0x0B05
#define PRODUCT_ID_ASUS_USB_AX56        0x1997

#define VENDOR_ID_NETGEAR               0x0846
#define `VENDOR_ID_EDIMAX                0x7392
`

It is unfortunate it kills the USB stack.
I was hoping you could plug/unplug it and nothing would be affected.

bjlockie, 2021-12-04:
Where is this from?
`[  187.101756] usb 1-6: Product: 802.11ac WLAN Adapter`

bjlockie, 2021-12-04:
I'm going to blame Ubuntu.
I need a new distro too. :-(
>Lubuntu like all Ubuntu flavors, use Ubuntu kernels, built by the Ubuntu Kernel Team .
https://wiki.ubuntu.com/KernelTeam

s-2, 2021-12-04:
I'm on Mate 20.04.3 LTS, thus still using 5.4, though as far as I could tell from `Readme`, it should not even build if the kernel were the issue? Will setup a recent Debian VM though, maybe that helps :)

> Where is this from?
> `[ 187.101756] usb 1-6: Product: 802.11ac WLAN Adapter`

It's just the Product name string for the USB device (D-Link DWA-X1850), they forgot to update that for the new adapter I suppose, not a big deal, no one would see it anyways, as the Windows driver just shows the string as given in the driver's `.inf` file.

The USB VID / PID structure should indeed see some refactoring, it was neither sorted alphabetically or by VID. Also Netgear and Edimax do not have any of these dongles out there yet, these VIDs seem to be remainders from an earlier .ac driver?

s-2, 2021-12-06:
Finally got debian 11 up and running, same issues with kernel 5.10, e.g.:

```
[  166.775188] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.776261] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.777048] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.778008] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.779070] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.779765] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.781002] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.781949] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.784727] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.784731] rtwifi_usb 1-2:1.0: failed to submit urb, ret=-19
[  166.784733] rtwifi_usb 1-2:1.0: [ERR] Polling timeout
[  166.784734] rtwifi_usb 1-2:1.0: [ERR] addr: 6, 6
[  166.784736] rtwifi_usb 1-2:1.0: [ERR] val: 0, 2
[  166.784737] rtwifi_usb 1-2:1.0: failed to setup chip information
[  166.784771] BUG: unable to handle page fault for address: 0000000fffffffe8
[  166.784773] #PF: supervisor write access in kernel mode
[  166.784774] #PF: error_code(0x0002) - not-present page
[  166.784775] PGD 0 P4D 0 
[  166.784779] Oops: 0002 [#1] SMP PTI
[  166.784782] CPU: 0 PID: 1862 Comm: systemd-udevd Tainted: G           OE     5.10.0-9-amd64 #1 Debian 5.10.70-1
[  166.784783] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[  166.784808] RIP: 0010:skb_dequeue+0x4e/0x80
[  166.784810] Code: 39 e3 74 3a 8b 43 10 48 89 ef 83 e8 01 89 43 10 49 8b 44 24 08 49 c7 44 24 08 00 00 00 00 49 8b 14 24 49 c7 04 24 00 00 00 00 <48> 89 42 08 48 89 10 e8 76 1e 20 00 4c 89 e0 5b 5d 41 5c c3 45 31
[  166.784812] RSP: 0018:ffffa44d41cebbf0 EFLAGS: 00010097
[  166.784814] RAX: ffff8fb0862aaa88 RBX: ffff8fb089955870 RCX: 0000000080400038
[  166.784816] RDX: 0000000fffffffe0 RSI: 0000000000000286 RDI: ffff8fb089955884
[  166.784817] RBP: ffff8fb089955884 R08: 0000000000000001 R09: 0000000000000000
[  166.784818] R10: 0000000000000001 R11: 0000000000000000 R12: ffff8fb0862aaa80
[  166.784819] R13: 0000000fffffffe0 R14: ffff8fb089950800 R15: ffff8fb089951f00
[  166.784821] FS:  00007f16a71938c0(0000) GS:ffff8fb0fdc00000(0000) knlGS:0000000000000000
[  166.784823] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  166.784824] CR2: 0000000fffffffe8 CR3: 00000000020cc003 CR4: 00000000000706f0
[  166.784828] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  166.784829] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  166.784830] Call Trace:
[  166.784844]  skb_queue_purge+0x18/0x30
[  166.784862]  rtw_usb_probe.cold+0x3f4/0x626 [rtw89core]
[  166.784877]  usb_probe_interface+0xe2/0x2a0 [usbcore]
[  166.784881]  really_probe+0x37b/0x480
[  166.784893]  driver_probe_device+0xe1/0x150
[  166.784896]  device_driver_attach+0xa1/0xb0
[  166.784899]  __driver_attach+0x8a/0x150
[  166.784901]  ? device_driver_attach+0xb0/0xb0
[  166.784903]  ? device_driver_attach+0xb0/0xb0
[  166.784905]  bus_for_each_dev+0x78/0xc0
[  166.784908]  bus_add_driver+0x12b/0x1e0
[  166.784911]  driver_register+0x8b/0xe0
[  166.784920]  usb_register_driver+0x84/0x120 [usbcore]
[  166.784923]  ? 0xffffffffc0bab000
[  166.784927]  do_one_initcall+0x44/0x1d0
[  166.784950]  ? do_init_module+0x23/0x260
[  166.784955]  ? kmem_cache_alloc_trace+0xf5/0x200
[  166.784957]  do_init_module+0x5c/0x260
[  166.784967]  __do_sys_finit_module+0xb1/0x110
[  166.784974]  do_syscall_64+0x33/0x80
[  166.784984]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  166.784986] RIP: 0033:0x7f16a764c9b9
[  166.784989] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a7 54 0c 00 f7 d8 64 89 01 48
[  166.784990] RSP: 002b:00007ffde3cb5b38 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[  166.784992] RAX: ffffffffffffffda RBX: 0000556d50e10010 RCX: 00007f16a764c9b9
[  166.784993] RDX: 0000000000000000 RSI: 00007f16a77d7e2d RDI: 0000000000000011
[  166.784995] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000556d50c94820
[  166.784996] R10: 0000000000000011 R11: 0000000000000246 R12: 00007f16a77d7e2d
[  166.784997] R13: 0000000000000000 R14: 0000556d50e9a940 R15: 0000556d50e10010
[  166.784999] Modules linked in: rtw89core(OE+) mac80211 cfg80211 libarc4 uas usb_storage vboxvideo(OE) rfkill intel_rapl_msr intel_rapl_common intel_pmc_core_pltdrv intel_pmc_core intel_powerclamp ghash_clmulni_intel aesni_intel snd_intel8x0 libaes snd_ac97_codec crypto_simd joydev ac97_bus cryptd snd_pcm glue_helper rapl vboxguest(OE) snd_timer snd soundcore serio_raw evdev pcspkr sg ac fuse msr configfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic hid_generic usbhid hid sd_mod t10_pi crc_t10dif sr_mod cdrom crct10dif_generic ata_generic vmwgfx ttm crct10dif_pclmul crct10dif_common ahci libahci ata_piix crc32_pclmul drm_kms_helper cec crc32c_intel psmouse libata scsi_mod i2c_piix4 ohci_pci ehci_pci ohci_hcd ehci_hcd usbcore usb_common e1000 drm video button
[  166.785049] CR2: 0000000fffffffe8
[  166.785051] ---[ end trace 4a404770aa901859 ]---
[  166.785053] RIP: 0010:skb_dequeue+0x4e/0x80
[  166.785055] Code: 39 e3 74 3a 8b 43 10 48 89 ef 83 e8 01 89 43 10 49 8b 44 24 08 49 c7 44 24 08 00 00 00 00 49 8b 14 24 49 c7 04 24 00 00 00 00 <48> 89 42 08 48 89 10 e8 76 1e 20 00 4c 89 e0 5b 5d 41 5c c3 45 31
[... cut ...]
[  166.785063] R13: 0000000fffffffe0 R14: ffff8fb089950800 R15: ffff8fb089951f00
[  166.785065] FS:  00007f16a71938c0(0000) GS:ffff8fb0fdc00000(0000) knlGS:0000000000000000
[  166.785066] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  166.785067] CR2: 0000000fffffffe8 CR3: 00000000020cc003 CR4: 00000000000706f0
[  166.785069] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  166.785070] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
sebastian@debian-rtw89:~$ 
```

lwfinger, 2021-12-06:
I now have the Realtek USB driver for the rtl8852au ready at https://github.com.lwfinger/rtl8852au.git.

From the README.md for that repo:
This driver currently handles the following devices:

ASUS USB-AX56 with USB ID 0b05:1997
D-Link DWA-X1850 with USB IF 2001:3321
Realtek Demo Board with USB ID 0bda:8832
Realtek Demo Board with USB ID 0bda:885a
Realtek Demo Board with USB ID 0bda:885c

The D-Link DWA-X1850 comes with a configuration that appears to be a USB disk,
which contains a Windows driver. If a 'lsusb' command shows the ID 0bda:1a2b,
then this disk is mounted. The way to avoid this is to edit either file
/usr/lib/udev/rules.d/40-usb_modeswitch.rules, or
/lib/udev/rules.d/40-usb_modeswitch.rules, whichever is on your system, and add
the following lines:

# D-Link DWA-X1850 Wifi Dongle
ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="usb_modeswitch '/%k'"

When a USB device is plugged in, or detected at boot, this rule causes the utulity
usb_modeswitch to unload any 0bda:1a2b devices that it finds. If you have a
device with different ID, change the rule accordingly.

Now that I have a device (DWA-X1850), I will be fixing the errors with the USB part of rtw89.

bjlockie, 2021-12-06:
>I now have the Realtek USB driver for the rtl8852au ready at https://github.com.lwfinger/rtl8852au.git.
https://github.com/lwfinger/rtl8852au.git :-)

Can the rtw89 automatically do a modeswitch?

>Now that I have a device (DWA-X1850), I will be fixing the errors with the USB part of rtw89.
Awesome. :-)

s-2, 2021-12-06:
Omg, that is awesome =) 
Will try within the next days, thanks.

The D-Link indeed is the only one of the 4 currently available devices that uses the USB disk, with an SPI NOR; I wonder how this is factory-flashed, apparently using undocumented OEM commands in the chip?
Probably available only under NDA from Realtek, thus not gonna happen in any official driver. Not that `flashrom` support would be of any importance, but at least from a security perspective, I wonder how easily the NOR contents could be modified...
Reminds me of DWA-192 (the "death orb"), which was RTL8814AU, and IIRC used CD-ROM emulation that would auto-start and isntall a driver

One more thing, there is currently only the Fenvi device that comes with a WPS button (D-Link has it unpopulated on the PCB). Is there anything in the OEM driver hinting towards how this could be detected? Although there is no WPS support in most distros by default (even Android removed it completely), it is still a button after all... 

> Can the rtw89 automatically do a modeswitch?

I think this should happen via `usb_modeswitch` from userspace, not from a kernel driver. And some people might want to mess with the mass storage drive, who knows...

bjlockie, 2021-12-06:
>Will try within the next days, thanks.

I think that is expecting too much.

I'd settle for something next year that doesn't kill the USB stack. :-)

The flash drive is presumably not writeable and is of 0 use to Linux users.
I'd prefer the default to be do the modeswitch.
Whatever other USB drivers default to is fine with me. :-)

s-2, 2021-12-06:
> > Will try within the next days, thanks.
> 
> I think that is expecting too much.
> I'd settle for something next year that doesn't kill the USB stack. :-)

This was just meant to say: I won't find time to re-compile and test this once more today, but maybe within the next couple days, out of mere curiosity, looking forward to whatever more is about to come :) 

Just realized this is yet based on OEM driver, thus should not immediately crash the system.
Although I'm currently lacking the skill to estimate how much the USB interface implementation differs between OEM and rtw89 drivers, I wonder whether there might ultimately be two separate drivers, or USB would indeed be merged into rtw89 one day.

lwfinger, 2021-12-06:
I would not have announces the driver UNLESS it worked. It is stable and gets about 300 Mbps up/down when connecting to an AC1900 router. I do not have an 802.11ax (wifi6) router. It does NOT crash either itself or the kernel.

If you want to play around with the USB disk, plug the device into a Windows machine or VM that does not have a driver for the 8852au. That way you can read anything on the device. I much prefer usb_modeswitch to get rid of it without ever seeing it.

If you had been reading my comments, the goal is to merge rtw89usb with the rest of rtw89.

bjlockie, 2021-12-07:
Which repo is most likely to have code of good enough quality to make it into the kernel?

rtl8852au (working and based on crappy quality Realtek code?)
rtw89-usb (unknown)
rtw89 (currently crashes USB stack?)

lwfinger, 2021-12-07:
rtl8852au is a short-term fix to allow people to get their devices to work. No one will do the 6-months work to get that into kernel condition!!

rtw89-usb was folded into rtw89. I am working on the latter, but not the former. If I am successful, that is the one that will make it into the kernel.

bjlockie, 2021-12-07:
Ok, I will check out the rtw89 in the new year when I get it.
I have a really old Realtek USB device (I forget the chipset) and it was becoming hard to get an updated driver for recent kernels.
I much prefer just plug it in and it works. :-)
What was the biggest roadblock from getting earlier Realtek USB devices in the kernel since they had a PCIe driver?

lwfinger, 2021-12-07:
No one wrote one, other than for the RTL8192CU. Adding a USB driver for a chip that has a PCI driver is not trivial!

bjlockie, 2021-12-07:
I have these 2 and I have had other Realtek ones.
They all worked out of the box so I assumed they were in-kernel drivers.

```
Ethernet controller: Realtek Semiconductor Co., Ltd. RTL810xE PCI Express Fast Ethernet controller (rev 02)

Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)

$ modinfo r8169

license:        GPL
softdep:        pre: realtek
description:    RealTek RTL-8169 Gigabit Ethernet driver
author:         Realtek and the Linux r8169 crew <netdev@vger.kernel.org>
depends:        
retpoline:      Y
intree:         Y
name:           r8169
```

Why is this PCIe driver "intree."?

bjlockie, 2021-12-09:
PCI drivers for Realtek wireless cards seem to be in the kernel.
The 2 modules I checked say 'intree".
How do they get properly written drivers?
Is the PCI division that much better than the USB division? 
My hope for a Realtek wifi 6 usb driver is within the next 10 years. :-)
Maybe I'll have wifi 8 by then. :-)

closed 2022-09-24 by lwfinger