lwfinger/rtw89 #90: make: kmodsign: Command not found

emrng, 2021-12-04:
Hello,

I'm using:
Linux suse 5.3.18-59.34-default #1 SMP Thu Nov 11 12:18:45 UTC 2021 (a2a53aa) x86_64 x86_64 x86_64 GNU/Linux

I'm trying to install the driver. The 'make' seemed to go fine. However, for the next step, I found this message.

> sudo bash -c 'MOK_KEY_DIR=/var/lib/shim-signed/mok make sign-install'
[sudo] password for root: 
make -C /lib/modules/5.3.18-59.34-default/build M=/home/eva/Downloads/rtw89-main modules
make[1]: Entering directory '/usr/src/linux-5.3.18-59.34-obj/x86_64/default'
  Building modules, stage 2.
  MODPOST 3 modules
make[1]: Leaving directory '/usr/src/linux-5.3.18-59.34-obj/x86_64/default'
kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der rtw89core.ko
make: kmodsign: Command not found
make: *** [Makefile:95: sign] Error 127

Has anyone found a workaround for this?

lwfinger, 2021-12-04:
I do not support signing; however, I checked and openSUSE does not provide that utility.

Have you checked https://github.com/lwfinger/rtw89/issues/74?  That worked for me, up to the point where it failed because I do not have an efi system.

emrng, 2021-12-11:
Thanks for the prompt response.
I ended up disabling Secure Boot entirely. Not the most elegant solution, but I need WiFi to work.
Thanks so much for your help.