lwfinger/rtw89 #93: Error on sudo make install

coco-nebula, 2021-12-09:
Since a few days my WLAN isn't working anymore on Ubuntu, Kernel: 5.11.0-41-generic. 
I connected via wire and followed the steps:

`git pull
make
sudo make install
`
this is the output:

```
make -C /lib/modules/5.11.0-41-generic/build M=/home/moongoat/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-41-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-41-generic'
cp: cannot stat 'rtl8852au*.bin': No such file or directory
make: *** [Makefile:61: install] Error 1
```

coco-nebula, 2021-12-09:
I cloned the repo from scratch, after that everything worked fine. Thank you very much.

closed 2021-12-09 by coco-nebula

lwfinger, 2021-12-10:
Ubuntu has screwed up the kernel headers such that something that should be rebuilt.  Do a 'make clean' first. That will save you the download.

Oh, I really hate Ubuntu!!!!!!

coco-nebula, 2021-12-10:
Thank you for the information!
Lately I've considered switching the distro, I don't want to run into issues in the future. It's a pain seeing the WLAN icon vanish every few weeks 😂
Are there known issues for Arch if i may ask?
Thank you

lwfinger, 2021-12-10:
As long as you use an out-of-kernel driver, it will disappear every time the kernel is updated. That is true for most distros that run a kernel older than v5.15.. The distros that have rtw89 built in are openSUSE Tumbleweed, which runs a v5.15 kernel, and openSUSE Leap 15.3, which has a standard package for rtw89. There may be others that have the driver built in or backported.

I do not use Arch, thus I have no ideas what problems will occur.