lwfinger/rtw89 #95: sign-install error on kernel 5.11.0-41-generic ubuntu 20.04

ahammedshamil, 2021-12-11:
ubuntu2024@ubuntu:~/Downloads/rtw89$ uname -r
5.11.0-41-generic

ubuntu2024@ubuntu:~/Downloads/rtw89$ sudo make sign-install
make -C /lib/modules/5.11.0-41-generic/build M=/home/ubuntu2024/Downloads/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-41-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-41-generic'
mkdir -p ~/mok
pushd ~/mok
~/mok /home/ubuntu2024/Downloads/rtw89
openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"
Generating a RSA private key
.....+++++
.....................................................................+++++
writing new private key to 'MOK.priv'
-----
mokutil --import MOK.der
input password: 
input password again: 
popd
/bin/bash: line 0: popd: directory stack empty
make: *** [Makefile:99: sign] Error 1

ahammedshamil, 2021-12-11:
issue fixed by using branch V7

closed 2021-12-11 by ahammedshamil

lwfinger, 2021-12-11:
How do you propose that the directory stack is empty when 3 statements earlier, something was pushed on that very stack. Your distro is broken!! Switching to v7 is only a temporary fix as it will soon disappear as the code has already been copied to branch main.

lwfinger, 2021-12-12:
I reworked the Makefile to do away with the pushd/popd combo as that was failing. I also moved the signing files MOK.* to the root directory. In addition, I test for the existence of MOK.der before creating the keys, and skip the process if it is found. That will prevent spamming the EFI space. A simple 'make && sudo make sign-install' will sign and install the modules.

One thing: If you blow away the repo and reclone it, you will force the regeneration of the keys with the associated spamming of EFI space. Do a 'make clean && git pull' instead.

ahammedshamil, 2021-12-12:
System stuck after reboot. boot to the recovery kernel is the only way to unbrick my system. I reinstalled the kernel. installing the 5.16rc4 kernel is also a boot loop. Waiting for a stable release of the 5.16 kernel.

lwfinger, 2021-12-12:
I have no idea what happened. With the 5.16rc4 kernel, you had no need to deal with an external rtw89 driver. It is built in. It is possible with both rtw89_pci (The in-kernel version) and rtw89pci (The external version) that the boot process got confused. Reinstall 5.16rc4, do not generate and install this repo, and you should be OK.