lwfinger/rtw89 #97: association timeout

cunsdorfer, 2021-12-14:
The driver (master branch and all branches before) is not working for me.
The connection is not associated with following message:

```
[ 7684.786736] wlp4s0: authenticate with cc:32:e5:53:92:3d
[ 7685.022974] wlp4s0: send auth to cc:32:e5:53:92:3d (try 1/3)
[ 7685.025283] wlp4s0: authenticated
[ 7685.029376] wlp4s0: associate with cc:32:e5:53:92:3d (try 1/3)
[ 7685.133651] wlp4s0: associate with cc:32:e5:53:92:3d (try 2/3)
[ 7685.245401] wlp4s0: associate with cc:32:e5:53:92:3d (try 3/3)
[ 7685.357626] wlp4s0: association with cc:32:e5:53:92:3d timed out
```
```
dmesg | grep "rtw89"
[    4.792232] rtw89_pci 0000:04:00.0: enabling device (0000 -> 0003)
[    4.800409] rtw89_pci 0000:04:00.0: Firmware version 0.13.30.0, cmd version 0, type 1
[    4.800414] rtw89_pci 0000:04:00.0: Firmware version 0.13.30.0, cmd version 0, type 3
[    4.819352] rtw89_pci 0000:04:00.0: chip rfe_type is 1
[    4.826118] rtw89_pci 0000:04:00.0 wlp4s0: renamed from wlan0
```

Lubuntu 64 Bit - Kernel 5.8.0-63
Lenovo Notebook AMD Ryzen 7 - RTL8852AE 802.11ax PCIe Wireless Network Adapter

lspci: Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:8852]

The funny thing is, that the WiFi connection is working if I connect to FritzBox 7950. Connections to all other APs (I tried) is breaking with the error above.

Any idea?
Thanks in advance.

lwfinger, 2021-12-14:
Without seeing the "in the air" traffic, I have no idea what may be wrong. Is there any noted difference in the strengths of the signals? Use sudo iw dev wlp4s0 scan | egrep "signal|BSSD|freq" to get that information. Post that info along with the identification of the one that works and the ones that fail. You have authenticated, thus your wifi secret is correct.

Are you in the EU? Is your country code set appropriately?

What are the makes and models of the APs that fail?

cunsdorfer, 2021-12-15:
```
BSS cc:32:e5:53:92:3c(on wlp4s0)
        freq: 5180
        signal: -65.00 dBm
                 * OBSS non-GF present: 0
        Overlapping BSS scan params:
                 * BSS width channel transition delay factor: 5
                 * OBSS Scan Activity Threshold: 0.25 %
                 * BSS Transition
                 * center freq segment 1: 42
                 * center freq segment 2: 0
BSS cc:32:e5:53:92:3d(on wlp4s0)
        freq: 2422
        signal: -61.00 dBm
                 * OBSS non-GF present: 0
        Overlapping BSS scan params:
                 * BSS width channel transition delay factor: 5
                 * OBSS Scan Activity Threshold: 0.25 %
                 * BSS Transition
```
These values are with old "manual" frequency settings (not working)

I'm in the EU (Germany). 

The AP (one of them) that is failing is an TP-Link Archer C6.

While playing around with the AP settings, I got it working.

I changed from manual settings for 2.4 GHz / 5 GHz to "Smart Connect".
I manual settings I had disabled the 5GHz radio. In Smart Connect, both frequencies are enabled.

New values, that are working now:
```
BSS cc:32:e5:53:92:3c(on wlp4s0) -- associated
        freq: 5180
        signal: -72.00 dBm
                 * OBSS non-GF present: 0
        Overlapping BSS scan params:
                 * BSS width channel transition delay factor: 5
                 * OBSS Scan Activity Threshold: 0.25 %
                 * BSS Transition
                 * center freq segment 1: 42
                 * center freq segment 2: 0
BSS cc:32:e5:53:92:3d(on wlp4s0)
        freq: 2412
        signal: -56.00 dBm
                 * OBSS non-GF present: 0
        Overlapping BSS scan params:
                 * BSS width channel transition delay factor: 5
                 * OBSS Scan Activity Threshold: 0.25 %
                 * BSS Transition

```
So it seems to have something to do with the 5GHz frequency, that was disabled manually, but the AP firmware did not disable the sender completely ??

lwfinger, 2021-12-15:
That is strange that the AP was still broadcasting an SSID for 5G, but not prepared for anything to hook to it. That is definitely a TP-Link firmware bug.

closed 2021-12-15 by lwfinger