lwfinger/rtw89 #104: try to compile on linux 4.18 but fail

siyy123, 2021-12-26:
[make.log](https://github.com/lwfinger/rtw89/files/7776474/make.log)

XOMUS, 2021-12-26:
You on arch/manjaro?

XOMUS, 2021-12-26:
sudo pacman -Sy linux-$(uname -r)-headers yay
sudo pacman -Syu dkms base-devel --needed
yay -Sy rtw89-dkms-git

restart your pc/notebook

XOMUS, 2021-12-26:
And done

siyy123, 2021-12-26:
No I'm on centos 8 :(

XOMUS, 2021-12-26:
OK wait

XOMUS, 2021-12-26:
Try this and install
sudo yum install kernel-headers kernel-devel
sudo yum group install "C Development Tools and Libraries"

siyy123, 2021-12-26:
I'm sure that the procedures are not wrong, and this issue is only for helping backport. The kernel on centos 8 is too old. Though I can try elrepo kernels or use centos9, the kernels are not properly signed for secure boot and can't boot on my computer:(

XOMUS, 2021-12-26:
Use arch or manjaro cent os from servers

XOMUS, 2021-12-26:
Or debian

XOMUS, 2021-12-26:
Or mint mate

XOMUS, 2021-12-26:
https://media.tenor.com/images/1e144c0ed826a3fb112b1b18c75cedf1/tenor.gif

siyy123, 2021-12-26:
ok i know but just make the issue clear.

XOMUS, 2021-12-26:
Something is missing for make

XOMUS, 2021-12-26:
Try sudo yum install make gcc linux-headers-$(uname -r) build-essential git

lwfinger, 2021-12-26:
The problem is that his kernel is 4.18, and so far, I support only back to 5.4 for this driver. @XOMUS - I know you are trying to be helpful, but you need to watch a bit more before trying to help.

closed 2022-09-24 by lwfinger