lwfinger/rtw89 #105: No 5GHz Connectivity

DreadPirateLynx, 2021-12-28:
Recently bought a new laptop (HP Envy x360), and the wifi adapter requires the use of this driver. Unfortunately, I quickly began to run into connectivity issues, and I eventually identified 5GHz network signals as the problem. I can connect to 2.4GHz SSIDs and access the internet without issue. I cannot connect to 5GHz SSIDs at all. If a router is using the same SSID for both 2.4GHz & 5GHz signals, I can connect to the network, but lose the ability to send and receive all data within a minute of connecting. I can manually force the connection to the 2.4GHz MAC address to regain this ability.

This was the behavior I experienced in Garuda Linux. I've tried using the precompiled package from the distro repository as well as make installing the git repo, with no changes in behavior. I reinstalled Windows 11 to check that it was a driver issue and not a hardware one; the adapter had no issues with 5GHz networks with the Windows adapter. I've also tried installing Linux Mint, but experienced the exact same behavior as with Garuda. I've also tried changing the options available in the .conf file, but I couldn't find any combination of options that had any positive impact.

Not sure what more information you need from my terminal, but here's the info from lspci at least:
```
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
Subsystem: Hewlett-Packard Company Device 88e1
Flags: bus master, fast devsel, latency 0, IRQ 88, IOMMU group 9
I/O ports at f000 [size=256]
Memory at fce00000 (64-bit, non-prefetchable) [size=1M]
Capabilities: [40] Power Management version 3
Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
Capabilities: [70] Express Endpoint, MSI 00
Capabilities: [100] Advanced Error Reporting
Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-88-52-01
Capabilities: [158] Latency Tolerance Reporting
Capabilities: [160] L1 PM Substates
Kernel driver in use: rtw89_pci
Kernel modules: rtw89pci
```
Let me know what other information I can provide

lwfinger, 2021-12-28:
I am surprised that you cannot connect on the 5G band. I use it regularly without problems. As 5G and 2,4G use the same SSID, my guess is that your device is seeing a stronger signal on 5G and is doing a roaming operation.

Two things to check: 1. When the connection fails, run 'iw dev' and post the results. 2. run 'iw list > iwlist.txt' and attach iwlist.txt to this issue. That will show what capabilities are activated.

DreadPirateLynx, 2021-12-28:
```
phy#0
     Interface wlp2s0
          ifindex 2
          wdev 0x1
          addr c8:94:02:81:fa:e5
          ssid RWells Home
          type managed
          channel 157 (5785 MHz), width: 80 MHz, center1: 5775 MHz
          txpower 20.00 dBm
          multicast TXQ:
               qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
               0          0           0         0          0           0          0            0                  0
```
[iwlist.txt](https://github.com/lwfinger/rtw89/files/7786546/iwlist.txt)

lwfinger, 2021-12-28:
As I suspected, your interface has roamed to the 5G band (channel 157). I do not see anything wrong with the iw list.

Please run the following command and post the output:
sudo iw dev wlp2s0 scan | egrep "SSID|signal|freq"

DreadPirateLynx, 2021-12-28:
```
freq: 2437
signal: -35.00 dBm
SSID: MeowMe
freq: 2412
signal: -48.00 dBm
SSID: CenturyLink2725
freq: 2462
signal: -38.00 dBm
SSID: 
* SSID List
freq: 2462
signal: -57.00 dBm
SSID: EverdayImBuffering
* SSID List
freq: 2462
signal: -57.00 dBm
SSID: 
* SSID List
freq: 2462
signal: -38.00 dBm
SSID: RWells Home
* SSID List
freq: 2462
signal: -57.00 dBm
SSID: 
* SSID List
freq: 2462
signal: -57.00 dBm
SSID: 
* SSID List
freq: 2462
signal: -37.00 dBm
SSID: 
* SSID List
freq: 2462
signal: -38.00 dBm
SSID: 
* SSID List
freq: 2437
signal: -59.00 dBm
SSID: 56F069
freq: 5785
signal: -38.00 dBm
SSID: RWells Home
* SSID List
* center freq segment 1: 155
* center freq segment 2: 0
freq: 5180
signal: -43.00 dBm
SSID: MeowMe 5G
* center freq segment 1: 42
* center freq segment 2: 0
freq: 2462
signal: -80.00 dBm
SSID: Byrgenwerth
* center freq segment 1: 0
* center freq segment 2: 0
freq: 5785
signal: -58.00 dBm
SSID: CoxWiFi
* SSID List
* center freq segment 1: 155
* center freq segment 2: 0
freq: 5785
signal: -58.00 dBm
SSID: EverdayImBuffering
* SSID List
* center freq segment 1: 155
* center freq segment 2: 0
freq: 2462
signal: -81.00 dBm
SSID: Guest Network
* UTF-8 SSID
freq: 2462
signal: -80.00 dBm
SSID: Conference Room
* UTF-8 SSID
freq: 2462
signal: -81.00 dBm
SSID: IAG
* UTF-8 SSID
freq: 2412
signal: -79.00 dBm
SSID: NETGEAR04
freq: 5785
signal: -62.00 dBm
SSID: Tijavon5
* SSID List
* center freq segment 1: 155
* center freq segment 2: 0
freq: 5785
signal: -50.00 dBm
SSID: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
* center freq segment 1: 157
* center freq segment 2: 0
freq: 2412
signal: -56.00 dBm
SSID: 
* SSID List
freq: 2437
signal: -73.00 dBm
SSID: ARRIS-4E09
* center freq segment 1: 0
* center freq segment 2: 0
freq: 5785
signal: -39.00 dBm
SSID: CoxWiFi
* center freq segment 1: 155
* center freq segment 2: 0
* SSID List
freq: 5785
signal: -37.00 dBm
SSID: 
* center freq segment 1: 155
* center freq segment 2: 0
* SSID List
freq: 5785
signal: -60.00 dBm
SSID: 
* center freq segment 1: 155
* center freq segment 2: 0
* SSID List
freq: 2412
signal: -56.00 dBm
SSID: 
* SSID List
freq: 2412
signal: -56.00 dBm
SSID: 
* SSID List
freq: 2412
signal: -80.00 dBm
SSID: Conference Room
* UTF-8 SSID
freq: 2412
signal: -77.00 dBm
SSID: MM
* SSID List
* center freq segment 1: 0
* center freq segment 2: 0
freq: 2412
signal: -56.00 dBm
SSID: 
* SSID List
freq: 2412
signal: -76.00 dBm
SSID: IAG
* UTF-8 SSID
freq: 5785
signal: -34.00 dBm
SSID: 
* SSID List
freq: 5785
signal: -58.00 dBm
SSID: 
* center freq segment 1: 155
* center freq segment 2: 0
* SSID List
freq: 2412
signal: -56.00 dBm
SSID: Tijavon2.4
* SSID List
freq: 2412
signal: -79.00 dBm
SSID: Guest Network
* UTF-8 SSID
freq: 2412
signal: -78.00 dBm
SSID: Stinson-Internal
* UTF-8 SSID
freq: 2412
signal: -78.00 dBm
SSID: MM-UID0
* SSID List
* center freq segment 1: 0
* center freq segment 2: 0
freq: 2462
signal: -81.00 dBm
SSID: Guest Network
* UTF-8 SSID
freq: 2462
signal: -81.00 dBm
SSID: Conference Room
* UTF-8 SSID
freq: 5580
signal: -38.00 dBm
SSID: 
* SSID List
freq: 5640
signal: -37.00 dBm
SSID: 
* SSID List
freq: 2462
signal: -80.00 dBm
SSID: IAG
* UTF-8 SSID
freq: 5785
signal: -62.00 dBm
SSID: 
* center freq segment 1: 155
* center freq segment 2: 0
* SSID List
freq: 5785
signal: -38.00 dBm
SSID:
* center freq segment 1: 155
* center freq segment 2: 0
* SSID List
freq: 5785
signal: -39.00 dBm
SSID:
* center freq segment 1: 155
* center freq segment 2: 0
* SSID List
freq: 5785
signal: -62.00 dBm
SSID: CoxWiFi
* SSID List
* center freq segment 1: 155
* center freq segment 2: 0
freq: 5785
signal: -59.00 dBm
SSID:
* center freq segment 1: 155
* center freq segment 2: 0
* SSID List
```

lwfinger, 2021-12-29:
I am surprised that you get the same signal for both bands (-38 dBm). I get -24 dBm on the 2.4G band and -42 on 5G.

I was looking for a way to disable the 5G band on a system, but I found no way to do it. It seems that you do not own the AP, and cannot change the SSID for the 5G band the way that MeowMe is set up.

DreadPirateLynx, 2021-12-30:
\
Well, I have admin access to _this_ AP, but my ability to use this laptop's wifi anywhere other than here is the issue. There's a surprising lack of 2.4GHz public signals in my area.

Good news is that if this can't be fixed, I still have time to return the laptop.

lwfinger, 2021-12-30:
Then you had better return it. I did set my router to use the same SSID for both bands. Once I did that, I could no longer connect to the 5G band. The scans in the NetworkManager applet only showed the 2.4 GHz one.

If you change the 5G SSID to something different, does %G then work?

DreadPirateLynx, 2021-12-30:
No. If I separate the two bands I can't connect to the 5Ghz one at all

lwfinger, 2021-12-30:
I got an Email that you had separated the bands, but I do not see it here. Can Windows 11 connect to the 5G now?

DreadPirateLynx, 2021-12-31:
Windows 11 could always connect. Separating the bands doesn't do anything to help my connection

lwfinger, 2021-12-31:
Do you mean that with the bands separated. that you still lose your 2.4 G connection? If that is true, then the only other thing I can suggest is that you try a couple of module parameters. Before I suggest those, please run the following command after the connection is lost:
dmesg > dmesg.txt

Attach the file dmesg.txt to this thread. That way I will see if any errors or warnings are being triggered.

DreadPirateLynx, 2022-01-06:
Here's the output of dmesg after losing connectivity on a shared SSID:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/7824494/dmesg.txt)

lwfinger, 2022-01-07:
I think I see your problem. It is in the line "Rate marked as a VHT rate but data is invalid: MCS: 10, NSS: 2" in your dmesg output. The problem is that your AP is sending some invalid data. That issue is discussed and worked-around in a recent submission. The problem is that I had not yet installed a number of earlier patches. When I tried them, I broke the driver, and ran out of time trying to find that problem.

I will not be available until Monday. I should be able to find the problem be next Wednesday at the latest.

DreadPirateLynx, 2022-01-07:
That's great news! It's still within my return window! I really don't want to have to find a replacement for this laptop, because other than this particular issue, I've been absolutely loving it. The significant open-box discount I got on it was great at the time, but if I needed to find a replacement for it, I'd either need to pony up more dough or settle for something worse.

DreadPirateLynx, 2022-01-07:
Sorry, closed by accident. There should really be a confirmation dialog on that option, lol.

lwfinger, 2022-01-09:
I do not think you need to return the laptop; however, I will need more time to include that patch. Keep the 2.4 and 5G bands separate for the moment.

lwfinger, 2022-01-10:
As you will see from the notification above, I have pushed a fix for your 5G problems. Please do a 'git pull' on the master branch and try the new code.

DreadPirateLynx, 2022-01-10:
after doing a pull/clean/make/install and reboot I'm still having the same issue. Most recent dmesg.txt:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/7842180/dmesg.txt)

lwfinger, 2022-01-11:
Actually, there has to be a companion patch to the kernel:

diff --git a/net/mac80211/rx.c b/net/mac80211/rx.c
index 0544563ede52..d2e8b84ed283 100644
--- a/net/mac80211/rx.c
+++ b/net/mac80211/rx.c
@@ -4924,7 +4924,7 @@ void ieee80211_rx_list(struct ieee80211_hw *hw, struct ieee80211_sta *pubsta,
                                goto drop;
                        break;
                case RX_ENC_VHT:
-                       if (WARN_ONCE(status->rate_idx > 9 ||
+                       if (WARN_ONCE(status->rate_idx > 11 ||
                                      !status->nss ||
                                      status->nss > 8,
                                      "Rate marked as a VHT rate but data is invalid: MCS: %d, NSS: %d\n",

Until that change is made, the error will continue.

lwfinger, 2022-01-11:
Unless you want to wait for kernel 5.17 in about 2 months, you need to install the kernel source and edit file net/mac80211/rx.c. In line 4947, change the "9" to an "11". Build and install that kernel, then rebuild and install this driver.

DreadPirateLynx, 2022-01-11:
That's kind of what I was leaning towards after looking at the kernel git repo. Is this definitely in the pipeline for 5.17 then? I've never built a kernel from source before and it kind of intimidates me. I'm definitely going to give it a try, but knowing that this will be in the next version means I don't have to worry if I can't get it working myself. It would also make me more comfortable with allowing my return window to expire before I get it working.

lwfinger, 2022-01-12:
No, it is not necessarily in the 5.17 kernel. It was submitted too late for 5.16, but not yet accepted.

Is your machine available via SSH? If so, I am willing to build a kernel for you. It is not too difficult. If not, I will guide you through the process.

DreadPirateLynx, 2022-01-12:
Well, if this is something I could potentially need to do again the next time I update my kernel, it's probably better if I learn how to do it myself rather than relying on you to do it for me. I'm reading through some tutorials and plan to take a crack at it tomorrow. I wouldn't object if you'd like to provide some guidance though. 

Thanks again for all your help, btw.

DreadPirateLynx, 2022-01-12:
Well, I don't have any config- files in my /boot, so I just had to use the defaults provided by 'make menuconfig'. After running 'sudo make modules_install install', I get "Cannot find LILO." When I reboot, I still only have the ability to choose between my default and fallback kernel, with no option for the new one. I'm guessing it isn't configured properly for grub, but not sure what I need to change.

lwfinger, 2022-01-12:
That LILO error means that your distro does not implement one of the scripts needed to install a new kernel. We could probably work around that problem, but it is likely better to follow their method, whatever it is.

I hope the default config has what is needed to make your system work. One other thing to check. Does your system have a file named /proc/config.gz. If so, then run the command 'zcat /proc/config.gz > .config" while in your kernel source directory.  

You are going to need to follow the instructions for your distro.

DreadPirateLynx, 2022-01-12:
you know, given Arch's commitment to the KISS principle, it's mind-boggling how much more complicated they manage to make everything

lwfinger, 2022-01-13:
Simple means what the devs understand. It is not user centric. Keep plugging.

lwfinger, 2022-01-19:
I found one patch for this problem that I missed earlier. With it, you should be able to run with a standard kernel. Note that this fix is only in the repo, and not in the kernel.

DreadPirateLynx, 2022-01-19:
Funny story:

I ended up returning the laptop over the weekend and started looking for another one. Unfortunately, there weren't a lot of options considering I was wanting to avoid intel and nvidia, and I happened upon a heavily discounted, upgraded version of the laptop I had just returned. I confirmed that the wifi adapter was indeed the same, but it was the only part in the laptop that was, and the computer came with a 90-day return policy, so I figured it couldn't hurt to keep trying, especially since this wasn't a widespread problem, and there was a chance it would just work. 

It didn't, but today I saw this post, pulled and rebuilt the driver, and everything is working! Thank you so much for all your help!

DreadPirateLynx, 2022-01-19:
So...
Did a system update that upgraded to kernel 5.16 and everything broke. After installing the new kernel, I tried clean/make/installing the git repo, and it seemed to work fine. But when I modprobe -rv rtw89pci, I get no output to the console, and when I modprobe -v rtw89pci, it says device in use.

lwfinger, 2022-01-20:
You have a conflict between rtw89_pci from the kernel and rtw89pci from this repo. You need to blacklist the one from the kernel.

DreadPirateLynx, 2022-01-20:
Blacklisting rtw89_pci and reinstalling rtw89pci got everything working again. Thanks again for all your help! I don't really know how driver updates get incorporated into kernel releases, so any idea what the earliest (version number) the kernel provided one will include this update?

closed 2022-01-20 by DreadPirateLynx

lwfinger, 2022-01-20:
Kernel 5.17 would be the earliest that this would be combined, and that will not be out for at least 2 months.

I would like you to do one thing for me. The main developer and I are identifying the APs with this problem in order to apply quirks for those with the difficulty so as not to throttle the speed for APs that are OK. Please run the command

sudo iw dev wlp2s0 scan | egrep "\(on |SSID"

Find the SSID for your AP and send me the BSS line that is just before the SSID output. If you want to preserve anonymity, you can replace the last 2 octants (the serial number) with zeros. All I need are the first 3, which identify the manufacturer.

DreadPirateLynx, 2022-01-20:
I tried running it but the only output I got was
```
grep: Unmatched ( or \(
command failed: No such device (-19)
```
After changing the device to wlo1 (different hardware configuration than before, and this is the device reported by iw dev):
```
grep: Unmatched ( or \(
```
Then it pauses for a few seconds, then gives me no further output before my normal terminal prompt.

lwfinger, 2022-01-20:
Yes, that is it. This block is owned by Technicolor CH USA Inc. Thus far, all users with the problem have the same vendor.

DreadPirateLynx, 2022-01-20:
I must have really bad luck then, because after I couldn't get it to work at the first four places, I didn't bother trying anywhere else. I can probably pull that info from those APs as well over the next couple of days just to verify that they're all the same manufacturer.