lwfinger/rtw89 #106: can't load USB firmware

bjlockie, 2021-12-29:
Does it first load as if it is a disk and then reload as a wifi adapter?
```
 sudo dmesg
[   45.357055] usb 1-1: new high-speed USB device number 6 using xhci_hcd
[   45.505481] usb 1-1: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 0.00
[   45.505490] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   45.505493] usb 1-1: Product: DISK
[   45.505495] usb 1-1: Manufacturer: Realtek
[   45.545589] usb-storage 1-1:1.0: USB Mass Storage device detected
[   45.545705] scsi host2: usb-storage 1-1:1.0
[   45.545761] usbcore: registered new interface driver usb-storage
[   45.547651] usbcore: registered new interface driver uas
[   46.093828] usb 1-1: USB disconnect, device number 6
[   46.869072] usb 1-1: new high-speed USB device number 7 using xhci_hcd
[   47.017655] usb 1-1: New USB device found, idVendor=2001, idProduct=3321, bcdDevice= 0.00
[   47.017665] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   47.017668] usb 1-1: Product: 802.11ac WLAN Adapter
[   47.017671] usb 1-1: Manufacturer: Realtek
[   47.017673] usb 1-1: SerialNumber: 00e04c000001
[   47.047425] rtw89core: loading out-of-tree module taints kernel.
[   47.047795] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[   47.051444] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (0):
[   47.051448] rtwifi_usb 1-1:1.0: bLength=7
[   47.051449] rtwifi_usb 1-1:1.0: bDescriptorType=5
[   47.051450] rtwifi_usb 1-1:1.0: bEndpointAddress=84
[   47.051451] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[   47.051452] rtwifi_usb 1-1:1.0: bInterval=0
[   47.051453] rtwifi_usb 1-1:1.0: USB: dir in endpoint num 4
[   47.051454] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (1):
[   47.051455] rtwifi_usb 1-1:1.0: bLength=7
[   47.051455] rtwifi_usb 1-1:1.0: bDescriptorType=5
[   47.051456] rtwifi_usb 1-1:1.0: bEndpointAddress=5
[   47.051457] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[   47.051458] rtwifi_usb 1-1:1.0: bInterval=0
[   47.051459] rtwifi_usb 1-1:1.0: USB: out endpoint num 5
[   47.051460] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (2):
[   47.051461] rtwifi_usb 1-1:1.0: bLength=7
[   47.051461] rtwifi_usb 1-1:1.0: bDescriptorType=5
[   47.051462] rtwifi_usb 1-1:1.0: bEndpointAddress=6
[   47.051463] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[   47.051464] rtwifi_usb 1-1:1.0: bInterval=0
[   47.051465] rtwifi_usb 1-1:1.0: USB: out endpoint num 6
[   47.051465] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (3):
[   47.051466] rtwifi_usb 1-1:1.0: bLength=7
[   47.051467] rtwifi_usb 1-1:1.0: bDescriptorType=5
[   47.051468] rtwifi_usb 1-1:1.0: bEndpointAddress=7
[   47.051468] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[   47.051469] rtwifi_usb 1-1:1.0: bInterval=0
[   47.051470] rtwifi_usb 1-1:1.0: USB: out endpoint num 7
[   47.051471] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (4):
[   47.051472] rtwifi_usb 1-1:1.0: bLength=7
[   47.051472] rtwifi_usb 1-1:1.0: bDescriptorType=5
[   47.051473] rtwifi_usb 1-1:1.0: bEndpointAddress=9
[   47.051474] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[   47.051475] rtwifi_usb 1-1:1.0: bInterval=0
[   47.051475] rtwifi_usb 1-1:1.0: USB: out endpoint num 9
[   47.051476] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (5):
[   47.051477] rtwifi_usb 1-1:1.0: bLength=7
[   47.051478] rtwifi_usb 1-1:1.0: bDescriptorType=5
[   47.051479] rtwifi_usb 1-1:1.0: bEndpointAddress=a
[   47.051479] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[   47.051480] rtwifi_usb 1-1:1.0: bInterval=0
[   47.051481] rtwifi_usb 1-1:1.0: USB: out endpoint num 10
[   47.051482] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (6):
[   47.051483] rtwifi_usb 1-1:1.0: bLength=7
[   47.051483] rtwifi_usb 1-1:1.0: bDescriptorType=5
[   47.051484] rtwifi_usb 1-1:1.0: bEndpointAddress=b
[   47.051485] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[   47.051486] rtwifi_usb 1-1:1.0: bInterval=0
[   47.051486] rtwifi_usb 1-1:1.0: USB: out endpoint num 11
[   47.051487] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (7):
[   47.051488] rtwifi_usb 1-1:1.0: bLength=7
[   47.051489] rtwifi_usb 1-1:1.0: bDescriptorType=5
[   47.051490] rtwifi_usb 1-1:1.0: bEndpointAddress=c
[   47.051490] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[   47.051491] rtwifi_usb 1-1:1.0: bInterval=0
[   47.051492] rtwifi_usb 1-1:1.0: USB: out endpoint num 12
[   47.051493] rtwifi_usb 1-1:1.0: USB_SPEED_HIGH
[   47.051493] rtwifi_usb 1-1:1.0: USB: bulkout_size: 512
[   47.051667] rtwifi_usb 1-1:1.0: Direct firmware load for rtw89/rtl8852a_u3_nic.bin failed with error -2
[   47.051670] rtwifi_usb 1-1:1.0: failed to request firmware
[   47.051705] rtwifi_usb 1-1:1.0: failed to wait firmware completion
[   47.051710] rtwifi_usb 1-1:1.0: failed to setup chip information
[   47.051751] rtwifi_usb: probe of 1-1:1.0 failed with error -22
[   47.051770] usbcore: registered new interface driver rtwifi_usb

```

```
$ sudo modinfo rtw89usb
filename:       /lib/modules/5.13.0-22-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89usb.ko
license:        Dual BSD/GPL
description:    Realtek 802.11ax wireless USB driver
author:         Realtek Corporation
srcversion:     6FB92775AF2C9EC509E215D
alias:          usb:v2001p3321d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1997d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp885Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp885Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8832d*dc*dsc*dp*ic*isc*ip*in*
depends:        rtw89core,mac80211
retpoline:      Y
name:           rtw89usb
vermagic:       5.13.0-22-generic SMP mod_unload modversions
```

bjlockie, 2021-12-29:
`$ sudo cp /lib/firmware/rtw89/rtw8852a_fw.bin /lib/firmware/rtw89/rtl8852a_u3_nic.bin` fixed that.

lwfinger, 2021-12-29:
The driver is not working yet. It does start as a USB disk but the modeswitch facility removes that disk and exposes the wifi device.

If you want to use the device now, clone github.com/lwfinger/rtw8852au.git. That one works with the RTW8852AU devices.

closed 2021-12-29 by bjlockie

bjlockie, 2021-12-29:
Let me know if I can help.
Enjoy the holidays.