lwfinger/rtw89 #108: USB: failed to submit urb

bjlockie, 2021-12-29:
```
[ 5783.469803] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[ 5783.618573] usb 1-1: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 0.00
[ 5783.618582] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 5783.618585] usb 1-1: Product: DISK
[ 5783.618587] usb 1-1: Manufacturer: Realtek
[ 5783.620854] usb-storage 1-1:1.0: USB Mass Storage device detected
[ 5783.621053] scsi host2: usb-storage 1-1:1.0
[ 5784.211833] usb 1-1: USB disconnect, device number 8
[ 5784.985795] usb 1-1: new high-speed USB device number 9 using xhci_hcd
[ 5785.134741] usb 1-1: New USB device found, idVendor=2001, idProduct=3321, bcdDevice= 0.00
[ 5785.134748] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5785.134751] usb 1-1: Product: 802.11ac WLAN Adapter
[ 5785.134754] usb 1-1: Manufacturer: Realtek
[ 5785.134755] usb 1-1: SerialNumber: 00e04c000001
[ 5785.156760] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (0):
[ 5785.156765] rtwifi_usb 1-1:1.0: bLength=7
[ 5785.156766] rtwifi_usb 1-1:1.0: bDescriptorType=5
[ 5785.156767] rtwifi_usb 1-1:1.0: bEndpointAddress=84
[ 5785.156768] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[ 5785.156769] rtwifi_usb 1-1:1.0: bInterval=0
[ 5785.156770] rtwifi_usb 1-1:1.0: USB: dir in endpoint num 4
[ 5785.156771] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (1):
[ 5785.156772] rtwifi_usb 1-1:1.0: bLength=7
[ 5785.156773] rtwifi_usb 1-1:1.0: bDescriptorType=5
[ 5785.156774] rtwifi_usb 1-1:1.0: bEndpointAddress=5
[ 5785.156775] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[ 5785.156776] rtwifi_usb 1-1:1.0: bInterval=0
[ 5785.156777] rtwifi_usb 1-1:1.0: USB: out endpoint num 5
[ 5785.156778] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (2):
[ 5785.156779] rtwifi_usb 1-1:1.0: bLength=7
[ 5785.156780] rtwifi_usb 1-1:1.0: bDescriptorType=5
[ 5785.156781] rtwifi_usb 1-1:1.0: bEndpointAddress=6
[ 5785.156782] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[ 5785.156783] rtwifi_usb 1-1:1.0: bInterval=0
[ 5785.156784] rtwifi_usb 1-1:1.0: USB: out endpoint num 6
[ 5785.156785] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (3):
[ 5785.156786] rtwifi_usb 1-1:1.0: bLength=7
[ 5785.156787] rtwifi_usb 1-1:1.0: bDescriptorType=5
[ 5785.156787] rtwifi_usb 1-1:1.0: bEndpointAddress=7
[ 5785.156788] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[ 5785.156789] rtwifi_usb 1-1:1.0: bInterval=0
[ 5785.156790] rtwifi_usb 1-1:1.0: USB: out endpoint num 7
[ 5785.156791] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (4):
[ 5785.156792] rtwifi_usb 1-1:1.0: bLength=7
[ 5785.156793] rtwifi_usb 1-1:1.0: bDescriptorType=5
[ 5785.156794] rtwifi_usb 1-1:1.0: bEndpointAddress=9
[ 5785.156795] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[ 5785.156796] rtwifi_usb 1-1:1.0: bInterval=0
[ 5785.156797] rtwifi_usb 1-1:1.0: USB: out endpoint num 9
[ 5785.156798] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (5):
[ 5785.156799] rtwifi_usb 1-1:1.0: bLength=7
[ 5785.156800] rtwifi_usb 1-1:1.0: bDescriptorType=5
[ 5785.156801] rtwifi_usb 1-1:1.0: bEndpointAddress=a
[ 5785.156802] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[ 5785.156803] rtwifi_usb 1-1:1.0: bInterval=0
[ 5785.156803] rtwifi_usb 1-1:1.0: USB: out endpoint num 10
[ 5785.156804] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (6):
[ 5785.156805] rtwifi_usb 1-1:1.0: bLength=7
[ 5785.156806] rtwifi_usb 1-1:1.0: bDescriptorType=5
[ 5785.156807] rtwifi_usb 1-1:1.0: bEndpointAddress=b
[ 5785.156808] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[ 5785.156809] rtwifi_usb 1-1:1.0: bInterval=0
[ 5785.156810] rtwifi_usb 1-1:1.0: USB: out endpoint num 11
[ 5785.156811] rtwifi_usb 1-1:1.0: 
               usb endpoint descriptor (7):
[ 5785.156812] rtwifi_usb 1-1:1.0: bLength=7
[ 5785.156813] rtwifi_usb 1-1:1.0: bDescriptorType=5
[ 5785.156814] rtwifi_usb 1-1:1.0: bEndpointAddress=c
[ 5785.156815] rtwifi_usb 1-1:1.0: wMaxPacketSize=512
[ 5785.156816] rtwifi_usb 1-1:1.0: bInterval=0
[ 5785.156817] rtwifi_usb 1-1:1.0: USB: out endpoint num 12
[ 5785.156818] rtwifi_usb 1-1:1.0: USB_SPEED_HIGH
[ 5785.156819] rtwifi_usb 1-1:1.0: USB: bulkout_size: 512
[ 5785.157307] rtwifi_usb 1-1:1.0: Firmware version 0.13.33.0, cmd version 0, type 1
[ 5785.157311] rtwifi_usb 1-1:1.0: Firmware version 0.13.33.0, cmd version 0, type 3
[ 5785.157416] rtwifi_usb 1-1:1.0: MAC has already powered on
[ 5785.678158] usb 1-1: USB disconnect, device number 9
[ 5785.678975] rtwifi_usb 1-1:1.0: failed to submit urb, ret=-19
...
[ 5787.161295] rtwifi_usb 1-1:1.0: failed to submit urb, ret=-19
[ 5787.161299] rtwifi_usb 1-1:1.0: [ERR] Polling timeout
[ 5787.161300] rtwifi_usb 1-1:1.0: [ERR] addr: 6, 6
[ 5787.161302] rtwifi_usb 1-1:1.0: [ERR] val: 0, 2
[ 5787.161305] rtwifi_usb 1-1:1.0: failed to setup chip information
[ 5787.161531] rtwifi_usb: probe of 1-1:1.0 failed with error -16
[ 5787.161603] usbcore: registered new interface driver rtwifi_usb
[ 5787.433769] usb 1-1: new full-speed USB device number 10 using xhci_hcd
[ 5787.561790] usb 1-1: device descriptor read/64, error -71
[ 5787.797734] usb 1-1: device descriptor read/64, error -71
[ 5788.033780] usb 1-1: new full-speed USB device number 11 using xhci_hcd
[ 5788.161780] usb 1-1: device descriptor read/64, error -71
[ 5788.397804] usb 1-1: device descriptor read/64, error -71
[ 5788.505822] usb usb1-port1: attempt power cycle
[ 5789.157722] usb 1-1: new full-speed USB device number 12 using xhci_hcd
[ 5789.157891] usb 1-1: Device not responding to setup address.
[ 5789.365937] usb 1-1: Device not responding to setup address.
[ 5789.573732] usb 1-1: device not accepting address 12, error -71
[ 5789.701753] usb 1-1: new full-speed USB device number 13 using xhci_hcd
[ 5789.701929] usb 1-1: Device not responding to setup address.
[ 5789.909962] usb 1-1: Device not responding to setup address.
[ 5790.117740] usb 1-1: device not accepting address 13, error -71
[ 5790.117843] usb usb1-port1: unable to enumerate USB device

```

XOMUS, 2022-01-06:
Connect to rear usb port

lwfinger, 2022-01-06:
@XOMUS: Do you really think that changing the port is going to fix a faulty driver? The USB driver for rtl8852au is incomplete, and I have little time to work on it!

closed 2022-01-06 by lwfinger