lwfinger/rtw89 #112: Driver suddenly stopped working

tomerarazy, 2022-01-04:
Sorry for the not so informative title, I'm not really sure what's happening.
I've been using this driver (awesome job btw) for the last 6 month or so with no problem, and suddenly today it stopped working - not after restart or anything, just "suddenly".
Now there is no WiFi adapter, i've tried reinstalling the driver (from the latest `main` branch), unloading and loading the module but nothing.
Everything (installation, loading and unloading) seems to be fine but there is WiFi adapter.
I'm using Ubuntu 20.04 and Lenovo ideapad 5 Pro 16".

This is the `dmesg` log:
```
root@tomer-ideapad:~# dmesg | grep "rtw"
[    2.558129] rtw89core: loading out-of-tree module taints kernel.
[    2.609018] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[    2.624890] rtw89_pci 0000:03:00.0: enabling device (0000 -> 0003)
[    2.630919] rtw89_pci 0000:03:00.0: Firmware version 0.13.33.0, cmd version 0, type 1
[    2.630929] rtw89_pci 0000:03:00.0: Firmware version 0.13.33.0, cmd version 0, type 3
[    2.650205] rtw89_pci 0000:03:00.0: chip rfe_type is 1
[    2.656070] rtw89_pci 0000:03:00.0 wlo1: renamed from wlan0
```

tomerarazy, 2022-01-04:
After doing some digging with the Network manager it seems that the WiFI was turned off for some reason, fixed it by running `nmcli radio wifi on`.

Not sure if this has anything to do with the driver or not...

lwfinger, 2022-01-04:
It does not. You managed to get it into airplane mode. Once you did that, the wifi was silenced, which is what the dmesg output said. The driver loaded the firmware, and udev renamed the device. At that point, it could only wait until airplane mode was turned off. I expect that a function key handles this.

closed 2022-01-04 by lwfinger