lwfinger/rtw89 #113: polling timeout after resume

ceisserer, 2022-01-09:
Already filed against the upstream codebase: https://bugzilla.kernel.org/show_bug.cgi?id=215243

On my Ideapad 5 Pro (ryzen 5600h based), wifi is broken after each standby/resume. 
Happens with the in-tree version of 5.16-rc as well as with the latest version here on github.

ceisserer, 2022-01-09:
[rtw89_supend_resume.txt](https://github.com/lwfinger/rtw89/files/7834653/rtw89_supend_resume.txt)

lwfinger, 2022-01-09:
Your error is "rtw89_pci 0000:03:00.0: can't change power state from D3hot to D0 (config space inaccessible)."

Have you done what README.md tells you to di in that case?

ceisserer, 2022-01-10:
Thanks for the hint, indeed I haven't had the work-around in place.

However, even with the pre/post suspend script (or when manually unloading/loading the module), I still get the same issue:

[  124.433619] rtw89_pci 0000:03:00.0: can't change power state from D3cold to D0 (config space inaccessible)
[  124.436805] rtw89_pci 0000:03:00.0: no suitable firmware found
[  124.436811] rtw89_pci 0000:03:00.0: failed to recognize firmware
[  124.436812] rtw89_pci 0000:03:00.0: failed to setup chip information
[  124.437482] rtw89_pci: probe of 0000:03:00.0 failed with error -2

ceisserer, 2023-01-08:
could this be re-opened please - it has been closed without any update and was not needinfo.

after all the issue is still present with mainline linux ( 6.0.16-300.fc37.x86_64 ) on my ideapad 5 pro with the latest bios - which works fine running windows.

[16574.779123] rtw89_8852ae 0000:03:00.0: [ERR] Polling timeout
[16574.779129] rtw89_8852ae 0000:03:00.0: [ERR] addr: 5, 5
[16574.779130] rtw89_8852ae 0000:03:00.0: [ERR] val: FF, 0
[16576.779939] rtw89_8852ae 0000:03:00.0: [ERR] Polling timeout
[16576.779945] rtw89_8852ae 0000:03:00.0: [ERR] addr: 5, 5
[16576.779947] rtw89_8852ae 0000:03:00.0: [ERR] val: FF, 0
[16578.781118] rtw89_8852ae 0000:03:00.0: [ERR] Polling timeout
[16578.781122] rtw89_8852ae 0000:03:00.0: [ERR] addr: 5, 5
[16578.781123] rtw89_8852ae 0000:03:00.0: [ERR] val: FF, 0
[16580.782108] rtw89_8852ae 0000:03:00.0: [ERR] Polling timeout
[16580.782111] rtw89_8852ae 0000:03:00.0: [ERR] addr: 5, 5
[16580.782113] rtw89_8852ae 0000:03:00.0: [ERR] val: FF, 0
[16580.782115] rtw89_8852ae 0000:03:00.0: mac init fail, ret:-16
[16580.782117] ------------[ cut here ]------------
[16580.782117] Hardware became unavailable upon resume. This could be a software issue prior to suspend or a hardware issue.
[16580.782139] WARNING: CPU: 4 PID: 23188 at net/mac80211/util.c:2436 ieee80211_reconfig+0x88/0x1530 [mac80211]
[16580.782174] Modules linked in: tls rfcomm snd_seq_dummy snd_hrtimer cdc_ether r8152 vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) qrtr bnep sunrpc binfmt_misc vfat fat snd_sof_amd_renoir intel_rapl_msr intel_rapl_common uvcvideo snd_hda_codec_realtek snd_sof_amd_acp snd_sof_pci snd_hda_codec_generic videobuf2_vmalloc videobuf2_memops snd_sof ledtrig_audio snd_hda_codec_hdmi videobuf2_v4l2 videobuf2_common rtw89_8852ae snd_hda_intel btusb snd_sof_utils snd_intel_dspcfg rtw89_8852a ax88179_178a snd_intel_sdw_acpi btrtl videodev snd_soc_core rtw89_pci snd_hda_codec btbcm mc snd_compress usbnet edac_mce_amd mii ac97_bus rtw89_core kvm_amd btintel snd_hda_core snd_pcm_dmaengine snd_hwdep snd_rpl_pci_acp6x mac80211 btmtk snd_seq kvm snd_pci_acp6x snd_seq_device libarc4 irqbypass bluetooth snd_pci_acp5x sdhci_pci snd_pcm snd_rn_pci_acp3x pcspkr cfg80211 rapl snd_timer wmi_bmof cqhci joydev ideapad_laptop snd_acp_config sparse_keymap sdhci platform_profile snd_soc_acpi snd rfkill soundcore mmc_core
[16580.782211]  snd_pci_acp3x k10temp i2c_piix4 amd_pmc zram dm_crypt amdgpu bcache drm_ttm_helper ttm nvme iommu_v2 gpu_sched nvme_core drm_buddy drm_display_helper uas ucsi_acpi hid_sensor_hub hid_multitouch crct10dif_pclmul crc32_pclmul crc32c_intel polyval_clmulni usb_storage polyval_generic ccp ghash_clmulni_intel cec typec_ucsi serio_raw nvme_common sp5100_tco typec wmi video i2c_hid_acpi i2c_hid scsi_dh_rdac scsi_dh_emc scsi_dh_alua ip6_tables ip_tables dm_multipath fuse
[16580.782233] CPU: 4 PID: 23188 Comm: kworker/u32:16 Tainted: G           OE      6.0.16-300.fc37.x86_64 #1
[16580.782235] Hardware name: LENOVO 82L5/LNVNB161216, BIOS GSCN34WW 10/12/2022
[16580.782237] Workqueue: events_unbound async_run_entry_fn
[16580.782242] RIP: 0010:ieee80211_reconfig+0x88/0x1530 [mac80211]
[16580.782265] Code: 84 24 8d 05 00 00 00 4c 89 e7 e8 73 ff fb ff 41 89 c0 85 c0 0f 84 e9 02 00 00 48 c7 c7 a8 23 f3 c1 44 89 04 24 e8 d5 97 e8 cf <0f> 0b 44 8b 04 24 eb 35 84 c0 0f 85 62 01 00 00 c6 87 8d 05 00 00
[16580.782266] RSP: 0018:ffffaf7bc3e37d90 EFLAGS: 00010292
[16580.782268] RAX: 000000000000006d RBX: ffff98aa50380518 RCX: 0000000000000000
[16580.782269] RDX: 0000000000000001 RSI: ffffffff927ab312 RDI: 00000000ffffffff
[16580.782270] RBP: ffff98aa503803a0 R08: ffffffff930663a0 R09: ffffaf7bc3e37d28
[16580.782271] R10: 3fffffffffffffff R11: 000000000000006f R12: ffff98aa503808e0
[16580.782271] R13: 0000000000000010 R14: 0000000000000000 R15: ffff98acd09f26e8
[16580.782272] FS:  0000000000000000(0000) GS:ffff98ad1ef00000(0000) knlGS:0000000000000000
[16580.782273] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[16580.782274] CR2: 0000564131d6fcf6 CR3: 0000000375010000 CR4: 0000000000750ee0
[16580.782276] PKRU: 55555554
[16580.782276] Call Trace:
[16580.782279]  <TASK>
[16580.782281]  ? preempt_count_add+0x6a/0xa0
[16580.782285]  wiphy_resume+0x79/0x150 [cfg80211]
[16580.782307]  ? wiphy_suspend+0x2b0/0x2b0 [cfg80211]
[16580.782325]  dpm_run_callback+0x47/0x150
[16580.782330]  device_resume+0xa2/0x1f0
[16580.782332]  async_resume+0x19/0x30
[16580.782334]  async_run_entry_fn+0x2d/0x130
[16580.782336]  process_one_work+0x1c4/0x380
[16580.782340]  worker_thread+0x4d/0x380
[16580.782342]  ? rescuer_thread+0x380/0x380
[16580.782344]  kthread+0xe6/0x110
[16580.782347]  ? kthread_complete_and_exit+0x20/0x20
[16580.782349]  ret_from_fork+0x1f/0x30
[16580.782353]  </TASK>
[16580.782353] ---[ end trace 0000000000000000 ]---
[16580.782389] ------------[ cut here ]------------
[16580.782390] wlo1: Failed check-sdata-in-driver check, flags: 0x0
[16580.782398] WARNING: CPU: 4 PID: 23188 at net/mac80211/driver-ops.c:98 drv_remove_interface+0x110/0x120 [mac80211]
[16580.782421] Modules linked in: tls rfcomm snd_seq_dummy snd_hrtimer cdc_ether r8152 vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) qrtr bnep sunrpc binfmt_misc vfat fat snd_sof_amd_renoir intel_rapl_msr intel_rapl_common uvcvideo snd_hda_codec_realtek snd_sof_amd_acp snd_sof_pci snd_hda_codec_generic videobuf2_vmalloc videobuf2_memops snd_sof ledtrig_audio snd_hda_codec_hdmi videobuf2_v4l2 videobuf2_common rtw89_8852ae snd_hda_intel btusb snd_sof_utils snd_intel_ [... cut ...]
[16580.782877]  process_one_work+0x1c4/0x380
[16580.782880]  worker_thread+0x4d/0x380
[16580.782882]  ? rescuer_thread+0x380/0x380
[16580.782884]  kthread+0xe6/0x110
[16580.782885]  ? kthread_complete_and_exit+0x20/0x20
[16580.782887]  ret_from_fork+0x1f/0x30
[16580.782891]  </TASK>
[16580.782891] ---[ end trace 0000000000000000 ]---
[16580.782954] ieee80211 phy0: PM: dpm_run_callback(): wiphy_resume+0x0/0x150 [cfg80211] returns -16
[16580.782980] ieee80211 phy0: PM: failed to resume async: error -16