lwfinger/rtw89 #115: Not working after update

Madzahttr, 2022-01-13:
Installed this yesterday on my Thinkpad E15 Gen3 and all worked fine,
Ran updates today including this and the update from Kernel 5.15.13 to 5.16 and the driver as stopped working.
Uninstalled the driver and reinstalled it as per the updated instructions in the readme, and modprobe gives an error:

`modprobe: ERROR: could not insert 'rtw89pci': Exec format error`

this comes up in dmesg after running modprobe:

`rtw89pci: exports duplicate symbol rtw89_pm_ops (owned by rtw89_pci)`

Madzahttr, 2022-01-13:
full rtw related dmesg output:
```
rtw89core: loading out-of-tree module taints kernel.
rtw89core: module verification failed: signature and/or required key missing - tainting kernel
rtw89_pci 0000:03:00.0: enabling device (0000 -> 0003)
RIP: 0010:rtw89_chip_info_setup+0x22/0x100 [rtw89core]
 rtw89_pci_probe+0x9f1/0xba0 [rtw89_pci 0e71b302ed7215162a2aa04f0f97e959e2f49347]
Modules linked in: snd_hda_codec_generic snd_hda_codec_hdmi(+) kvm_amd(+) rtw89_pci(+) snd_hda_intel amdgpu(+) rtw89core(OE) uvcvideo btusb vfat snd_intel_dspcfg kvm fat videobuf2_vmalloc snd_intel_sdw_acpi btrtl videobuf2_memops irqbypass psmouse snd_hda_codec pcspkr btbcm rapl mac80211 r8169 videobuf2_v4l2 btintel snd_hda_core ucsi_acpi(+) snd_pci_acp6x libarc4 videobuf2_common typec_ucsi snd_pci_acp5x thinkpad_acpi snd_hwdep realtek gpu_sched k10temp bluetooth videodev snd_rn_pci_acp3x sp5100_tco snd_pcm typec drm_ttm_helper mdio_devres ledtrig_audio ecdh_generic cfg80211 mc snd_pci_acp3x ttm snd_timer libphy i2c_piix4 roles platform_profile wmi tpm_crb rfkill snd soundcore video tpm_tis pinctrl_amd mac_hid acpi_cpufreq tpm_tis_core i2c_scmi pkcs8_key_parser crypto_user fuse bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 dm_crypt cbc encrypted_keys dm_mod trusted asn1_encoder tee tpm serio_raw atkbd crct10dif_pclmul libps2 crc32_pclmul crc32c_intel
RIP: 0010:rtw89_chip_info_setup+0x22/0x100 [rtw89core]
rtw89pci: exports duplicate symbol rtw89_pm_ops (owned by rtw89_pci)
```

Madzahttr, 2022-01-13:
did a complete reinstall of arch, and wifi now works by default with the 5.16 kernel

closed 2022-01-13 by Madzahttr

lwfinger, 2022-01-13:
Using the driver in the kernel for 5.16 is the correct solution. This repo is a duplicate of that code with the necessary additions so that it will build on older kernels.

I just pushed a fix for your original problem, but that will not matter for you.