lwfinger/rtw89 #116: severe kernel issues

RiedleroD, 2022-01-13:
linux 5.16.arch1-1 now includes the driver for my device (a rtw89 variant) natively 😃 
rtw89 now crashes with a nullptr and fucks everything in the ass ☹️ 

can post logs if needed, but I doubt they'll be helpful lol

lwfinger, 2022-01-14:
If you have another Linux host, then look at setting up netconsole to capture the crash. There is a potential problem with some illegal VHT codes from some APs, but that only generates a warning.

Do a 'git pull' on this repo, make and install it. Does that help?

RiedleroD, 2022-01-17:
sorry, I don't have enough time to continue testing this. My original installation came from the AUR, so it was already the newest version; just to rule that issue out. You can close this issue if you can't reproduce it yourself with the newest linux kernel.

lwfinger, 2022-01-17:
I do not see this problem with the latest kernel from openSUSE Tumbleweed, or with kernels built from the mainline kernel GIT repo. Per your instructions, I am closing the issue.

closed 2022-01-17 by lwfinger