lwfinger/rtw89 #117: Not working after kernel upgrade

yooouxin, 2022-01-15:


lwfinger, 2022-01-15:
To help, more information is needed. If you are using this repo, you are likely getting interference with the driver built into the kernel. What does 'lsmod | grep 89' show?

CNB-Kevin, 2022-01-15:
I have the same issue. I just happened to be doing updates. I'm on Linux 5.16.0-arch1-1 x86_64

lsmod | grep 89
```
rtw89_pci              69632  1
rtw89core            1032192  1 rtw89_pci
mac80211             1232896  2 rtw89_pci,rtw89core
kvm                  1089536  1 kvm_amd
cfg80211             1077248  2 rtw89core,mac80211
```

mrsbh, 2022-01-15:
On 5.16.0-arch1, I got wifi working on kernel provided rtw89_pci, just uninstall the AUR package.

CNB-Kevin, 2022-01-15:
Oh yeah. I just read the readme. 5.16.0-arch1 _should_ have working wifi. Guess I'm on my own to try and fix my issues, this isn't the place to investigate since it's to support older kernels.

I didn't install through the AUR package. I'm thinking in theory if I reinstall Linux package it should copy the new kernel and this driver should get nuked (hence the instruction to do make install again when changing kernels). But I'm probably wrong and I need to manually remove it from... somewhere.

mrsbh, 2022-01-15:
> I didn't install through the AUR package. I'm thinking in theory if I reinstall Linux package it should copy the new kernel and this driver should get nuked (hence the instruction to do make install [...]
The package provides `make uninstall` which should remove the modules.

lwfinger, 2022-01-15:
Yes, 'sudo make uninstall' is the correct way to uninstall this driver. Reinstalling the kernel to get rid of some modules is definite overkill.

closed 2022-01-15 by lwfinger