lwfinger/rtw89 #118: I can't make it work

AbhayRojas, 2022-01-15:
Hi, I followed the tutorial step by step but I can't make it work. My PC is a Lenovo Legion 5 and I'm running Linux MInt

```
bergfalk@pc:~$ sudo modprobe -v rtw89pci
insmod /lib/modules/5.4.0-94-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko 
modprobe: ERROR: could not insert 'rtw89pci': Operation not permitted
```
I hope you can help me, thanks :)

lwfinger, 2022-01-16:
Look at the output of dmesg to see why it was permitted.

kelebek333, 2022-01-16:
> modprobe: ERROR: could not insert 'rtw89pci': Operation not permitted

Did You try to disable secureboot?

lwfinger, 2022-01-16:
@kelebek333: Until he reports the error from dmesg, we do not know if he needed to sign the module, or not. Disabling secureboot is probably not warranted.

closed 2022-12-14 by lwfinger