lwfinger/rtw89 #120: dkms question on ubuntu

white-gecko, 2022-01-16:
The kernel module works, thank you for providing it in this way. Also I'm very happy that the build files for a debian package are included. I was able to build a package with `dpkg-buildpackage -us -uc` (I had to add the source as .tar.gz to the parrent directory as well). However, after installing the package it does not appear to work as dkms, it does not show up when calling `dkms status`. Has anyone, who has successfully installed the package to work with dkms some instuctions for me?

kelebek333, 2022-01-16:
Hi,

If You want, You can install from PPA. 

```
sudo add-apt-repository ppa:kelebek333/kablosuz
sudo apt-get update
sudo apt install rtw89-dkms
```
I update packages in PPA repository periodically.

diplatis, 2022-03-16:
I, too, have the same issue in Debian.

closed 2022-12-14 by lwfinger