lwfinger/rtw89 #121: Facing a lot of `CTRL-EVENT-BEACON-LOSS`

unkemptArc99, 2022-01-16:
Hi, I am facing a lot of `CTRL-EVENT-BEACON-LOSS` using this driver. The WiFi works perfectly on my laptop in Windows, but not on Linux. Here is my system summary -
```
❯ inxi -Fxxxz
System:
  Kernel: 5.13.0-22-generic x86_64 bits: 64 compiler: N/A 
  Desktop: Cinnamon 5.0.7 wm: muffin 5.0.2 dm: LightDM 1.30.0 
  Distro: Linux Mint 20.2 Uma base: Ubuntu 20.04 focal 
Machine:
  Type: Laptop System: LENOVO product: 82JW v: Legion 5 15ACH6 
  serial: <filter> Chassis: type: 10 v: Legion 5 15ACH6 serial: <filter> 
  Mobo: LENOVO model: LNVNB161216 v: SDK0R32862 WIN serial: <filter> 
  UEFI: LENOVO v: HHCN11WW date: 05/13/2021 
Battery:
  ID-1: BAT0 charge: 62.0 Wh condition: 62.0/60.0 Wh (103%) volts: 17.4/15.4 
  model: Celxpert L20C4PC0 type: Li-poly serial: <filter> status: Full 
  cycles: 7 
CPU:
  Topology: 8-Core model: AMD Ryzen 7 5800H with Radeon Graphics bits: 64 
  type: MT MCP arch: Zen 3 L2 cache: 4096 KiB 
  flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm 
  bogomips: 102208 
  Speed: 1256 MHz min/max: 1200/3200 MHz boost: enabled Core speeds (MHz): 
  1: 1168 2: 1294 3: 3845 4: 4436 5: 3662 6: 2143 7: 1613 8: 1198 9: 1832 
  10: 1565 11: 1332 12: 1297 13: 1297 14: 2547 15: 1981 16: 1297 
Graphics:
  Device-1: NVIDIA vendor: Lenovo driver: nvidia v: 495.46 bus ID: 01:00.0 
  chip ID: 10de:25e0 
  Display: x11 server: X.Org 1.20.13 driver: nvidia 
  unloaded: fbdev,modesetting,nouveau,vesa 
  resolution: 1920x1080~60Hz, 1920x1080~165Hz 
  OpenGL: renderer: NVIDIA GeForce RTX 3050 Ti Laptop GPU/PCIe/SSE2 
  v: 4.6.0 NVIDIA 495.46 direct render: Yes 
Audio:
  Device-1: NVIDIA driver: snd_hda_intel v: kernel bus ID: 01:00.1 
  chip ID: 10de:2291 
  Device-2: AMD Raven/Raven2/FireFlight/Renoir Audio Processor 
  vendor: Lenovo driver: N/A bus ID: 06:00.5 chip ID: 1022:15e2 
  Device-3: AMD Family 17h HD Audio vendor: Lenovo driver: snd_hda_intel 
  v: kernel bus ID: 06:00.6 chip ID: 1022:15e3 
  Sound Server: ALSA v: k5.13.0-22-generic 
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet 
  vendor: Lenovo driver: r8169 v: kernel port: 2000 bus ID: 03:00.0 
  chip ID: 10ec:8168 
  IF: eno1 state: down mac: <filter> 
  Device-2: Realtek vendor: Lenovo driver: rtw89_pci v: N/A port: 1000 
  bus ID: 04:00.0 chip ID: 10ec:8852 
  IF: wlp4s0 state: up mac: <filter> 
  IF-ID-1: gpd0 state: down mac: N/A 
Drives:
  Local Storage: total: 953.87 GiB used: 180.35 GiB (18.9%) 
  ID-1: /dev/nvme0n1 vendor: SK Hynix model: HFS512GDE9X084N 
  size: 476.94 GiB speed: 31.6 Gb/s lanes: 4 serial: <filter> rev: 41010C22 
  scheme: GPT 
  ID-2: /dev/nvme1n1 vendor: Team model: TM8FP4001T size: 953.87 GiB 
  speed: 31.6 Gb/s lanes: 4 serial: <filter> rev: ECFM13.4 scheme: GPT 
Partition:
  ID-1: / size: 930.09 GiB used: 180.32 GiB (19.4%) fs: ext4 
  dev: /dev/nvme1n1p4 
Sensors:
  Message: No sensors data was found. Is sensors configured? 
Info:
  Processes: 433 Uptime: 2d 18h 10m Memory: 15.48 GiB used: 8.92 GiB (57.6%) 
  Init: systemd v: 245 runlevel: 5 Compilers: gcc: 9.3.0 alt: 9 Shell: zsh 
  v: 5.8 running in: gnome-terminal inxi: 3.0.38 
```

I am attaching a sample capture of the `CTRL-EVENT-BEACON-LOSS` events. ([temp.txt](https://github.com/lwfinger/rtw89/files/7877637/temp.txt)) The command I used to capture these logs was `journalctl -rb | grep wlp4s0`. As you can see from the file, the frequency of this event is huge. Can you help me with this?

lwfinger, 2022-01-16:
We need to see if anything else happened before these messages. Change the grep part of your capture command to egrep "wlp4s0|rtw89" and post that file.

unkemptArc99, 2022-01-17:
[temp1.txt](https://github.com/lwfinger/rtw89/files/7878695/temp1.txt)
Here is the requested file.

lwfinger, 2022-01-17:
The only problem with journalctl output is that what you see may not come from the same boot. In any case. a little searching suggests that the problem is caused by the wifi card sleeping when the AP does not expect it. Something is funky between the rtw89 and AP handshaking. What is the make and model of that AP?

Try creating (as root) a file named /etc/modprobe.d/50-rtw89.conf and adding a line that says
options rtw89_core disable_ps_mode=y

If that file already exists, add the line to it. That will keep the rtw89 from going to low-power mode and should fix this problem. You will experience decreased battery life.

unkemptArc99, 2022-01-17:
So, my WiFi card (as you might have guessed because I am using this driver :sweat_smile: ) is Realtek 8852AE, and my AP is an xFi Advanced Gateway router (Tech Specs - https://www.xfinity.com/support/articles/broadband-gateways-userguides). 

Also, I have not used the sleep profile that comes with the repository (suspend_rtw89). Would not doing that be causing some problem?

Also, I will try your suggestion and let you know what I find.

lwfinger, 2022-01-17:
You are confusing the sleep of the laptop with the power-saving feature of the wireless connection. The wifi interface (NIC) can request that the connection be put to sleep to save power while the laptop is running. The messages indicate that the NIC thought it made such a request, but the AP did not honor it. Some protocol is wrong. On my APs, I do not see this problem, nor has anyone else reported it. Is the firmware in the router up to date? I found it strange that the link you provided, which gave lots of specifications, did not mention Wifi5 or Wifi6.

unkemptArc99, 2022-01-17:
I am now curious as to why the NIC is trying to set the connection should be put to sleep, when I am actually playing a multiplayer game (which has now become very infuriating). 

I am also very weirded about the fact that Xfinity does not provide any tech specs on the default router that they send with their internet connection. I dug deeper and found out that this router is essentially a ARRIS TG3482G model. I could then find only 2 good links describing its specs - http://en.techinfodepot.shoutwiki.com/wiki/Arris_TG3482ER and https://www.dslreports.com/hardware/ARRIS-TG3482-h4594.

But here is my link connection details -
```
Connected to <redacted> (on wlp4s0)
	SSID: <redacted>
	freq: 5785
	RX: 172174414 bytes (133619 packets)
	TX: 15959901 bytes (51307 packets)
	signal: -41 dBm
	rx bitrate: 866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
	tx bitrate: 866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2

	bss flags:	short-preamble short-slot-time
	dtim period:	3
	beacon int:	100
```
It seems the connection is using 5GHz 802.11a standard. This is an overlap of WiFi 5 and WiFi 6. When I connect my MacBook Air to the AP, it says it is using 802.11ac (WiFi 5). So, I assume mine should also be a WiFi 5 connection. Do you know any way to get the PHY mode or WiFi standard using linux commands?

unkemptArc99, 2022-01-17:
Also, just an update from your workaround to prevent the NIC going into low power mode, attached is the journalctl logs since I made the workaround. There are definitely some `CTRL-EVENT-BEACON-LOSS` events, but in very low frequency (compared to without the workaround).
[temp2.txt](https://github.com/lwfinger/rtw89/files/7884763/temp2.txt)

lwfinger, 2022-01-17:
The connection will be put into power save mode whenever nothing is being transmitted or received. It should not affect your latency.

The connection will use whatever is common to both AP and NIC. The NIC is Wifi6, but my AP is Wifi5, thus my connection uses 5.

This stuff listed below indicated a warning or bug situation:

Jan 17 13:49:06 <redacted> kernel:  rtw89_pci_interrupt_threadfn+0x13c/0x1f0 [rtw89pci]
Jan 17 13:49:06 <redacted> kernel:  rtw89_pci_napi_poll+0x37f/0x520 [rtw89pci]
Jan 17 13:49:06 <redacted> kernel:  rtw89_core_rx+0x628/0x7c0 [rtw89core]
Jan 17 13:49:06 <redacted> kernel:  rtw89_core_rx_to_mac80211+0xaf/0xe0 [rtw89core]

Can you provide that section from the jounalctrl output with no filtering, of from the dmesg output?

unkemptArc99, 2022-01-17:
Here is the journalctl output with no filter -
```
Jan 17 13:49:06 <redacted> kernel: ---[ end trace 10a4eaff710fb1ac ]---
Jan 17 13:49:06 <redacted> kernel:  ret_from_fork+0x22/0x30
Jan 17 13:49:06 <redacted> kernel:  ? set_kthread_struct+0x40/0x40
Jan 17 13:49:06 <redacted> kernel:  kthread+0x12b/0x150
Jan 17 13:49:06 <redacted> kernel:  ? irq_thread_check_affinity+0xf0/0xf0
Jan 17 13:49:06 <redacted> kernel:  ? irq_forced_thread_fn+0x90/0x90
Jan 17 13:49:06 <redacted> kernel:  irq_thread+0xda/0x170
Jan 17 13:49:06 <redacted> kernel:  irq_thread_fn+0x28/0x60
Jan 17 13:49:06 <redacted> kernel:  ? irq_finalize_oneshot.part.0+0xf0/0xf0
Jan 17 13:49:06 <redacted> kernel:  rtw89_pci_interrupt_threadfn+0x13c/0x1f0 [rtw89pci]
Jan 17 13:49:06 <redacted> kernel:  __local_bh_enable_ip+0x50/0x60
Jan 17 13:49:06 <redacted> kernel:  </IRQ>
Jan 17 13:49:06 <redacted> kernel:  do_softirq+0x66/0x80
Jan 17 13:49:06 <redacted> kernel:  __do_softirq+0xe0/0x29b
Jan 17 13:49:06 <redacted> kernel:  net_rx_action+0x23f/0x290
Jan 17 13:49:06 <redacted> kernel:  __napi_poll+0x31/0x140
Jan 17 13:49:06 <redacted> kernel:  rtw89_pci_napi_poll+0x37f/0x520 [rtw89pci]
Jan 17 13:49:06 <redacted> kernel:  rtw89_core_rx+0x628/0x7c0 [rtw89core]
Jan 17 13:49:06 <redacted> kernel:  rtw89_core_rx_to_mac80211+0xaf/0xe0 [rtw89core]
Jan 17 13:49:06 <redacted> kernel:  ieee80211_rx_napi+0x38/0xa0 [mac80211]
Jan 17 13:49:06 <redacted> kernel:  ? __alloc_skb+0x92/0x1d0
Jan 17 13:49:06 <redacted> kernel:  <IRQ>
Jan 17 13:49:06 <redacted> kernel: Call Trace:
Jan 17 13:49:06 <redacted> kernel: PKRU: 55555554
Jan 17 13:49:06 <redacted> kernel: CR2: 00007ffc339b5000 CR3: 000000010e446000 CR4: 0000000000750ee0
Jan 17 13:49:06 <redacted> kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 17 13:49:06 <redacted> kernel: FS:  0000000000000000(0000) GS:ffff9a349e040000(0000) knlGS:0000000000000000
Jan 17 13:49:06 <redacted> kernel: R13: ffff9a3193c05640 R14: 0000000000000000 R15: ffffb905803ccd48
Jan 17 13:49:06 <redacted> kernel: R10: 0000000000000001 R11: 0000000000000001 R12: ffff9a318c83f500
Jan 17 13:49:06 <redacted> kernel: RBP: ffffb905803ccd38 R08: ffff9a349e0589c0 R09: ffffb905803cca90
Jan 17 13:49:06 <redacted> kernel: RDX: 0000000000000027 RSI: 00000000ffffdfff RDI: ffff9a349e0589c8
Jan 17 13:49:06 <redacted> kernel: RAX: 0000000000000000 RBX: ffff9a3193c00820 RCX: 0000000000000027
Jan 17 13:49:06 <redacted> kernel: RSP: 0018:ffffb905803cccb8 EFLAGS: 00010286
Jan 17 13:49:06 <redacted> kernel: Code: 86 68 02 00 00 80 3d 12 e8 08 00 00 75 92 41 0f b6 54 24 4a 0f b6 f1 48 c7 c7 d0 c9 8d c0 c6 05 f9 e7 08 00 01 e8 63 bd f2 e0 <0f> 0b >
Jan 17 13:49:06 <redacted> kernel: RIP: 0010:ieee80211_rx_list+0x221/0x5a0 [mac80211]
Jan 17 13:49:06 <redacted> kernel: Hardware name: LENOVO 82JW/LNVNB161216, BIOS HHCN11WW 05/13/2021
Jan 17 13:49:06 <redacted> kernel: CPU: 9 PID: 809 Comm: irq/91-rtw89pci Tainted: P           O      5.13.0-25-generic #26~20.04.1-Ubuntu
Jan 17 13:49:06 <redacted> kernel:  sparse_keymap cm32181 industrialio mac_hid sch_fq_codel msr parport_pc ppdev lp parport drm ip_tables x_tables autofs4 btrfs blake2b_generi>
Jan 17 13:49:06 <redacted> kernel: Modules linked in: ip6table_filter ip6_tables iptable_filter bpfilter vboxnetadp(O) vboxnetflt(O) vboxdrv(O) ccm rfcomm cmac algif_hash algi>
Jan 17 13:49:06 <redacted> kernel: WARNING: CPU: 9 PID: 809 at net/mac80211/rx.c:4857 ieee80211_rx_list+0x221/0x5a0 [mac80211]
Jan 17 13:49:06 <redacted> kernel: Rate marked as a VHT rate but data is invalid: MCS: 10, NSS: 2
Jan 17 13:49:06 <redacted> kernel: ------------[ cut here ]------------ 
```

And the following is output from dmesg -
```
[   10.603119] wlp4s0: send auth to a8:97:cd:2f:a3:ec (try 1/3)
[   10.603925] wlp4s0: authenticated
[   10.608375] wlp4s0: associate with a8:97:cd:2f:a3:ec (try 1/3)
[   10.610843] wlp4s0: RX AssocResp from a8:97:cd:2f:a3:ec (capab=0x1511 status=0 aid=1)
[   10.720703] wlp4s0: associated
[   10.720749] wlp4s0: Limiting TX power to 30 (30 - 0) dBm as advertised by a8:97:cd:2f:a3:ec
[   10.788366] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready
[   12.087255] vboxdrv: Found 16 processor cores
[   12.104744] vboxdrv: TSC mode is Invariant, tentative frequency 3194040987 Hz
[   12.104751] vboxdrv: Successfully loaded version 6.1.26_Ubuntu r145957 (interface 0x00300000)
[   12.112784] VBoxNetFlt: Successfully started.
[   12.118162] VBoxNetAdp: Successfully started.
[   12.137189] bpfilter: Loaded bpfilter_umh pid 1498
[   12.137370] Started bpfilter
[   61.421199] ------------[ cut here ]------------
[   61.421202] Rate marked as a VHT rate but data is invalid: MCS: 10, NSS: 2
[   61.421218] WARNING: CPU: 9 PID: 809 at net/mac80211/rx.c:4857 ieee80211_rx_list+0x221/0x5a0 [mac80211]
[   61.421243] Modules linked in: ip6table_filter ip6_tables iptable_filter bpfilter vboxnetadp(O) vboxnetflt(O) vboxdrv(O) ccm rfcomm cmac algif_hash algif_skcipher af_alg bnep nvidia_uvm(PO) nvidia_drm(PO) nvidia_modeset(PO) intel_rapl_msr intel_rapl_common nvidia(PO) rtw89pci(O) rtw89core(O) nls_iso8859_1 uvcvideo edac_mce_amd snd_hda_codec_realtek videobuf2_vmalloc snd_hda_codec_generic btusb videobuf2_memops ledtrig_audio snd_hda_codec_hdmi btrtl videobuf2_v4l2 btbcm btintel snd_seq_midi snd_hda_intel kvm_amd videobuf2_common snd_intel_dspcfg snd_intel_sdw_acpi bluetooth snd_seq_midi_event videodev snd_hda_codec kvm joydev mc ecdh_generic rapl ecc hid_multitouch mac80211 input_leds snd_rawmidi drm_kms_he [... cut ...]
[   61.421370]  __local_bh_enable_ip+0x50/0x60
[   61.421371]  rtw89_pci_interrupt_threadfn+0x13c/0x1f0 [rtw89pci]
[   61.421373]  ? irq_finalize_oneshot.part.0+0xf0/0xf0
[   61.421375]  irq_thread_fn+0x28/0x60
[   61.421376]  irq_thread+0xda/0x170
[   61.421377]  ? irq_forced_thread_fn+0x90/0x90
[   61.421378]  ? irq_thread_check_affinity+0xf0/0xf0
[   61.421379]  kthread+0x12b/0x150
[   61.421381]  ? set_kthread_struct+0x40/0x40
[   61.421382]  ret_from_fork+0x22/0x30
[   61.421385] ---[ end trace 10a4eaff710fb1ac ]---
```

ShaunTheQuietGamer, 2022-01-17:
I'm having the same symptoms on linux-zen 5.16 (so I would think its an issue with the actual driver?) I also have the exact same AP  `(Xfinity) ARRIS TG3482G`. [dmesg.txt](https://github.com/lwfinger/rtw89/files/7885185/dmesg.txt) of the warning/bug is attached unfiltered in case its of any use. possibly relevant specs below.

```System:
  Kernel: 5.16.0-zen1-1-zen x86_64 bits: 64 compiler: gcc v: 11.1.0
    Desktop: GNOME 41.3 tk: GTK 3.24.31 wm: gnome-shell dm: GDM
    Distro: Arch Linux
Machine:
  Type: Laptop System: LENOVO product: 82JW v: Legion 5 15ACH6
    serial: <filter> Chassis: type: 10 v: Legion 5 15ACH6 serial: <filter>
  Mobo: LENOVO model: LNVNB161216 v: SDK0R32862 WIN serial: <filter>
    UEFI: LENOVO v: HHCN23WW date: 11/08/2021
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet
    vendor: Lenovo driver: r8169 v: kernel port: 3000 bus-ID: 03:00.0
    chip-ID: 10ec:8168
  IF: eno1 state: down mac: <filter>
  Device-2: Realtek RTL8852AE 802.11ax PCIe Wireless Network Adapter
    vendor: Lenovo driver: rtw89_pci v: kernel port: 2000 bus-ID: 04:00.0
    chip-ID: 10ec:8852
  IF: wlp4s0 state: up mac: <filter>```

unkemptArc99, 2022-01-18:
@ShaunTheQuietGamer Actually since you are on 5.16 kernel, have you tried the in-kernel rtw89 driver (https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/net/wireless/realtek/rtw89?h=v5.16)? I have been thinking of moving to the same, but I don't want to move too soon on 5.16 as it is just released.

Also, I haven't seen the full wifi card device name being populated in my summary. Have you configured it manually? Here is my Network summary for comparison -
```
System:    Host: abhishek-Legion-5-15ACH6 Kernel: 5.13.0-25-generic x86_64 bits: 64 Desktop: Cinnamon 5.2.7 
           Distro: Linux Mint 20.3 Una 
Machine:   Type: Laptop System: LENOVO product: 82JW v: Legion 5 15ACH6 serial: <superuser/root required> 
           Mobo: LENOVO model: LNVNB161216 v: SDK0R32862 WIN serial: <superuser/root required> UEFI: LENOVO v: HHCN11WW 
           date: 05/13/2021 
Network:   Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet driver: r8169 
           IF: eno1 state: down mac: 90:2e:16:40:06:c2 
           Device-2: Realtek driver: rtw89_pci 
           IF: wlp4s0 state: up mac: 14:5a:fc:0e:2c:63 
           IF-ID-1: gpd0 state: down mac: N/A 
```

ShaunTheQuietGamer, 2022-01-18:
@unkemptArc99 Thats what I am currently using, with the same issues you are describing. Thats why i said i thnink its an issue with the driver itself; its something that isn't just contained to this repo. I imagine that's also why my device name was automatically populated.

lwfinger, 2022-01-18:
The issue has been fixed in the driver. What remains is an error in the kernel compounded by unexpected behavior of some APs. For more description, see the following:
https://marc.info/?l=linux-wireless&m=164117374213027&w=2
https://marc.info/?l=linux-wireless&m=164117374013022&w=2
https://marc.info/?l=linux-wireless&m=164117374313030&w=2

Patches 2 and 3 are already in this repo, but not in the kernel. Patch 1 must be applied to the kernel. Until it propagates through the kernel update process, you will need to install the source code, patch the necessary file, and generate your own custom kernel. One user found that using 2.4 G band avoided the problem.

unkemptArc99, 2022-01-18:
@lwfinger Thank you for the investigation :) Do you know which kernel (or kernels) are being targeted for these patches? Apologies if the question seems too naive (I am a bit noob in kernel development). Basically, I want to confirm whether the patches would roll down to 5.13 or directly go into 5.16.

lwfinger, 2022-01-18:
These patches will go into 5.17, at the earliest. I think it unlikely that they would be propagated as far down as 5.13.

lwfinger, 2022-01-19:
I missed one of the patches posted that fix the problem. I have added that patch and pushed the update along with a fix for builds with kernel 5.17.

If you have this problem, then you must use this repo as it it ahead of the 5.16 kernel.

unkemptArc99, 2022-01-19:
Thanks a lot! If my understanding is correct, the fixes that needed to go into kernel are still pending, right?

lwfinger, 2022-01-19:
Yes, but the one I just added should take care of the problem. Until the kernel catches up, you need to use this repo.

unkemptArc99, 2022-01-19:
Awesome! I will update my driver and let you know about the results :)

ShaunTheQuietGamer, 2022-01-20:
Dumb question - Is it possible to use this repo with a 5.16 kernel that has the un-patched driver already integrated?

lwfinger, 2022-01-20:
Yes, but the driver in 5.16 is not patched to fix the beacon loss  problem. Only the one in this repo has this fix. You need to blacklist rtw89_pci to keep the built-in driver from loading.

lwfinger, 2022-01-20:
If you have had the BEACON-LOSS or the invalid MCS-10 errors, and the driver as updated here fixes the problem, please run the following command:

sudo iw dev wlXXX scan | egrep "SSID|on "

Change the wlXXX to the device name for your wifi card. It may be wlp4s0 or wl01, etc. In the output, find the SSID for the AP you are using. Two lines up, you will find a line that starts with "BSS" followed by some hexadecimal numbers separated by colons. Please post the first 3 of those numbers. These indicate the manufacturer of your AP, and posting them does not have any effect on your security or privacy. We are collecting that information to see if we can add a quirk that will avoid MCS-10 and MCS-11 for affected APs without having to eliminate those speeds for APs that are coded correctly. Thanks.

ShaunTheQuietGamer, 2022-01-21:
I believe `70:54:25` is what you are looking for. So far all of the complains I've seen are from people using the ARRIS TG3482G that Xfinity leases out to residential customers. (It seems there's two slightly different AP with the same model number, TG3482G. There is the stock ARRIS one, and the Xfinity one.)

ShaunTheQuietGamer, 2022-01-21:
@unkemptArc99 out of curiosity, what hardware revision is your router? You can find it in Gateway > Hardware > System Hardware on the AP's management page, (in case you don't know what I'm talking about, go to http://10.0.0.1/, if i remember correctly the default username *and* password is 'admin'.) Mine is revision 15. I'm wondering if its the model in general, or if its a specific revision.

lwfinger, 2022-01-21:
@unkemptArc99 - That vendor code 70:54:25 is not in the database. I have at least one other instance where the vendor was not ARRIS.

@ShaunTheQuietGamer - It is certainly possible that the vendor has fixed this problem.

ShaunTheQuietGamer, 2022-01-21:
>  That vendor code 70:54:25 is not in the database. I have at least one other instance where the vendor was not ARRIS.

That was me that sent that. I thought that I grabbed the right numbers...

lwfinger, 2022-01-21:
I checked a second database that showed the vendor to be Arris Group, Inc. as you reported. I now have 3 vendor codes for APs that show the problem.

riquezjp, 2022-03-08:
I have `wpa_supplicant[686]: wlp4s0: CTRL-EVENT-BEACON-LOSS` in my logs roughly twice a minute & all the way since v4 of the driver when I first got this laptop. (Lenovo ThinkBook 14 G2 ARE)

I did report it, but back then I was more concerned about the[ log flooding with SIGNAL-CHANGE](https://github.com/lwfinger/rtw89/issues/14) rather than BEACON-LOSS, so that detail wasnt mentioned directly.

Since then I have used various routers & AP's as I have moved country. & still get this msg in logs. I began with an asus AC68U router, then various hotels & friends houses & now using mobile phone sharing. The Beacon-loss msg follows me everywhere! (although It doesnt seem to affect my connection, afik)

I was on kernel 5.11, then 5.13 using rtw89 v7, now on 5.16

```
 Device-2: Realtek RTL8852AE 802.11ax PCIe Wireless Network Adapter
    vendor: Lenovo driver: rtw89_pci v: kernel pcie: gen: 1 speed: 2.5 GT/s
    lanes: 1 port: 2000 bus-ID: 04:00.0 chip-ID: 10ec:8852 class-ID: 0280
  IF: wlp4s0 state: up mac: <filter>

```

lissyx, 2022-03-30:
> I checked a second database that showed the vendor to be Arris Group, Inc. as you reported. I now have 3 vendor codes for APs that show the problem.

I'm facing that as well since I upgrade to Ubuntu 22.04, kernel 5.15, and updated to latest current main of this current repo. I'm using Ubiquiti Unifi APs (nanoHD as well as UAP-AC-PRO).

apollo13, 2022-05-16:
I also see `wlp3s0: CTRL-EVENT-BEACON-LOSS` every second in the journal.

I am on Fedora 36 with a kernel from rawhide:
```
Linux apollo13 5.18.0-0.rc6.47.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Mon May 9 14:00:37 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

```
Connected to ab:cd:ef:01:23:45 (on wlp3s0)
	SSID: XXX
	freq: 5220
	RX: 18447953 bytes (32993 packets)
	TX: 2496165 bytes (10410 packets)
	signal: -35 dBm
	rx bitrate: 400.0 MBit/s VHT-MCS 9 40MHz short GI VHT-NSS 2
	tx bitrate: 400.0 MBit/s VHT-MCS 9 40MHz short GI VHT-NSS 2

	bss flags:	short-slot-time
	dtim period:	3
	beacon int:	100
```

Like @lissyx I have Unifi AC-PRO access points; their BSS as of now: `b4:fb:e4` and `f0:9f:c2`

apollo13, 2022-05-16:
Hi @lwfinger, thank you it seems as if disable_aspm_l1=y did help. Since I have no idea what those do, does it make to just keep this one on or should I add the others as well?

lwfinger, 2022-05-16:
If you unload the modules using 'sudo modprobe -rv rtw_8852ae', and reload with 'sudo modprobe -v rtw_8852ae', you should see the options on the rtw89pci line.

mikaelhg, 2022-08-20:
My workaround for the issue is to listen to journal messages, and reassociate when receiving continuous `CTRL-EVENT-BEACON-LOSS` errors:

```python
#!/bin/env python

# journalctl -u wpa_supplicant -f -o json | python event_listener.py

import json
import sys
import subprocess

__BEACON_LOSS = 'CTRL-EVENT-BEACON-LOSS'
__REASSOCIATE = ['/usr/bin/sudo', '/usr/sbin/wpa_cli', 'reassociate']
__WINDOW = 10 * 1_000_000
__THRESHOLD = 5

def main():
    messages = []
    for line in sys.stdin:
        msg = json.loads(line)
        print(msg['SYSLOG_TIMESTAMP'], msg['MESSAGE'])
        messages.insert(0, msg)
        if len(messages) > 100:
            messages.pop()
        t0 = int(msg['__REALTIME_TIMESTAMP'])
        tc = t0 - __WINDOW
        if __BEACON_LOSS in msg['MESSAGE']:
            error_count = 1
            for h in messages:
                if __BEACON_LOSS in h['MESSAGE']:
                    error_count += 1
                if int(h['__REALTIME_TIMESTAMP']) < tc:
                    break
            if error_count > __THRESHOLD:
                print(f'more than 5 in 10, reassociating...')
                cp = subprocess.run(__REASSOCIATE)
                print(f'Reassociate succeeded: {cp.returncode == 0}')

if __name__ == '__main__':
    main()

```

riquezjp, 2022-09-15:
Since moving house (& ISP =  new router) I started getting BEACON-LOSS spamming the log 1/second.
Im using kernel provided driver in 5.18 / 5.19 & followed all the 'options' advice carefully, confirming they were correct when reloading the modules. 
The flood was temporarily halted for a few mins until `kernel: rtw89_8852ae 0000:04:00.0: c2h class 1 func 2 not support` then it would flood.
However, shifting to 2.4Ghz does seem to have now stopped the BEACON-LOSS flood (with no options set.)

Still there is a lot of grumbling in the journal related to wifi, particularly 'func 2 not support' which is now the new log spam, but far less.

The router, for ref, is provided by ISP TalkTalk in the UK ~ BSS 64:fd:96

lwfinger, 2022-09-15:
The 5G signal must have been low. It does not have the range of 2.4 G. An iw scan would show that.

That "func 2..." message has been reported upstream. I thought Realtek had fixed it, but the latest source still does not contain a fix.

riquezjp, 2022-09-15:
> The 5G signal must have been low. It does not have the range of 2.4 G. An iw scan would show that.
Actually its good -41db on 2.4G & -56db on 5G the router is in the same room only 2 metres away.

lwfinger, 2022-09-16:
In my environment, I get the following:

AP       2.4G       5G
#1        -42        -52
#2        -42        -50
#3        -50        -63

#1 and #3 are the same distance away (~2m horizontally), and #2 is about 1m vertically. I never connect to #3.

graytron, 2022-09-22:
I'm running Debian 11 (bullseye) with Linux 5.19 kernel on a Lenovo IdeaPad 5 Pro 16ACH6 laptop connected to an old, but rather reliable Buffalo WZR-HP-AG300H router (`4c:e6:76`), which itself runs OpenWrt 21.02.3. I get regular CTRL-EVENT-BEACON-LOSS events with the rtw89_8852ae driver and WiFi connection "hangs" every now and then.

Did anyone figure out how to easily fix this issue, preferably without scripts and/or recompilations?

lwfinger, 2022-09-23:
One thing you might try is to set an option for rtw89_core. Using your favorite text editor and running as root, create the file /etc/modprobe.d/60-rtw89core.conf. In this file, add the line "options rtw89_core disable_ps_mode=y". Of course, do not include the ", only the material between them.

closed 2022-12-14 by lwfinger