lwfinger/rtw89 #125: extreme rubberbanding/stuttering happens in many games

Zaechus, 2022-01-24:
I narrowed the problem of stuttering happening in online games to my 8852AE wifi card. Everything works completely fine and lag free with an alternate USB wifi device I tested.

I'm not sure yet if there's a work around, if the card is just really bad, or if there's a problem with the driver itself.

Zaechus, 2022-01-24:
Also, if I'm on a really slow network and push the card too hard, dmesg gets flooded with errors (I'll paste them when I get them again) and the card stops working and requires a full system reboot. This same error is sometimes shown in small doses after waking from sleep or just randomly.

Zaechus, 2022-01-24:
Naturally, once I post the issue I figure out how to fix it. It seems that removing "Enable powersave by default" (CONFIG_CFG80211_DEFAULT_PS) in the kernel solves this problem. Of course, this shows that a problem with the card/driver doesn't let it play well with a sane option in the kernel.

I will keep this issue open unless someone else deems this a non-issue.

lwfinger, 2022-01-24:
Making that change says that the driver and your AP are not working well together. You could have gotten to the same place with the "disable_ps_mode=y" for rtw89core or rtw89_core. The latter is for the in-kernel driver.

Zaechus, 2022-02-13:
On kernel version 5.15.23 (still happened with older versions of 5.15 anyways), the driver will randomly crash with the following in `journalctl`:
```
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: queue 0 txwd 2 is not idle
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: queue 0 txwd 506 is not idle
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: queue 0 txwd 507 is not idle
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: FW status = 0xaa001108
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: FW BADADDR = 0x1860ca0c
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: FW EPC/RA = 0x0
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: FW MISC = 0xb89911c7
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: R_AX_HALT_C2H = 0x10
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: R_AX_SER_DBG_INFO = 0x3200000e
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89920fd
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb899210f
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb899201b
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb899200b
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8992107
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8992027
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8992113
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89920ed
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8992009
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb899210f
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb899202f
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8992003
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb899210b
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb899202b
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8992001
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: --->
                                  err=0x10
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: R_AX_SER_DBG_INFO =0x3200000e
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: R_AX_CMAC_ERR_ISR =0x00000000
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: R_AX_DMAC_ERR_ISR =0x00000000
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: R_AX_RPQ_RXBD_IDX =0x00000000
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: R_AX_DBG_ERR_FLAG=0x00000000
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: R_AX_LBC_WATCHDOG=0x00000041
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: <---
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: ser event = 0x0010
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: pci dmach busy1 0x200100
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: pci dmach busy1 0x200100
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: lv1 rcvy pci stop dma fail
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: sec cam entry is empty
Feb 13 14:18:50 gentoopad kernel: ieee80211 phy0: Hardware restart was requested
Feb 13 14:18:50 gentoopad kernel: rtw89_pci 0000:03:00.0: MAC has already powered on
Feb 13 14:18:51 gentoopad kernel: rtw89_pci 0000:03:00.0: failed to get available bssid cam
Feb 13 14:18:51 gentoopad kernel: rtw89_pci 0000:03:00.0: failed to init bssid cam
Feb 13 14:18:52 gentoopad kernel: rtw89_pci 0000:03:00.0: failed to get available bssid cam
```
The last three messages repeat constantly until a manual shutdown and reboot is enforced.

This happens on different wireless networks with different routers and ISPs. This latest crash was on powersave mode; I'll do more study on whether or not it crashes when kept in performance mode the whole time. This is difficult since this issue only cares maybe one time in a whole week.

Zaechus, 2022-02-18:
Error occurred again, retesting with all of the "disable_" options for modprobe.

Zaechus, 2022-03-01:
swapping the 8852ae for an ax200 resolved the issue :+1:

closed 2022-03-01 by Zaechus