lwfinger/rtw89 #126: Partial hung up almost daily, syslog filled with repeated logs

Elktro, 2022-01-26:
I'm running rtw89 driver for the Realtek Semiconductor Co., Ltd. Device 8852 for wifi and bluetooth on my Yoga Slim 7 Pro (with 5.13.0-27-generic x86_64). I, however, experience almost daily partial hung ups of my laptop due to the driver. (I can operate some of the user space programs, but need to cold boot the computer. `sudo shutdown now` does not work. It just hangs)

Logs are filled with repeating messages:

syslog excerpt (see attachment for detail):
Jan 24 15:52:34 computer kernel: [30504.114207] rtw89_pci 0000:02:00.0: failed to get available bssid cam
Jan 24 15:52:34 computer kernel: [30504.114218] rtw89_pci 0000:02:00.0: failed to init bssid cam
Jan 24 15:52:34 computer kernel: [30504.226127] rtw89_pci 0000:02:00.0: failed to get available bssid cam
Jan 24 15:52:34 computer kernel: [30504.226133] rtw89_pci 0000:02:00.0: failed to init bssid cam
Jan 24 15:52:34 computer kernel: [30504.338279] rtw89_pci 0000:02:00.0: failed to get available bssid cam
Jan 24 15:52:34 computer kernel: [30504.338299] rtw89_pci 0000:02:00.0: failed to init bssid cam
Jan 24 15:52:34 computer kernel: [30504.450345] rtw89_pci 0000:02:00.0: failed to get available bssid cam
Jan 24 15:52:34 computer kernel: [30504.450356] rtw89_pci 0000:02:00.0: failed to init bssid cam

[inxi.txt](https://github.com/lwfinger/rtw89/files/7944245/inxi.txt)
[modinfo.txt](https://github.com/lwfinger/rtw89/files/7944246/modinfo.txt)
[syslog.txt](https://github.com/lwfinger/rtw89/files/7944247/syslog.txt)

lwfinger, 2022-01-26:
Have you tried any of the module options?

Elktro, 2022-01-26:
Added `options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y` to /etc/modprobe.d/rtw89.conf. let's see if that helps.

Still, the driver should not fill logs with endless messages. Is there an obvious way to fix that?

ralfstx, 2022-02-13:
I'm having the same problem. Lenovo P14s, Ubuntu 21.10.

lwfinger, 2022-02-13:
Did you add the options? These messages usually appear in Lenovo laptops due to an error-filled BIOS that prevents incorrect operation of the PCIe.

For those of us with a correct BIOS, we get no spamming of the logs. Of course, the driver could be changed to limit the number of such outputs. If you wish this, prepare a patch and submit it to linux-wireless@vger.kernel.org.

closed 2022-02-13 by lwfinger

Elktro, 2022-02-13:
After adding the options, no more hung ups observed.

lissyx, 2022-04-14:
> After adding the options, no more hung ups observed.

On my laptop, ThinkPad P14s Gen 2 AMD, then even with those options no help at all. In the past it would happen only once in a while but since recently (a few days ?), then it would basically happen after a few minutes.

I'm running with UEFI 1.15 (which seems to have been pulled off by Lenovo, is this related?).

lwfinger, 2022-04-14:
There is one additional option to try "disable_clkreq=y".

This is most certainly caused by Lenovo, either in the BIOS or perhaps UEFI 1.15.

I have owned one Lenovo laptop in my career, and would never buy another!!!!

lissyx, 2022-04-14:
> There is one additional option to try "disable_clkreq=y".

Probably, although, no offense to you or the Realtek people working on that, but I've lost way too much time on workarounds for that driver, this week I spent half of my work time rebooting because of it, so I'll switch to an ax210 card as soon as it arrives, and I'll wait for driver and UEFI to mature a few months more.

> 
> This is most certainly caused by Lenovo, either in the BIOS or perhaps UEFI 1.15.

Yes, I'm in touch with some people from Lenovo on another issue (WWAN firmware crashing when I have "more than ping" usage on linux, and it would depend on the environment because I dont hit the crash when I'm not home), I will probably pass it over.

> 
> I have owned one Lenovo laptop in my career, and would never buy another!!!!

I have owned many ThinkPad, and they always were super stable. But only Intel, since in the past there was no AMD option.

lwfinger, 2022-04-14:
Do what you have to do. My 8852ae has not burped in the past 3 months.

The ax210 is a good card, but be sure you are running at least a 5.16 kernel, 5.17 would be better. Intel has had some problems of late. I cannot say that they affected the ax210 as I have an Intel Corporation Wireless 7260, which was not affected.

lissyx, 2022-04-14:
> Do what you have to do. My 8852ae has not burped in the past 3 months.

But yours is not in a ThinkPad :)

As I said, I meant no offense, your work of packaging and updating the driver has helped me a lot, and it is only recently that I have been facing many problems, so I can only be thankful. I just wanted to write it down somewhere that maybe, with that setup (P14s Gen 2 AMD, UEFI 1.15), there is still something going on.

I'll keep an eye on the status of the driver and retry in the future :)