lwfinger/rtw89 #127: needs to unload and reload kernel modules all the times

pakyrs, 2022-01-31:
On vanilla Arch, KDE, nstalled the package via AUR and needs to unload and reload the kernel module on every startup, along with refreshing NetworkManager to show the available WiFi's any idea why?

I have been using these drivers for a fair while and always worked like charm.

I could script the reload of the modules upon reboot but would like to understand why this is happening.

Thanks

lwfinger, 2022-01-31:
I have no idea. That behavior does not happen here. Are these warm reboots, or is the power turned completely off. If Wake On Lan is enabled in your BIOS, that would prevent power off on shutdown.

My rtw89 chip is mounted on en Express Card extender, and I clearly see the power LED blink off for about 1 sec before the machine starts to reboot.

pakyrs, 2022-02-01:
It started to happen with the new 5.16.4 zen kernel, wasn't an issue before.

No wake on lan option, either a full shutdown and start or reboot resolves in the same issue. Not much info in dmesg

I am wondering if it could be because of AUR package now.

Didn't have the issue by compiling myself before.

lwfinger, 2022-02-01:
I would guess that the zen kernel has more effect than the AUR package. You can test that by building and installing this repo, which will remove the AUR part.

I do not know enough about zen to know if drivers are loaded in the hypervisor, or in the overlaid kernel.

closed 2022-09-24 by lwfinger