lwfinger/rtw89 #128: modprobe: ERROR: could not insert 'rtw89pci': Exec format error

Beagle, 2022-02-02:
Building from commit hash: `72c62621c207eb4e8b68b92522fd104ebc32fa69`

Build procedure:
```
git clone
make
sudo make install
```

All successful.

`sudo modprobe -v rtw89pci`

fails with the follow message:

> insmod /lib/modules/5.13.0-28-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko
> modprobe: ERROR: could not insert 'rtw89pci': Exec format error

lwfinger, 2022-02-03:
Check the dmesg output to see what the problem is.

teomanyuksel, 2022-02-04:
Not OP but I have the same error. dmesg output (last 9 lines):

`[  216.254061] audit: type=1400 audit(1643989564.569:83): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/discord/discord.desktop" pid=2388 comm="pool-org.gnome." requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[  216.254069] audit: type=1400 audit(1643989564.569:84): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/var/lib/snapd/hostfs/usr/share/gdm/greeter/applications/gnome-initial-setup.desktop" pid=2388 comm="pool-org.gnome." requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[  216.633367] audit: type=1326 audit(1643989564.949:85): auid=1000 uid=1000 gid=1000 ses=3 subj=snap.snap-store.ubuntu-software pid=2388 comm="pool-org.gnome." exe="/snap/snap-store/558/usr/bin/snap-store" sig=0 arch=c000003e syscall=93 compat=0 ip=0x7f32021544fb code=0x50000
[  273.862989] rtw89core: exports duplicate symbol __rtw89_debug (owned by rtw89_core)
[  369.706156] rtw89core: exports duplicate symbol __rtw89_debug (owned by rtw89_core)`

lwfinger, 2022-02-04:
@teomanyuksel: I do not know what kernel you are running, but I suspect you have tried to install drivers from this repo, but you have drivers from the kernel available as well.

What does 'lspci | grep 89' show?

Beagle, 2022-02-06:
Output of dmesg:
`[147280.307071] rtw89core: exports duplicate symbol __rtw89_debug (owned by rtw89_core)`

`lspci|grep 89` returns nothing

`lsmod|grep 89` returns:
```
rtw89_pci              49152  0
rtw89_core            925696  1 rtw89_pci
mac80211             1028096  2 rtw89_core,rtw89_pci
cfg80211              888832  2 rtw89_core,mac80211
```

lwfinger, 2022-02-06:
Those modules ar the ones from the kernel. If you want to use the drivers in this repo, you will need to blacklist rtw89_pci. This driver is named rtw89pci.

Beagle, 2022-02-06:
I wasn't aware that 5.13 had drivers for rtw89. I thought it was only introduced in 5.16. Would you recommend using the modules included in my kernel or continue to use the ones from this repository?

gati3478, 2022-02-06:
Happened the same to me on Fedora 35 kernel version 5.15.18-200
I didn't have WiFi available at all after the update to this kernel version

lwfinger, 2022-02-06:
The drivers in this repo have been modified to keep up with the wireless-drivers-next repo. At the moment, these are equivalent with what will be in kernel 5.18. If the in-kernel version works for you, use it. If you have problems, switch to this repo. If you still have problems, then we will try to solve them.

lwfinger, 2022-02-06:
@gati3478: After any kernel change, ALL out of kernel modules must be rebuilt and reinstalled. No exceptions.

scottbarnesg, 2022-02-07:
@lwfinger blacklisting the new driver for 5.16 fixed the issue for me. It looks like a version of the driver was included with this kernel, but its clearly not working for all hardware (I couldn't get wifi working with it).

For those who want to blacklist `rtw89_pci`, the following worked for me on Fedora 35:

Create a file in `/etc/modprobe.d/` containing:

```
blacklist rtw89_pci
```

For reference, I saw this issue on the following system:

Device: ThinkPad T14 Gen2 (AMD)
OS: Fedora 35
Kernel Linux 5.16.5-200.fc35.x86_64 x86_64

Thank you for looking into this (and all the work you put into this project).

lwfinger, 2022-02-07:
No problem. Notice how I did not tell you how to blacklist a driver? You remember better when you solve the problem.

closed 2022-02-07 by lwfinger