lwfinger/rtw89 #129: i cant do "make" in rtw89

darkmurcie, 2022-02-07:
Hi!, i am trying in my lenovo legion 5 and in Manjaro, I did the first things step by step but when i have to execute "make" in rtw89, the konsole send me this error: 
                                                                            ✔ 
make -C /lib/modules/5.16.2-1-rt19-MANJARO/build M=/home/panamiguel/rtw89 modules
make[1]: *** /lib/modules/5.16.2-1-rt19-MANJARO/build: No such file or directory.  Stop.
make: *** [Makefile:58: all] Error 2

i am noob in linux, i hope u guys can help me, thx!

lwfinger, 2022-02-07:
You have not installed the kernel headers. Read the "### Installation instruction" instructions in README.md.

darkmurcie, 2022-02-07:
> You have not installed the kernel headers. Read the "### Installation instruction" instructions in README.md.

I did it, maybe wrong. 
i wrote "sudo pacman -S linux-headers"
and then it give me those options
::: Repository core
   1) linux414-headers  2) linux419-headers  3) linux44-headers  4) linux49-headers  5) linux510-headers
   6) linux515-headers  7) linux516-headers  8) linux517-headers  9) linux54-headers
:: Repository community
   10) linux510-rt-headers  11) linux515-rt-headers  12) linux516-rt-headers

i think i choose number 6, because my kernel was 5.15.19-1, maybe i choose the wrong one. later, because the error, i uptate to 5.16.2_rt19-1 and detele de headers of 515 and install de option 7, maybe wrong again.

can u tell me how to do it well? thank you, im very noob and im not sure of anything y do haha, i dont mind if i have to reinstall manjaro or something, I only wanna fix this :c 

Pd. sorry for my english haha

darkmurcie, 2022-02-07:
> Your headers MUST match the kernel you are running. Why they should offer you headers as old as 4.14 is beyond me.
> 
> I do not suggest that you run an -rt kernel unless you know exactly what you are doing, and your workload requires it!

but with 5.15 u said i havent got the headers, so i downloaded wrong one, so can u tell me with 5.15.19-1 what are the correct headers? thx!

lwfinger, 2022-02-07:
Run the command 'uname -r'. The headers MUST match them. In addition, 'ls /lib/modules/$(uname -r)/build' must show a directory named "include".

closed 2022-09-24 by lwfinger