lwfinger/rtw89 #130: Installing Without Internet Connection

micah-williamson, 2022-02-19:
Maybe a stupid question but how are you supposed to install this to fix network drivers when the first steps require installing deps from the internet? Are people downloading `kernel-devel` manually from a different machine and transferring via usb? Those deps have their own deps as well.

lwfinger, 2022-02-20:
Most people connect through a wire to satisfy the requisites. Transferring distro packages via USB is not trivial due to the interdependencies of packages that you noted but are solved by the distro's tool.

If you installed from a DVD, or a USB image of the DVD, you should be able to add that medium to your package tool's list of sources. This becomes the equivalent of adding the required packages at installation.

If your box has a USB port, then I suggest that you acquire a simple USB wifi device. As you will not need great speed, almost any of the Mediatek RTxxxx or Realtek RTL8192CU will work just fine. These should be available on E-Bay for $10 US or less. The drivers are built into Linux, thus they will be plug-and-play. If you have any questions as to whether a particular device will work, post the link here, and I will check it for you.

micah-williamson, 2022-02-20:
Hah of course. Can't believe I didn't consider that. Thank you.

closed 2022-02-20 by micah-williamson