lwfinger/rtw89 #131: Null pointer

rubydesign, 2022-02-23:
I got a new lenovo yoga 14 book, the one with 8-core amd.
And luckily found this and got the wifi working, thanks a lot.

Until my fedora 35 upgraded to 5.16.9-200.fc35.x86_64  (from 16.8)
and wifi stopped working and i get this in dmesg.

I use unsigned module and have secure boot off.

I would be happy to help, but amd out of my depth by a long shot with kenel development. So this is just info, i will wait it out or buy a usb stick. I can handle a terminal if there is instructions:-)

PS: to compile for the updated kernel i did git pull, and the firmware got updated. So i don't know if it is the new kernel or the new firmware.
PPS: i also did not manage to get it working on older kernels (fedora keeps 3)
PPPS: there was a warning that the gcc version mismatched (kernel/module) in the mini-minor. can's think that is a cause, but just in case

[    3.676821] rtw89_pci 0000:01:00.0: enabling device (0000 -> 0003)
[    3.677885] BUG: kernel NULL pointer dereference, address: 0000000000000048
[    3.677888] #PF: supervisor read access in kernel mode
[    3.677889] #PF: error_code(0x0000) - not-present page
[    3.677890] PGD 0 P4D 0 
[    3.677893] Oops: 0000 [#1] PREEMPT SMP NOPTI
[    3.677895] CPU: 6 PID: 724 Comm: systemd-udevd Tainted: G           O      5.16.9-200.fc35.x86_64 #1
[    3.677897] Hardware name: LENOVO 82MS/LNVNB161216, BIOS GZCN20WW 06/08/2021
[    3.677898] RIP: 0010:rtw89_chip_info_setup+0x22/0x120 [rtw89core]
[    3.677908] Code: 00 00 00 00 0f 1f 40 00 0f 1f 44 00 00 41 54 be f0 00 00 00 41 bc 00 f0 00 00 55 48 8b 87 40 01 00 00 48 89 fd f3 4d 0f bc e4 <48> 8b 40 48 0f ae e8 ff d0 44 89 e1 25 00 f0 00 00 d3 e8 89 c2 3c
[    3.677910] RSP: 0018:ffffad92415bfc00 EFLAGS: 00010202
[    3.677911] RAX: 0000000000000000 RBX: 00000000000000ff RCX: 0000000000000000
[    3.677912] RDX: ffff99ee653ba7f8 RSI: 00000000000000f0 RDI: ffff99ee67402040
[    3.677913] RBP: ffff99ee67402040 R08: 0000000000000000 R09: 0000000000000a20
[    3.677914] R10: 0000000000000003 R11: 00000001ffffe000 R12: 000000000000000c
[    3.677915] R13: ffff99eee2f7c040 R14: ffff99ee67402040 R15: ffff99ee41ae90d0
[    3.677916] FS:  00007ffa920a8b40(0000) GS:ffff99f0dff80000(0000) knlGS:0000000000000000
[    3.677917] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    3.677918] CR2: 0000000000000048 CR3: 0000000104ba8000 CR4: 0000000000750ee0
[    3.677920] PKRU: 55555554
[    3.677920] Call Trace:
[    3.677922]  <TASK>
[    3.677924]  rtw89_pci_probe+0x9c6/0xb40 [rtw89_pci]
[    3.677928]  local_pci_probe+0x45/0x80
[    3.677932]  ? pci_match_device+0xd7/0x130
[    3.677933]  pci_device_probe+0xbe/0x1a0
[    3.677935]  really_probe+0x1f5/0x3f0
[    3.677937]  __driver_probe_device+0xfe/0x180
[    3.677938]  driver_probe_device+0x1e/0x90
[    3.677939]  __driver_attach+0xc0/0x1c0
[    3.677940]  ? __device_attach_driver+0xe0/0xe0
[    3.677941]  ? __device_attach_driver+0xe0/0xe0
[    3.677942]  bus_for_each_dev+0x64/0x90
[    3.677944]  bus_add_driver+0x149/0x1e0
[    3.677945]  driver_register+0x8f/0xe0
[    3.677947]  ? 0xffffffffc01c1000
[    3.677948]  do_one_initcall+0x44/0x200
[    3.677950]  ? kmem_cache_alloc_trace+0x163/0x2c0
[    3.677953]  do_init_module+0x5c/0x280
[    3.677955]  __do_sys_init_module+0x11d/0x180
[    3.677957]  do_syscall_64+0x3b/0x90
[    3.677959]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[    3.677961] RIP: 0033:0x7ffa92c4292e
[    3.677963] Code: 48 8b 0d fd 74 0e 00 f7 d8 64

lwfinger, 2022-02-23:
That location for the NULL pointer dereference does not make much sense.

With kernels as new as 5.16.9, you should be using the drivers built into the kernel, not this repo, which exists to handle bugs that have been located, but not yet fixed in the main kernel, or to backport the driver to older kernels. Neither case applies here.

Did you blacklist rtw89_core? If not, there is a potential for both rtw89_pci and rtw89pci to be loaded at the same time, which leads to the kinds of errors seen here. You should probably add rtw89_pci to /etc/modules.d/50-blacklist.conf and reboot 
so that you are ensured of getting only rtw89pci. Once rebooted, run  'sudo make uninstall' is the rtw89 directory, remove the blacklist entry created above, and reboot. That should work. After this reboot, run 'lsmod | brep rtw89'. You should see only rtw89_pci and rtw89_core.

rubydesign, 2022-02-24:
Thanks for answering. And true, the current kernel seems to work without this repo and has the rtw89_core and rtw89_pci loaded.

I think i started with a 14 kernel and then it was not working. I do try to read docs but even now can't see a clear mention of the fact that newer (is it 16+) kernels do not need this, nay will break with it. Maybe worth putting in as a google search for the realtek card will lead here quite quickly (and so the fact that 16 is the magic line after which things just work will go unnoticed)

Still, thanks for the work and the answer,
Torsten

closed 2022-02-24 by rubydesign

lwfinger, 2022-02-24:
The in-kernel driver only appeared with 5.16. For kernels older than that, you need this repo, unless your distro has backported the driver the way that openSUSE has.

lwfinger, 2022-02-24:
By the way, the first sentence of README.md states "This branch was created from the version merged into the wireless-drivers-next repo, which is in the 5.16 kernel."

rubydesign, 2022-02-24:
Yes that is what i meant: It in no way says clearly: Don't use this for kernels over 5.16

There is lots of CAPS, like YOU ARE ON YOUR OWN!!!!!
So those messages are clear and say what you mean

lwfinger, 2022-02-24:
OK, I yelled some more.