lwfinger/rtw89 #133: unstable connection

colttt, 2022-03-18:
I've no stable internet connection (a simple ping leads to >50% data loss, it looks like after every 4-5th ping there are 5missing pings)
```
	ping -c 60 8.8.8.8 -I wlp3s0
	PING 8.8.8.8 (8.8.8.8) from 192.168.178.56 wlp3s0: 56(84) bytes of data.
	64 bytes from 8.8.8.8: icmp_seq=2 ttl=118 time=17.7 ms
	64 bytes from 8.8.8.8: icmp_seq=3 ttl=118 time=16.1 ms
	64 bytes from 8.8.8.8: icmp_seq=4 ttl=118 time=16.0 ms
	64 bytes from 8.8.8.8: icmp_seq=5 ttl=118 time=16.5 ms
	64 bytes from 8.8.8.8: icmp_seq=13 ttl=118 time=1045 ms
	64 bytes from 8.8.8.8: icmp_seq=14 ttl=118 time=21.8 ms
	64 bytes from 8.8.8.8: icmp_seq=15 ttl=118 time=16.6 ms
	64 bytes from 8.8.8.8: icmp_seq=16 ttl=118 time=14.8 ms
	64 bytes from 8.8.8.8: icmp_seq=17 ttl=118 time=16.7 ms
	64 bytes from 8.8.8.8: icmp_seq=26 ttl=118 time=1045 ms
	64 bytes from 8.8.8.8: icmp_seq=27 ttl=118 time=20.7 ms
	64 bytes from 8.8.8.8: icmp_seq=28 ttl=118 time=15.4 ms
	64 bytes from 8.8.8.8: icmp_seq=29 ttl=118 time=16.3 ms
	64 bytes from 8.8.8.8: icmp_seq=34 ttl=118 time=1043 ms
	64 bytes from 8.8.8.8: icmp_seq=35 ttl=118 time=19.6 ms
	64 bytes from 8.8.8.8: icmp_seq=36 ttl=118 time=15.7 ms
	64 bytes from 8.8.8.8: icmp_seq=44 ttl=118 time=20.8 ms
	64 bytes from 8.8.8.8: icmp_seq=45 ttl=118 time=15.6 ms
	64 bytes from 8.8.8.8: icmp_seq=46 ttl=118 time=16.7 ms

	--- 8.8.8.8 ping statistics ---
	60 packets transmitted, 19 received, 68.3333% packet loss, time 60017ms
	rtt min/avg/max/mdev = 14.834/179.499/1045.424/374.572 ms, pipe 3
```

Debian Release: 11.2

Kernel: Linux 5.16.0-0.bpo.3-amd64 (SMP w/16 CPU threads; PREEMPT)

lshw -C Network
```
[...]
  *-network
       description: Wireless interface
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:03:00.0
       logical name: wlp3s0
       version: 00
       serial: e0:0a:f6:69:05:5b
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw89_pci driverversion=5.16.0-0.bpo.3-amd64 firmware=N/A ip=192.168.178.56 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:109 ioport:3000(size=256) memory:fd800000-fd8fffff
[...]
```
modinfo rtw89_pci rtw89_core
```
filename:       /lib/modules/5.16.0-0.bpo.3-amd64/kernel/drivers/net/wireless/realtek/rtw89/rtw89_pci.ko
license:        Dual BSD/GPL
description:    Realtek 802.11ax wireless PCI driver
author:         Realtek Corporation
alias:          pci:v000010ECd0000A85Asv*sd*bc*sc*i*
alias:          pci:v000010ECd00008852sv*sd*bc*sc*i*
depends:        rtw89_core,mac80211
retpoline:      Y
intree:         Y
name:           rtw89_pci
vermagic:       5.16.0-0.bpo.3-amd64 SMP preempt mod_unload modversions
sig_id:         PKCS#7
signer:         Debian Secure Boot CA
sig_key:        4B:6E:F5:AB:CA:66:98:25:17:8E:05:2C:84:66:7C:CB:C0:53:1F:8C
sig_hashalgo:   sha256
signature:      14:B0:E0:87:29:FA:6A:84:2B:1E:A4:F3:5C:C7:D8:66:85:8C:08:4C:
                3F:A7:57:93:3B:2C:0F:D4:EA:C6:D0:9D:92:23:3E:D3:99:EE:BE:6B:
                0B:FB:64:BD:9E:8D:08:64:B0:42:E4:E1:FD:E5:39:96:42:DC:94:C3:
                75:4E:A6:CB:77:4E:7D:19:E1:E3:7A:D2:E2:B2:04:E5:FA:52:65:17:
                39:7D:09:AA:F6:33:AD:7B:96:B7:F2:AE:ED:76:5F:03:00:AC:E6:03:
                17:6B:63:55:55:BE:E9:66:EE:F2:89:C6:0E:8D:C6:28:2C:FD:8E:31:
                91:E6:B4:5A:56:33:BC:3B:12:6B:F4:A1:21:F1:1D:33:C2:FA:5C:F1:
                89:D2:0F:1C:69:52:24:B9:F7:9E:26:98:E2:26:3D:66:3E:DD:22:36:
                DA:15:02:C8:4E:BB:B6:82:97:13:F1:74:18:C4:01:F9:95:77:7D:56:
                E5:73:EF:33:41:6C:C3:2F:C0:7B:8E:4D:27:0D:F5:7C:E9:4B:B2:0D:
                73:CB:10:13:36:49:6B:E9:43:EE:50:A2:AE:2B:52:A0:86:D2:0B:7D:
                92:3B:7E:B4:79:B8:59:35:E2:9E:D1:BD:D3:5A:56:3B:04:0C:B7:CA:
                91:99:83:52:77:17:0D:DA:B4:E9:28:D7:DA:AE:0F:89
parm:           disable_clkreq:Set Y to disable PCI clkreq support (bool)
parm:           disable_aspm_l1:Set Y to disable PCI ASPM L1 support (bool)
parm:           disable_aspm_l1ss:Set Y to disable PCI L1SS support (bool)
filename:       /lib/modules/5.16.0-0.bpo.3-amd64/kernel/drivers/net/wireless/realtek/rtw89/rtw89_core.ko
license:        Dual BSD/GPL
description:    Realtek 802.11ax wireless core module
author:         Realtek Corporation
firmware:       rtw89/rtw8852a_fw.bin
depends:        mac80211,cfg80211
retpoline:      Y
intree:         Y
name:           rtw89_core
vermagic:       5.16.0-0.bpo.3-amd64 SMP preempt mod_unload modversions
sig_id:         PKCS#7
signer:         Debian Secure Boot CA
sig_key:        4B:6E:F5:AB:CA:66:98:25:17:8E:05:2C:84:66:7C:CB:C0:53:1F:8C
sig_hashalgo:   sha256
signature:      37:79:5C:38:36:11:42:5D:E7:5B:6A:49:94:98:88:84:D9:83:AC:27:
                2C:33:27:35:1B:67:78:FF:8E:B5:58:5C:E2:0D:EF:DE:CD:86:7B:4D:
                8A:43:22:D8:AF:BC:B2:96:C6:14:D0:2E:C5:D3:B1:48:78:1F:16:36:
                88:84:67:6C:BA:9B:9B:0A:31:80:C8:38:DB:53:20:B9:3D:15:07:67:
                88:44:5D:8F:25:42:79:54:FF:18:50:E2:A2:C9:DD:72:54:09:BB:30:
                0E:EF:4B:CF:FE:15:3C:C2:3B:8A:95:4C:35:50:6C:B2:34:0B:FC:42:
                9A:B0:73:9F:D3:84:C5:8D:4D:6D:3D:FB:CD:F0:A9:5E:A0:D3:7E:D0:
                30:41:54:34:47:B4:DA:DB:C3:B9:11:9E:48:FD:41:1B:87:E1:8E:1C:
                E2:8D:F8:13:6C:CC:59:BD:91:4F:38:67:E9:E2:A2:17:EE:40:DC:0C:
                D0:7A:95:27:B7:FD:7D:67:47:E3:9F:DA:3E:14:BE:9A:83:71:49:0F:
                F8:6C:43:7C:54:74:1B:24:A9:27:65:FF:53:1F:C3:84:E9:3E:E5:99:
                BB:BD:05:28:F8:E7:2D:2A:CC:3C:E2:5B:91:37:0E:BE:08:23:16:F8:
                1D:2C:B8:D6:E3:17:3D:F5:72:4D:C9:4A:F2:5C:4E:FE
parm:           disable_ps_mode:Set Y to disable low power mode (bool)
```

dmesg:
```
[ 2172.276570] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 2172.377306] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 2172.477665] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 2172.577527] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 2172.677964] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 2172.778385] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 2172.878858] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 2172.979191] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 2173.012134] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 2173.012156] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x94  0x00  0xe0  0x81
[ 2174.965023] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 2174.965045] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x94  0x00  0xe0  0x81
[ 2176.981149] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 2176.981196] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x94  0x00  0xe0  0x81
[ 2178.996667] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 2178.996705] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x94  0x00  0xe0  0x81
[ 2181.012787] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 2181.012808] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x94  0x00  0xe0  0x81
[ 2182.932550] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 2182.932571] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x94  0x00  0xe0  0x81
[ 2185.204745] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 2185.204766] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x94  0x00  0xe0  0x81
```

lwfinger, 2022-03-18:
Your BIOS is behaving badly. Is this a Lenovo laptop?

As root, create a file named /etc/modprobe.d/50-rtw89.conf

In that file, add a line saying
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y

colttt, 2022-03-20:
yes it is a Lenovo T14 Gen2 AMD:
```
BIOS Information
        Vendor: LENOVO
        Version: R1MET43W (1.13 )
        Release Date: 11/05/2021
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
           [...]
        BIOS Revision: 1.13
        Firmware Revision: 1.13
System Information
        Manufacturer: LENOVO
        Product Name: 20XL0012GE
        Version: ThinkPad T14 Gen 2a
        Serial Number: PF3JWH2K
        UUID: 1a13b84c-2a59-11b2-a85c-9fa8ddbef6b3
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_20XL_BU_Think_FM_ThinkPad T14 Gen 2a
        Family: ThinkPad T14 Gen 2a
```
The issue still exist:
```
 cat /etc/modprobe.d/50-rtw89.conf 
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y
```

another strange issue:  nearly no packet loss and very high ping times:
```
--- 8.8.8.8 ping statistics ---
100 packets transmitted, 100 received, 0% packet loss, time 99514ms
rtt min/avg/max/mdev = 798.277/4425.074/10320.159/2673.047 ms, pipe 11
```

but the error messages are still there in dmesg (and some new one):
```
[ 1532.499712] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1532.600273] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1532.700650] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1532.806014] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 1532.806038] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x4e  0x00  0xe0  0x81
[ 1533.398877] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1533.499233] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1533.599538] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1533.699889] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1533.800254] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1533.900582] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.000633] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.101214] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.201533] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.301885] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.402465] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.502509] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.602803] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.703138] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1534.725183] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
[ 1534.725206] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x4e  0x00  0xe0  0x81
[ 1535.401467] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1535.501547] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1535.602014] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1535.702482] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1535.802844] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1535.903198] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1536.003188] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1536.103822] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1536.203911] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1536.304272] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1536.404867] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[... 79 lines cut ...]
[ 1713.534650] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1713.634930] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1713.713215] rtw89_pci 0000:03:00.0:  [FWBB][WARN]Ttlen = 0x0018, macidcfg = 0x4e  0x00  0xe0  0x81
[ 1713.735048] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1713.835606] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1713.935949] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.035943] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.136497] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.237004] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.337285] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.437632] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.537901] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.638311] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.738646] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.839027] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1714.939089] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1715.039401] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1715.139861] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1715.240011] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1715.340797] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1715.440742] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1715.541131] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1715.641829] rtw89_pci 0000:03:00.0:  [FWBB][WARN][CMW] No specific sta connect
[ 1715.750333] rtw89_pci 0000:03:00.0: firmware failed to ack for entering ps mode
```

lwfinger, 2022-03-20:
Use the suggestion in the last paragraph of README.md. I have no further suggestions for you.

colttt, 2022-03-25:
latest version fixed that

closed 2022-03-25 by colttt

Jmacea, 2023-03-24:
Which latest version solved your problem?