lwfinger/rtw89 #134: Random disconnects

Leviathan13186, 2022-03-19:
Hello,
I have been having random disconnects the last few weeks. It would randomly loss connection and the signal strength indicator just shows a question mark. I thought it might be my WiFi but during this state my system can not see any other APs. This will go on for a minute or 2 then it would reconnect. I'm still pretty new so I'm sorry if this isn't descriptive enough. 
I am running PopOS 21.10 on a legion 5 15ach6 on kernel 5.16.11-76051611-generic.

lwfinger, 2022-03-19:
Your BIOS is behaving badly. Is this a Lenovo laptop?

As root, create a file named /etc/modprobe.d/50-rtw89.conf

In that file, add a line saying
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y

colttt, 2022-03-25:
did you try the latest version here from github? it works for me very well

closed 2022-12-14 by lwfinger