lwfinger/rtw89 #135: 22/03/2022 update caused Wireless Not Working on Manjaro

salimp2009, 2022-03-22:
Hi
Currently on Manjaro Gnomo with kernel 5.15
I was using your driver on my thinkpad e14 laptop for wireless 
todays update caused to wireless driver disappear . I removed / reinstalled again.
When I search from pacman or Aur it shows installed 

`package version :r170.ff422f1`

When i run inxi -N on terminal it shows;

```
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet
    driver: r8169
  Device-2: Realtek RTL8852AE 802.11ax PCIe Wireless Network Adapter
    driver: N/A
```

this is dkms.conf file shows;
```

PACKAGE_NAME="@_PKGBASE@"
PACKAGE_VERSION="@PKGVER@"
BUILT_MODULE_NAME[0]="${PACKAGE_NAME}core"
BUILT_MODULE_NAME[1]="${PACKAGE_NAME}pci"
MAKE[0]="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build modules"
CLEAN="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build clean"
DEST_MODULE_LOCATION[0]=/kernel/drivers/net/wireless/realtek/rtw89
DEST_MODULE_LOCATION[1]=/kernel/drivers/net/wireless/realtek/rtw89
AUTOINSTALL=yes
BUILD_EXCLUSIVE_KERNEL="^(4\.[0-9]+\.|5\.[0-9]\.|5\.1[0-5]\.)"
```
and this .install fille

```
post_upgrade() {
  if [[ $(vercmp "1:r0.4dc5371-1" "$2") -eq 1 ]]; then
    # upgraded from epoch 0, kernel <= 5.15
    for i in /usr/lib/modules/*/; do
      i="${i%/*}"
      echo "==> rtw89-dkms-git: depmod ${i}"
      depmod "${i##*/}"
    done
  fi
}
```

lkno0705, 2022-03-22:
Got the same Problem on Manjaro KDE, after todays update. I did use the Chaotic-AUR compiled binary to install the driver.

salimp2009, 2022-03-22:
> Got the same Problem on Manjaro KDE, after todays update. I did use the Chaotic-AUR compiled binary to install the driver.

I looked at Chaotic it looks likes it updates building from source; so if we clone the repo and built or if we can revert back to prev. version but not sure how to track it and do it to be honest.

lwfinger, 2022-03-22:
I need to see the log. Run 'dmesg > dmesg.txt' and attach dmesg.txt.

If you want to revert to an older version, git has that capability built in in its checkout function.

Chaotic sounds just that!

salimp2009, 2022-03-22:
> I need to see the log. Run 'dmesg > dmesg.txt' and attach dmesg.txt.
> 
> If you want to revert to an older version, git has that capability built in in its checkout function.
> 
> Chaotic sounds just that!

i will send you the dmesg.txt as soon as i get to my computer. Thanks for doing .It saved me since this was my first linux dostro 👍👍

lwfinger, 2022-03-22:
Do not delete the line, but you can obfuscate any IP addresses. Funny, I do not remember any IP addresses in dmesg.

salimp2009, 2022-03-22:
> Do not delete the line, but you can obfuscate any IP addresses. Funny, I do not remember any IP addresses in dmesg.

ok i did another dmesg after removing the phone  and rebooting that way it does not show those

[dmesg.txt](https://github.com/lwfinger/rtw89/files/8327025/dmesg.txt)

lwfinger, 2022-03-22:
Your dmesg shows no mention of the rtw89 at all, thus the driver has not been built correctly/

Do you use dkms? The latest update of rtw89 split the drivers into 4 .ko routines rather than the two previously employed in preparation for adding a driver for the RTL8852CE.. I updated the debian/dkms.conf, but it probably was not updated in the dkms data base.

Find the rtw89 source directory, cd to it and do 'make' and 'sudo make install', then reboot. At this poiont, you should have a wirel4ess driver.

lkno0705, 2022-03-22:
> > Got the same Problem on Manjaro KDE, after todays update. I did use the Chaotic-AUR compiled binary to install the driver.
> 
> I looked at Chaotic it looks likes it updates building from source; so if we clone the repo and built or if we can revert back to prev. version but not sure how to track it and do it to be honest. or maybe directly install from the repo, clone it and use make ... Not sure what will be best ..wait for a fix if possible

Yeah it provides precompiled binarys. However I also used the aur to install it and had the same problems. For now i cloned the repo, reverted back to the commit before the update, compiled it from src and updated the kernel modules. That fixed it for now. But currently I am not able to use the version from the update of commit “update for new module version”. I didnˋt try the newest commit yet though.

lwfinger, 2022-03-22:
Please build the latest commit. I am trying to learn if this is a dkms problem, or a real problem with the new commit, which works fine here. Of course, dkms is forbidden on my machine. I want to know exactly what is happening.

salimp2009, 2022-03-22:
> Please build the latest commit. I am trying to learn if this is a dkms problem, or a real problem with the new commit, which works fine here. Of course, dkms is forbidden on my machine. I want to know exactly what is happening.

I did but still it does not work.

salimp2009, 2022-03-22:
here is the new dmesg output .

[dmesg.txt](https://github.com/lwfinger/rtw89/files/8327555/dmesg.txt)

salimp2009, 2022-03-22:
> Can you get me that dmesg output?

I saw your earlier reply and i went into usr/src and cd into rtw directory and build sudo make then sudo make install; it successfully built it; reboot it ; it works now. Great....thank you. 
But now I have question..if you sent another update..will it be OK to update thru Aur ?? I guess it will . Thank you so much . 

And i used the latest update you sent thru Aur so I did not revert back. So your latest update should be OK.  It is this package ;
rtw89-r171.ad0672c

lwfinger, 2022-03-22:
I have no idea what Aur is. I have never sent an update though them. My latest change is ad0672c, so that seems to be correct.

salimp2009, 2022-03-22:
> I have no idea what Aur is. I have never sent an update though them. My latest change is [ad0672c](https://github.com/lwfinger/rtw89/commit/ad0672c7bd32f096e8b1d112c008e671d7f9f92a), so that seems to be correct.

I meant AUR ;  https://aur.archlinux.org/packages (package manager on Manjaro:)

And thank you; now I see how the commits and package name relate to each other. 
Great ...Just in case i have a problem like this in future

closed 2022-03-22 by salimp2009