lwfinger/rtw89 #140: Realtek 8852BE Supported?

HRex39, 2022-04-08:
Hello,
I tried to build the code but it seems don't support Realtek 8852BE?
```
git clone git://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make install
sudo modprobe xxx
```
Did I do something wrong or should I update my kernel?
```
$ uname -r
$ 5.14.0-1032-oem
```

lwfinger, 2022-04-08:
To my knowledge, no Linux code has been released for 8852BE. I know that the Realtek group is working on the 8852CE, but I do not know about the BE. This is a question for linux-wireless@vger.kernel.org. The Realtek developers read that mailing list, but do not read these issues as the noise component is too high.

closed 2022-04-08 by lwfinger

HRex39, 2022-04-13:
> To my knowledge, no Linux code has been released for 8852BE. I know that the Realtek group is working on the 8852CE, but I do not know about the BE. This is a question for [linux-wireless@vger.kernel.org](mailto:linux-wireless@vger.kernel.org). The Realtek developers read that mailing list, but do not read these issues as the noise component is too high.

In fact, there is an arm board driver [here](https://github.com/TinkerBoard2/kernel). You need to take it out and compile it separately. I succeeded in this operation, but I don't know whether I have the authority to take it out alone and open another repo, which depends on the driver's author HHH.

HRex39, 2022-04-13:
1. You can enable CONFIG_PLATFORM_I386_PC in the Makefile. Line 149 and Line 160.
[image]

2. add below codes in os_dep/linux/recv_linux.c:339
[image]
```
It's not elegant, but it's useful.
```

lwfinger, 2022-04-13:
When the README contains a section "What is Linux", it does not seem very sophisticated.

lwfinger, 2022-06-06:
I just created a new rtw8852be repo. I will be working on the problems with kernels newer that 5.18. The basic problem is that they are using memcpy to copy a relatively large struct into a u32 variable. Kernel 5.19-rc1 generates a run-time warning.

HRex39, 2022-06-06:
> I just created a new rtw8852be repo. I will be working on the problems with kernels newer that 5.18. The basic problem is that they are using memcpy to copy a relatively large struct into a u32 variable. Kernel 5.19-rc1 generates a run-time warning.

According to some existing hardware tests, ```_rtw_memcpy```appeared in more than a thousand places (of course this is a bit of an exaggeration) in the code, I really don't know how to modify this API, maybe you have any good way to fix these warnings? 

About a month ago, I created another rtl8852be repo, and now it seems that your solution is more elegant. After all, you are the real expert.

lwfinger, 2022-06-06:
Removing the _os_mem_cpy() wrapper takes a bit of hand work as its first argument must be deleted. Removing _rtw_memcpy() can be done with sed (mostly) as it has the same arguments as memcpy()..

lwfinger, 2022-06-09:
I did get the problems fixed. The driver now builds without warning, and runs on kernel 5.19-rc1.

HRex39, 2022-06-10:
Thank you very much for your work. By the way, if I want to use the 8852BE Bluetooth,do I just need to add following code in ```btusb.c``` or do I need to add something else?
```
{ USB_DEVICE(0x0bda, 0x4853), .driver_info = BTUSB_REALTEK |
						     BTUSB_WIDEBAND_SPEECH },
```

```
$ lsusb
$ Bus 003 Device 004: ID 0bda:4853 Realtek Semiconductor Corp. Bluetooth Radio

$ sudo dmesg | grep -i blue
[    5.284941] Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000b lmp_ver=0b lmp_subver=8852
[    5.284945] Bluetooth: hci0: RTL: unknown IC info, lmp subver 8852, hci rev 000b, hci ver 000b
[    5.284947] Bluetooth: hci0: RTL: assuming no firmware upload needed
```

[Add USB_DEVICE Reference](https://github.com/lwfinger/rtw89/issues/48#issuecomment-928187089)

HRex39, 2022-06-10:
Based on the latest kernel, he adapted the 8852B in the btrtl .c
```
/* 8852B */
{ IC_INFO(RTL_ROM_LMP_8852A, 0xb, 0xb, HCI_USB),
	.config_needed = false,
	.has_rom_version = true,
	.has_msft_ext = true,
	.fw_name  = "rtl_bt/rtl8852bu_fw.bin",
	.cfg_name = "rtl_bt/rtl8852bu_config" },
};
```
it just looks like all I need to do is make the kernel accept my device?

lwfinger, 2022-06-10:
That is a mystery. I do not see that on my system. Please add the following patch and report back the output of your "Blue" grep of dmesg.

inspect_bt_params.txt](https://github.com/lwfinger/rtw89/files/8881708/inspect_bt_params.txt)

My device has a different USB ID than yours. I will take care of submitting a patch adding both devices. Please post the portion of /sys/kernel/debug/usb/devices pertaining to your BT device. Thanks.

HRex39, 2022-06-11:
Currently, I changed a few places in the LTS kernel in 5.15 and then the bluetooth can detect some other devices. But now I don't have more devices to test...  Anyway, it needs more tests.
```Bus 003 Device 004: ID 0bda:4853 Realtek Semiconductor Corp. Bluetooth Radio```
```
T:  Bus=03 Lev=01 Prnt=01 Port=06 Cnt=03 Dev#=  4 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  4
P:  Vendor=05ac ProdID=12a8 Rev=11.02
S:  Manufacturer=Apple Inc.
S:  Product=iPhone
S:  SerialNumber=00008020001165E80E2A002E
C:  #Ifs= 1 Cfg#= 1 Atr=c0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 3 Cls=06(still) Sub=01 Prot=01 Driver=
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=64ms
C:  #Ifs= 3 Cfg#= 2 Atr=c0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 0 Cls=01(audio) Sub=01 Prot=00 Driver=
I:  If#= 1 Alt= 0 #EPs= 0 Cls=01(audio) Sub=02 Prot=00 Driver=
I:  If#= 1 Alt= 1 #EPs= 1 Cls=01(audio) Sub=02 Prot=00 Driver=
E:  Ad=81(I) Atr=01(Isoc) MxPS= 192 Ivl=1ms
I:  If#= 2 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=
E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=125us
C:  #Ifs= 2 Cfg#= 3 Atr=c0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 3 Cls=06(still) Sub=01 Prot=01 Driver=
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=64ms
I:  If#= 1 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=fe Prot=02 Driver=
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
C:* #Ifs= 3 Cfg#= 4 Atr=c0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=06(still) Sub=01 Prot=01 Driver=(none)
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=83(I) Atr=03(Int.) MxPS=  64 Ivl=64ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=fe Prot=02 Driver=usbfs
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:  If#= 2 Alt= 0 #EPs= 0 Cls=ff(vend.) Sub=fd Prot=01 Driver=ipheth
I:* If#= 2 Alt= 1 #EPs= 2 Cls=ff(vend.) Sub=fd Prot=01 Driver=ipheth
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=05(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:  If#= 2 Alt= 2 #EPs= 2 Cls=ff(vend.) Sub=fd Prot=01 Driver=ipheth
E:  Ad=86(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=05(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
```