lwfinger/rtw89 #144: GUI freezes -> failed to get available addr cam

dakorpar, 2022-05-02:
So in some cases, specially after returning from lock or sleep ubuntu freezes completelly and on console I just get this errors out:

[image]

raphink, 2022-05-05:
Same here. Reported on Ubuntu as https://bugs.launchpad.net/ubuntu/+source/linux-firmware/+bug/1971656

carlos-m-rodrigues, 2022-05-17:
Hi @lwfinger, I'm experiencing the same problem with a Lenovo L15. I took a screenshot from the moment the module enters in infinite loop. 

[image]

I am to reproduce the problem when downloading big files, so if I can help please let me know.

lwfinger, 2022-05-17:
Have you added the module options for rtw88pci?

carlos-m-rodrigues, 2022-05-18:
Nop, I'm using a vanilla xubuntu with kernel 5.15.0-27. I tried kernel 5.18.0-rc7 with no success.

carlos-m-rodrigues, 2022-05-18:
Just another update! I've compiled the module from this repository and the problem persists:

[image]

[image]

From my point of view, it seems to be related with that message **"ieee80211 phy0: Hardware restart was requested"**. Do you have any clue what it can be?

lwfinger, 2022-05-18:
My comment was a "gentle" reminder for you to apply those options to the module.

The general cause for these kinds of failures are an inconsistency between the Lenovo or HP BIOS and the PCI handling.  The modules options that I referenced handle those problems in many cases. Obviously, I need to spell out the instructions, even though I have done so many, many times before!

As root, create a file named /etc/modprobe.d/50-rtw89.conf.
In that file, add two lines that say:
options rtw89pci disable_aspm_l1=y
options rtw89_pci disable_aspm_l1=y

The first line will handle the case where you are using the driver from this repo. The second is for the driver from the kernel.

If those lines do not help, append ' disable_aspm_l1ss=y' to each of those lines. If still no help, append ' disable_clkreq=y' to both lines.

carlos-m-rodrigues, 2022-05-18:
Even with this options the problem remains :-(. Is there something else I can do to help?

lwfinger, 2022-05-18:
If you have tested that the options are correct by unloading the drivers, and reloading them with the "-v" option for modprobe, then you have exhausted my capability and need to report the problem to the linux wireless mailing list as described in the last paragraph of README.md. Note that you must submit text messages - no html, thus your screen shots cannot be in-lined there. You should be able to find the logged information using journalctl, and copy it to the message. If not, you could attach the screen shot.

closed 2022-12-14 by lwfinger