lwfinger/rtw89 #146: Couldn't modprobe. Unknown symbol.

longPtrCall, 2022-06-11:
Couldn't modeprobe/insmod: invalid argument.
Dmesg output:
...
disagree about version of symbol ieee80211_txq_get_depth
rtw89core: Unknow symbol ieee80211_txq_get_depth (err -22)
...
And such errors for all symbols ieee80211_*.

lwfinger, 2022-06-11:
That error usually arises because the driver has been built against a different source header version than the running kernel.

The kernel version shown in the "make -C" line must be the same as shown in the 'uname -r' output.

lwfinger, 2022-06-11:
Actually, those symbols come from mac80211. Did you do a 'make installl'? If so, the depmod step will find such relations and modprobe should work. Without the installion, you will need a separate 'sudo modprobe -v mac80211'.

longPtrCall, 2022-06-11:
Now I'm trying to re-build it.
An error that i've receive:
./include/net/mac.80211.h:23:10: fatal error: net/ieee80211_radiotap.h: No such file or directory

IDK what I have to do. I don't even know which devel I need to install to get successfull build.

lwfinger, 2022-06-11:
That header file has been in the kernel since 4.11.File README.md lists the necessary requirements for a number of distros. If yours is not one of them, please tell me what distro you are using.

longPtrCall, 2022-06-11:
I'm using void linux (x86_64).
I've added a file that missing to location usr/include/net,  but have no result.

lwfinger, 2022-06-11:
I do not think you got the kernel headers with that new kernel.

I use Void Linux on a old PowerBook with a ppc32 processor. I have never been able to get a graphical screen, thus I use it in command-line mode. I am not very pleased with it.

On the Web, I found the following:

The linux meta package, installed by default, depends on one of the kernel packages, usually the package containing the latest mainline kernel that works with all DKMS modules. Newer kernels might be available in the repository, but are not necessarily considered stable enough to be the default; use these at your own risk. If you wish to use a more recent kernel and have DKMS modules that you need to build, install the relevant linux<x>.<y>-headers package, then use [xbps-reconfigure(1)](https://man.voidlinux.org/xbps-reconfigure.1) to reconfigure the linux<x>.<y> package you installed. This will build the DKMS modules.

My understanding of this is that you need to use xbps to install the linux5.15-headers package.

longPtrCall, 2022-06-11:
I already have this version of headers. Also I've download other devel libs, but it was redundant, still have no ieee80211 headers.
Reconfiguring doesn't helps too :[

lwfinger, 2022-06-11:
The critical thing is that 'ls /lib/modules/$(uname -r)/build/include' shows a number of directories such as linux, net among others.

longPtrCall, 2022-06-11:
Finaly DONE...
I've manually added file ieee80211_radiotap.h to the directory lib/modules/$(uname -r)/build/include/net.
File's content has been copied from original file in linux base (copy-pasted from linux site).

lwfinger, 2022-06-12:
Strange that one file was missing. Do you still have those unknown symbol errors? Is that the problem?

longPtrCall, 2022-06-12:
Yes, even after rebuilding I have unknown symbol error.

lwfinger, 2022-06-12:
Please do the following from directory rtw89, and post the output:

sudo modprobe -v mac80211
make
sudo make install
sudo modprobe -v rtw_8852ce

I am guessing at that last line. If your device is not an 8852ce, then adjust it accordingly.

longPtrCall, 2022-06-13:
I followed thos steps.
rtw_8852ce is not found, only rtw_8852ae are available, so I did sudo modprobe -v rtw_8852ae again, and result the same.

On the "make" stage there's some messages, like
''
Skipping BTF generation for rtw89core due to unavailability of vmlinux.
''
Similar messages for rtw89pci, rtw_8852a and rtw_8852ae.
Install stage success.
Modprobe failed as I've mentioned.

I've also tried to insert rtw_8852a, but get previous error again.

lwfinger, 2022-06-13:
Where is the "sudo modprobe -v mac80211". I need to see what you enter, and what gets output!''. Do not try any shortcuts. I would also like to see the output of 'lsmod | egrep "802|885".

After the 'sudo make install', you should 'sudo modprobe -v rtw_8852ae'. If you do not install the modules, then you woll need to do ALL of the following:

sudo modprobe -v mac80211
sudo insmod rtw89pci.ko
sudo insmod rtw89core.ko
sudo insmod rtw_8852a.ko
sudo insmod rtw_8852ae.ko

longPtrCall, 2022-06-13:
>> sudo modprobe -v mac80211

<<
libkmod: kmod_config_parse: /etc/modprobe.d/default.conf line 1: ignoring bad line starting with 'option'
insmod /lib/modules/5.15.45_1/kernel/lib/crypto/libarc4.ko.gz
insmod /lib/modules/5.15.45_1/updates/compact/compact.ko.gz
insmod /lib/modules/5.15.45_1/updates/net/wireless/cfg80211.ko.gz
insmod /lib/modules/5.15.45_1/updates/net/mac80211/mac80211.ko.gz

>> lsmod | egrep "802|885"

<<
mac80211   0
cfg80211     1
compat         2
libarc4          1
nls_iso8859_1    2
rfkill               5

lwfinger, 2022-06-13:
Something is wrong in file /etc/modprobe.d/default.conf. Perhaps that is messing up the depmod. That file is not on my system!

Symbol ieee80211_txq_get_depth is definitely in mac80211 - at least on my system.

Does the sequence
sudo insmod rtw89pci.ko
sudo insmod rtw89core.ko
sudo insmod rtw_8852a.ko
sudo insmod rtw_8852ae.ko
work? If so, that would verify that the depmod step is failing.

longPtrCall, 2022-06-13:
insmod failed on rt89pci. DMESG shows ieee symbols are missing

longPtrCall, 2022-06-14:
Here's my dmesg output. While booting rtw89core obtains that error.
[dmesg.txt](https://github.com/lwfinger/rtw89/files/8897803/dmesg.txt)

longPtrCall, 2022-06-16:
I've installed void core (without any additional libs). All packages has been installed manually.
XBPS doesn't have binary packages for Realtek wifi-bluetoth controller.
By the way, bluetooth services are shown as enabled in lsmod.

I'm not sure that VM can emulate this issue. I can provide any additional information from my PC.

lwfinger, 2022-06-16:
[image]

I cannot run the code in a Void VM, but I can build and load it.
After I got all the various pieces installed (git to download the code, the kernel headers, make and gcc), I did the usual make. sudo make install, and sudo modprobe -v rtw_8852ae. The tail end of the make part, and the rest are in the attsched screen shot. It loaded exactly the way I expected. Something is wrong with your system.

lwfinger, 2022-06-17:
No. I have at least 4 old kernels on my system, and have never seen this problem.

Please delete that file  /etc/modprobe.d/default.conf. It has a totally wrong format, and I worry that it is affecting the depmod command that is part of driver installation.

longPtrCall, 2022-06-17:
I've commented its content, so there's no strange warnings administrators anymore.

lwfinger, 2022-06-17:
Just to verify, have you done a 'sudo make install' since you commented out that strange line? If so, it is some problem with your system, at least it did not show in my installation of Void with an xfce desktop. I started with the live system and did the installation from there. You mentioned that you installed "void core". I hoped you selected the glibc version. I have no experience with musl. The specific file that I used is void-live-x86_64-20210930-xfce.iso. I noticed that the download page made reference to a tar file. I did not use that method.

longPtrCall, 2022-06-17:
https://alpha.de.repo.voidlinux.org/live/current/void-live-x86_64-20210930.iso - it's a link that I've used. 
I'll try to re-install again, but not sure that's helps.

lwfinger, 2022-06-17:
I expect that that file would give you a Gnome desktop, but that should not matter.

longPtrCall, 2022-06-18:
I'm using kde plasma

longPtrCall, 2022-06-19:
I've decided to analyse source code...

Here's errors:

File mac80211.c:

...
#include "cam.h" // core.h: file <linux/average.h>
...
static void rtw89_ops_tx(struct ieee80211_hw *hw,
			 struct ieee80211_tx_control *control, // Declaration will not be visible outside this function.
			 struct sk_buff *skb)
{
	struct rtw89_dev *rtwdev = hw->priv; // Incomplete definition of struct. Forwarded from core.h
	struct ieee80211_tx_info *info = IEEE80211_SKB_CB(skb); // Could not convert from int to pointer. Implicit declaration of function "IEEE80211_SKB_CB" is invalid in C99
	struct ieee80211_vif *vif = info->control.vif; // Incomplete struct definition
	struct ieee80211_sta *sta = control->sta; // Incomplete struct definition
	int ret, qsel;

	ret = rtw89_core_tx_write(rtwdev, vif, sta, skb, &qsel);
	if (ret) {
		rtw89_err(rtwdev, "failed to transmit skb: %d\n", ret); // Implicit declaration of function is invalid in C99
		ieee80211_free_txskb(hw, skb); // Implicit declaration of function is invalid in C99
		return;
	}
	rtw89_core_tx_kick_off(rtwdev, qsel);
}
static void rtw89_ops_wake_tx_queue(struct ieee80211_hw *hw,
				    struct ieee80211_txq *txq)
{
	struct rtw89_dev *rtwdev = hw->priv; // Incomplete definition of struct. Forwarded from core.h

	ieee80211_schedule_txq(hw, txq); // Implicit declaration of function is invalid in C99
	queue_work(rtwdev->txq_wq, &rtwdev->txq_work); // Implicit declaration of function is invalid in C99
}

static int rtw89_ops_start(struct ieee80211_hw *hw)
{
	struct rtw89_dev *rtwdev = hw->priv; // Incomplete definition of struct. Forwarded from core.h
	int ret;

	mutex_lock(&rtwdev->mutex); // Implicit declaration of function is invalid in C99
	ret = rtw89_core_start(rtwdev);
	mutex_unlock(&rtwdev->mutex); // Implicit declaration of function is invalid in C99

	return ret;
}

static void rtw89_ops_stop(struct ieee80211_hw *hw)
{
	struct rtw89_dev *rtwdev = hw->priv;

	mutex_lock(&rtwdev->mutex); // Implicit declaration of function is invalid in C99
	rtw89_core_stop(rtwdev);
	mutex_unlock(&rtwdev->mutex); // Implicit declaration of function is invalid in C99
}

static int rtw89_ops_config(struct ieee80211_hw *hw, u32 changed)
{
	struct rtw89_dev *rtwdev = hw->priv;

	/* let previous ips work finish to ensure we don't leave ips twice */
	cancel_work_sync(&rtwdev->ips_work); // Implicit declaration of function is invalid in C99

	mutex_lock(&rtwdev->mutex); // Implicit declaration of function is invalid in C99
	rtw89_leave_ps_mode(rtwdev);

	if ((changed & IEEE80211_CONF_CHANGE_IDLE) && // Undeclared identifier
	    !(hw->conf.flags & IEEE80211_CONF_IDLE)) // Undeclared identifier
		rtw89_leave_ips(rtwdev);

	if (changed & IEEE80211_CONF_CHANGE_PS) {
		if (hw->conf.flags & IEEE80211_CONF_PS) {
			rtwdev->lps_enabled = true;
		} else {
			rtw89_leave_lps(rtwdev);
			rtwdev->lps_enabled = false;
		}
	}

	if (changed & IEEE80211_CONF_CHANGE_CHANNEL) // Undeclared identifier
		rtw89_set_channel(rtwdev);

	if ((changed & IEEE80211_CONF_CHANGE_IDLE) && // Undeclared identifier
	    (hw->conf.flags & IEEE80211_CONF_IDLE))
		rtw89_enter_ips(rtwdev);

	mutex_unlock(&rtwdev->mutex);

	return 0;
}

File core.h:
struct rtw89_txpwr_byrate {
	s8 cck[RTW89_RATE_CCK_MAX]; // Unknown type (s8 is unknown in whole while. Should I define it as a signed char?)
	s8 ofdm[RTW89_RATE_OFDM_MAX];
	s8 mcs[RTW89_NSS_MAX][RTW89_RATE_MCS_MAX];
	s8 hedcm[RTW89_NSS_HEDCM_MAX][RTW89_RATE_HEDCM_MAX];
	s8 offset[RTW89_RATE_OFFSET_MAX];
};

In other files i also have unknown u32 (maybe can be fixed with typedef u32 unsigned int?)
GENMASK is undeclared.
struct _packed defined twice.
Undefined identifier __le32
Undefined identifier RTW89_DEF_CHAN_2G
File #include <linux/vmalloc.h> is not found
Undeclared identifier BIT
Unknown type u16 (unsigned short int?)
And same errors in other files.

This errors is shown by clang.

lwfinger, 2022-06-19:
If you want to use clang, you will need to fix these problems yourself. I do not have it installed, and I have no plans to use it.

longPtrCall, 2022-06-19:
No, I mean clang show this as errors.
Kate uses clang to analyse a code.
IDK how make competes its job with some files missing

lwfinger, 2022-06-19:
I do not support IDK either. Does 'make', 'sudo make install', and 'sudo modprobe -v rtw_8821c.ko' work now?

longPtrCall, 2022-06-19:
make and make install works.
modprobe doesn't.
Error the same.

lwfinger, 2022-06-19:
What happens with 'sudo depmod -a'.

longPtrCall, 2022-06-19:
Here's output:
[modules.txt](https://github.com/lwfinger/rtw89/files/8936376/modules.txt)

lwfinger, 2022-06-19:
That output shows that depmod is working.

Please post the output of 'lspci -nn'. Let's verify what card you have.

longPtrCall, 2022-06-19:
[output.txt](https://github.com/lwfinger/rtw89/files/8936484/output.txt)

longPtrCall, 2022-06-26:
After a weeks of suffering I found a solution.

First weird thing I have its another driver in dkms, probably there's a reason of incompability. I decide to delete other rtl drivers (looks like I suddenly download it, sry I'm new in linux, dont intendet to waste your time).

I installed new version of kernel and re-build this driver. After reboot all works.

The issue is closed. Thank for your attention and sorry for time I took.

closed 2022-06-26 by longPtrCall