lwfinger/rtw89 #148: Can't scan bluetooth devices on RTL8852AE

benjamimgois, 2022-06-29:
Hi, i have this network card TRL8852AE in a Xiaomi notebook. The wifi works fine with the rtw89 driver from Kernel 5.18. Bluetooth is detected but it can not scan for any device. Any help ?

```
inxi --bluetooth -xxx                                                              
Bluetooth:
  Device-1: Opticis Bluetooth Radio type: USB driver: btusb v: 0.8
    bus-ID: 1-4:4 chip-ID: 0cb8:c549 cla
ss-ID: e001 serial: 00e04c000001
  Report: rfkill ID: hci0 rfk-id: 1 state: up address: see --recommends
```

[dmesg.txt](https://github.com/lwfinger/rtw89/files/9006331/dmesg.txt)
[lspci_v.txt](https://github.com/lwfinger/rtw89/files/9006332/lspci_v.txt)

I'm on manjaro 5.18.7 kernel

lwfinger, 2022-06-29:
In your dmesg.txt, I saw no indication that the Bluetooth driver ever tried to load firmware.

When you do 'lsmod | grep bt', do you see btrtl listed?

Perhaps you should post the output of 'lsusb'. The BT device is interfaces through USB, not PCI.

0versun, 2022-06-29:
T:  Bus=01 Lev=01 Prnt=01 Port=03 Cnt=02 Dev#=  3 Spd=12   MxCh= 0
D:  Ver= 1.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0cb8 ProdID=c549 Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=Bluetooth Radio
C:* #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=1ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   0 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   0 Ivl=1ms
I:  If#= 1 Alt= 1 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   9 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   9 Ivl=1ms
I:  If#= 1 Alt= 2 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  17 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  17 Ivl=1ms
I:  If#= 1 Alt= 3 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  25 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  25 Ivl=1ms
I:  If#= 1 Alt= 4 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  33 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  33 Ivl=1ms
I:  If#= 1 Alt= 5 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  49 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  49 Ivl=1ms

lwfinger, 2022-06-29:
[add_new_usb.txt](https://github.com/lwfinger/rtw89/files/9013089/add_new_usb.txt)

The patch file above (given a .txt name to meet GitHub's conventions) will add your device. If you install your kernel source using git, the command 'git am add_new_usb.txt' will add your device to the tables. If you get the source from a tarball, then 'patch -p1 < add_new_usb.txt' will do the trick.

benjamimgois, 2022-06-29:
I'll try to compile it and post the results here. Thankyou very much @lwfinger !

lwfinger, 2022-06-30:
It is your lucky day! As I got two requests for addition of Opticis BT devices, I decided to create a new code repository for the modified BT codes.

In a console terminal, do the following:
git clone git@github.con:lwfinger/rtw89-BT.git
cd rtw89-bt
make
sudo make install

If you are using secure boot, replace that install step with
sudo make sign-install

If you need to do this, please read the appropriate section of README.md so that you do not end up making your computer unbootable.

Good luck.

0versun, 2022-06-30:
Unfortunately
`git clone git@github.com:lwfinger/rtw89-BT.git
Cloning into 'rtw89-BT'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
`

0versun, 2022-06-30:
> It is your lucky day! As I got two requests for addition of Opticis BT devices, I decided to create a new code repository for the modified BT codes.
> 
> In a console terminal, do the following: git clone [git@github.com](mailto:git@github.com):lwfinger/rtw89-BT.git cd rtw89-bt make sudo make install
> 
> If you are using secure boot, replace that install step with sudo make sign-install
> 
> If you need to do this, please read the appropriate section of README.md so that you do not end up making your computer unbootable.
> 
> Good luck.

i made this changes but its still not work, i will wait for [benjamimgois] try

lwfinger, 2022-06-30:
If it builds and loads but does not work, you have gotten to the point where I say "I do not do Bluetooth". Sorry.

@0versun: GitHub has changed their security. That command is what I need to use to clone one of my repos. You probably need to use

git clone https://github.com/lwfinger/rtw89-BT.git

0versun, 2022-06-30:
> If it builds and loads but does not work, you have gotten to the point where I say "I do not do Bluetooth". Sorry.
> 
> @0versun: GitHub has changed their security. That command is what I need to use to clone one of my repos. You probably need to use
> 
> git clone https://github.com/lwfinger/rtw89-BT.git

Maybe i need do something extra after this steps

ake -C /lib/modules/5.17.15-1-MANJARO/build M=/home/oversun/Downloads/rtw89-BT modules
make[1]: Entering directory '/usr/lib/modules/5.17.15-1-MANJARO/build'
make[1]: Leaving directory '/usr/lib/modules/5.17.15-1-MANJARO/build'
Install btusb/btrtl SUCCESS

lwfinger, 2022-06-30:
You likely have to unload and reload btusb to get the new drivers. You should see the Realtek device loading its config and firmware files.

0versun, 2022-06-30:
Something like that ?

T:  Bus=01 Lev=01 Prnt=01 Port=03 Cnt=02 Dev#=  3 Spd=12   MxCh= 0
D:  Ver= 1.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0cb8 ProdID=c549 Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=Bluetooth Radio
S:  SerialNumber=00e04c000001
C:* #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=1ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   0 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   0 Ivl=1ms
I:  If#= 1 Alt= 1 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   9 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   9 Ivl=1ms
I:  If#= 1 Alt= 2 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  17 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  17 Ivl=1ms
I:  If#= 1 Alt= 3 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  25 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  25 Ivl=1ms
I:  If#= 1 Alt= 4 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  33 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  33 Ivl=1ms
I:  If#= 1 Alt= 5 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  49 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  49 Ivl=1ms

0versun, 2022-06-30:
That one ?

 
sudo dmesg | grep Blue                                                                                                                                          ✔ 
[    1.652129] usb 1-4: Product: Bluetooth Radio
[    2.257783] Bluetooth: Core ver 2.22
[    2.257806] Bluetooth: HCI device and connection manager initialized
[    2.257810] Bluetooth: HCI socket layer initialized
[    2.257812] Bluetooth: L2CAP socket layer initialized
[    2.257816] Bluetooth: SCO socket layer initialized
[    4.102888] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.102891] Bluetooth: BNEP filters: protocol multicast
[    4.102896] Bluetooth: BNEP socket layer initialized
[   12.886427] Bluetooth: RFCOMM TTY layer initialized
[   12.886435] Bluetooth: RFCOMM socket layer initialized
[   12.886439] Bluetooth: RFCOMM ver 1.11

lwfinger, 2022-06-30:
No indication that the Realtek device was found. I get

usb 2-4: new full-speed USB device number 5 using xhci_hcd
usb 2-4: New USB device found, idVendor=8087, idProduct=07dc, bcdDevice= 0.01
usb 2-4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Bluetooth: hci1: Legacy ROM 2.5 revision 8.0 build 2 week 3 2013
Bluetooth: hci1: Intel Bluetooth firmware file: intel/ibt-hw-37.7.10-fw-1.80.2.3.d.bseq
Bluetooth: hci1: Intel BT fw patch 0x57 completed & activated
usb 2-13: USB disconnect, device number 3
usb 2-13: new full-speed USB device number 6 using xhci_hcd
usb 2-13: New USB device found, idVendor=0bda, idProduct=887b, bcdDevice= 0.00
usb 2-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 2-13: Product: Bluetooth Radio
usb 2-13: Manufacturer: Realtek
usb 2-13: SerialNumber: 00e04c000001
Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000b lmp_ver=0b lmp_subver=8852
Bluetooth: hci0: RTL: rom_version status=0 version=1
Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin
Bluetooth: hci0: command tx timeout

That tx timeout may be due to the fact that I have a RTW8852BE and I am not sure about it.

benjamimgois, 2022-07-01:
Hey guys, sorry for my late. I can only try this things at night. I followed all the instructions and got the same result as @0versun .  Here is my dmesg output right after i reload the btusb module.

```
[  742.695690] usbcore: registered new interface driver btusb
[  742.696819] audit: type=1106 audit(1656634802.438:197): pid=5756 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  742.696954] audit: type=1104 audit(1656634802.438:198): pid=5756 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  742.701759] audit: type=1130 audit(1656634802.441:199): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  747.710225] audit: type=1131 audit(1656634807.450:200): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  750.432385] audit: type=1101 audit(1656634810.169:201): pid=5791 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="benjamim" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  750.432710] audit: type=1110 audit(1656634810.169:202): pid=5791 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  750.433285] audit: type=1105 audit(1656634810.172:203): pid=5791 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  750.438946] audit: type=1106 audit(1656634810.175:204): pid=5791 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  750.439002] audit: type=1104 audit(1656634810.175:205): pid=5791 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  758.702489] audit: type=1101 audit(1656634818.440:206): pid=5813 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="benjamim" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  758.702839] audit: type=1110 audit(1656634818.440:207): pid=5813 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  758.703388] audit: type=1105 audit(1656634818.440:208): pid=5813 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'

```

lwfinger, 2022-07-01:
There is no indication that your device was recognized. What does lsusb show?

benjamimgois, 2022-07-01:
> There is no indication that your device was recognized. What does lsusb show?

Here is the `lsusb -v` output

[lsusb_v_519.txt](https://github.com/lwfinger/rtw89/files/9031986/lsusb_v_519.txt)

lwfinger, 2022-07-01:
That USB ID is in the table. Please do the following steps:

make clean
make
sudo make install
sudo modprobe -rv btusb
sudo modprobe -v btusb

Paste the console output into this thread.

At this point, do 'dmesg | less' and find a line that looks like "usb 2-13: new full-speed USB device number 6 using xhci_hcd". The details will differ depending on how your chip is interfaced. Please post that line and those that follow describing either "usb", or "Bluetooth".

benjamimgois, 2022-07-02:
Ok, here is the console output. (The squares are manjaro icons used on the ZSH theme).

 ```
   ~  cd rtw89-BT                                                                                    ✔ 
    ~/rtw89-BT    main ?1  make clean                                                              ✔ 
    ~/rtw89-BT    main ?1  make                                                                    ✔ 
make -C /lib/modules/5.18.8-1-MANJARO/build M=/home/benjamim/rtw89-BT modules
make[1]: Entrando no diretório '/usr/lib/modules/5.18.8-1-MANJARO/build'

  CC [M]  /home/benjamim/rtw89-BT/btusb.o
  LD [M]  /home/benjamim/rtw89-BT/btusb.prelink.o
  CC [M]  /home/benjamim/rtw89-BT/btrtl.o
  LD [M]  /home/benjamim/rtw89-BT/btrtl.prelink.o
  MODPOST /home/benjamim/rtw89-BT/Module.symvers
  CC [M]  /home/benjamim/rtw89-BT/btrtl.mod.o
  LD [M]  /home/benjamim/rtw89-BT/btrtl.ko
  BTF [M] /home/benjamim/rtw89-BT/btrtl.ko
  CC [M]  /home/benjamim/rtw89-BT/btusb.mod.o
  LD [M]  /home/benjamim/rtw89-BT/btusb.ko
  BTF [M] /home/benjamim/rtw89-BT/btusb.ko
make[1]: Saindo do diretório '/usr/lib/modules/5.18.8-1-MANJARO/build'

    ~/rtw89-BT    main ?1  sudo make install                                                ✔  3s  
[sudo] senha para benjamim: 
make -C /lib/modules/5.18.8-1-MANJARO/build M=/home/benjamim/rtw89-BT modules
make[1]: Entrando no diretório '/usr/lib/modules/5.18.8-1-MANJARO/build'

make[1]: Saindo do diretório '/usr/lib/modules/5.18.8-1-MANJARO/build'

Install btusb/btrtl SUCCESS
    ~/rtw89-BT    main ?1   sudo modprobe -rv btusb                                        ✔  12s  
rmmod btusb
rmmod btmtk
rmmod btintel
rmmod btbcm
rmmod btrtl
    ~/rtw89-BT    main ?1   sudo modprobe -v btusb                                                 ✔ 
insmod /lib/modules/5.18.8-1-MANJARO/kernel/drivers/bluetooth/btmtk.ko.xz 
insmod /lib/modules/5.18.8-1-MANJARO/kernel/drivers/bluetooth/btintel.ko.xz 
insmod /lib/modules/5.18.8-1-MANJARO/kernel/drivers/bluetooth/btbcm.ko.xz 
insmod /lib/modules/5.18.8-1-MANJARO/kernel/drivers/bluetooth/btrtl.ko.xz 
insmod /lib/modules/5.18.8-1-MANJARO/kernel/drivers/bluetooth/btusb.ko.xz reset=1 
    ~/rtw89-BT    main ?1                                                                   
```

Now the dmesg | less output. I guess these are the lines you were refering to. But i'll the the full output in a text file, just in case.

[dmesg_output.txt](https://github.com/lwfinger/rtw89/files/9032240/dmesg_output.txt)

```
[    1.678198] usb 1-4: new full-speed USB device number 4 using xhci_hcd
[    1.680439] ipmi device interface
[    1.684812] BTRFS info (device nvme0n1p2: state M): turning on async discard
[    1.684815] BTRFS info (device nvme0n1p2: state M): using free space tree
[    1.685371] systemd[1]: Finished Remount Root and Kernel File Systems.
[    1.685456] systemd[1]: First Boot Wizard was skipped because of a failed condition check (ConditionFirstBoot=yes).
[    1.686415] systemd[1]: Rebuild Hardware Database was skipped because of a failed condition check (ConditionNeedsUpdate=/etc).
[    1.686941] systemd[1]: Starting Load/Save Random Seed...
[    1.686975] systemd[1]: Create System Users was skipped because of a failed condition check (ConditionNeedsUpdate=/etc).
[    1.687358] systemd[1]: Starting Create Static Device Nodes in /dev...
[    1.692209] systemd[1]: Finished Load/Save Random Seed.
[    1.692276] systemd[1]: First Boot Complete was skipped because of a failed condition check (ConditionFirstBoot=yes).
[    1.698634] vboxdrv: loading out-of-tree module taints kernel.
[    1.698740] vboxdrv: module verification failed: signature and/or required key missing - tainting kernel
[    1.700523] systemd[1]: Finished Create Static Device Nodes in /dev.
[    1.700579] systemd[1]: Entropy Daemon based on the HAVEGE algorithm was skipped because of a failed condition check (ConditionKernelVersion=<5.6).
[    1.701031] vboxdrv: Found 12 processor cores
[    1.701378] systemd[1]: Starting Journal Service...
[    1.701983] systemd[1]: Starting Rule-based Manager for Device Events and Files...
[    1.714935] systemd[1]: Finished Coldplug All udev Devices.
[... 53 lines cut ...]
[   15.004948] wlp1s0: associate with b4:0f:3b:5c:e7:8c (try 1/3)
[   15.019747] wlp1s0: RX AssocResp from b4:0f:3b:5c:e7:8c (capab=0x411 status=0 aid=5)
[   15.128540] wlp1s0: associated
[   15.128616] wlp1s0: Limiting TX power to 30 (30 - 0) dBm as advertised by b4:0f:3b:5c:e7:8c
[   15.235053] IPv6: ADDRCONF(NETDEV_CHANGE): wlp1s0: link becomes ready
[   16.396227] kauditd_printk_skb: 14 callbacks suppressed
...
[   75.800544] usbcore: deregistering interface driver btusb
[   75.925605] audit: type=1130 audit(1656726943.052:117): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   76.049409] audit: type=1106 audit(1656726943.175:118): pid=2827 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[   76.049456] audit: type=1104 audit(1656726943.175:119): pid=2827 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[... cut ...]
[  225.833356] audit: type=1110 audit(1656727092.957:129): pid=3011 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  225.834164] audit: type=1105 audit(1656727092.957:130): pid=3011 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'

```

lwfinger, 2022-07-04:
I just pushed a change that will print out why the device is not being recognized. Do a git pull, make, install, unload and load, then run 'dmesg | egrep "******" and post the output.

benjamimgois, 2022-07-04:
Ok, here is the output of `dmesg | egrep "******"` running the latest git code.

[dmesg.txt](https://github.com/lwfinger/rtw89/files/9035755/dmesg.txt)

lwfinger, 2022-07-04:
I added a couple more diagnostics Rinse and repeat, but this time, post the output of dmesg. Thanks.

benjamimgois, 2022-07-04:
> I added a couple more diagnostics Rinse and repeat, but this time, post the output of dmesg. Thanks.

Here is the `dmesg` output from the latest git-code.  

[dmesg_full.txt](https://github.com/lwfinger/rtw89/files/9042254/dmesg_full.txt)

lwfinger, 2022-07-05:
I have no idea what is going on. None of my test printouts are being triggered. I will need to look deeper into btusb.

lwfinger, 2022-07-05:
I added some more diagnostic outputs. Pull, build, and post dmesg.

lwfinger, 2022-07-06:
I was trying to get the BT part of my rtl8852be working with btusb, when I got an E-mail telling me of a different driver for it. I discovered that the firmware was different. When I installed the new version, it worked. I then mounted my rtl8852ae, which also failed. Copying the firmware from the linux-firmware repo started it working as well.

I have pushed the new 8852a firmware to the repo. Please pull and try again.

lwfinger, 2022-07-06:
I pushed some more debugging. Please try again.

benjamimgois, 2022-07-06:
Ok, here it is.

[dmesg.txt](https://github.com/lwfinger/rtw89/files/9055723/dmesg.txt)

lwfinger, 2022-07-07:
For reasons I do not understand, the normal 2-step process of selecting a device in the driver is failing. As a temporary workaround, I added device Vendor=0cb8 ProdID=c549 to the first step, which should cause the driver to load.
Please do the following:

git pull
make
sudo make install
sudo modprobe -rv btusb
sudo modprobe -v btusb

Please post the dmesg after all that.

benjamimgois, 2022-07-08:
Sure, here is the new output,

[dmesg.txt](https://github.com/lwfinger/rtw89/files/9067956/dmesg.txt)

lwfinger, 2022-07-08:
I do not understand this lack of loading the BT driver. Do you have a blacklist entry for any of the btxxx drivers?

Are the drivers loaded? 'lsmod | grep bt'

Please run the following:
sudo modprobe -rv btusb
sudo modprobe -v btusb

Post the output from those commands, and the last few lines from dmesg. I do not need everything. Start with the line "usbcore: deregistering interface driver btusb" and continue to the end. Copy and paste the output into the GitHub box.

benjamimgois, 2022-07-08:
There's no blacklisted driver. Also, the btusb seems to be loading correctly:

lsmod | grep bt

```
btusb                  65536  0
btrtl                  28672  1 btusb
btbcm                  20480  1 btusb
btintel                49152  1 btusb
btmtk                  16384  1 btusb
bluetooth             876544  44 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
btrfs                1753088  1
libcrc32c              16384  1 btrfs
xor                    24576  1 btrfs
raid6_pq              122880  1 btrfs
```

sudo modprobe -rv btusb                                                                       
```
rmmod btusb
rmmod btmtk
rmmod btintel
rmmod btbcm
rmmod btrtl
```

sudo modprobe -v btusb                                                                        
```
insmod /lib/modules/5.18.9-1-MANJARO/kernel/drivers/bluetooth/btmtk.ko.xz 
insmod /lib/modules/5.18.9-1-MANJARO/kernel/drivers/bluetooth/btintel.ko.xz 
insmod /lib/modules/5.18.9-1-MANJARO/kernel/drivers/bluetooth/btbcm.ko.xz 
insmod /lib/modules/5.18.9-1-MANJARO/kernel/drivers/bluetooth/btrtl.ko.xz 
insmod /lib/modules/5.18.9-1-MANJARO/kernel/drivers/bluetooth/btusb.ko.xz reset=1 

```

sudo dmesg

```
[  399.925932] usbcore: deregistering interface driver btusb
[  400.048329] audit: type=1130 audit(1657316721.779:142): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  400.186231] audit: type=1106 audit(1657316721.915:143): pid=2983 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  400.186292] audit: type=1104 audit(1657316721.915:144): pid=2983 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  405.054441] audit: type=1131 audit(1657316726.786:145): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  405.708067] audit: type=1101 audit(1657316727.439:146): pid=2997 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="benjamim" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  405.708411] audit: type=1110 audit(1657316727.439:147): pid=2997 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  405.708954] audit: type=1105 audit(1657316727.439:148): pid=2997 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  405.747810] audit: type=1106 audit(1657316727.479:149): pid=2997 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  405.747927] audit: type=1104 audit(1657316727.479:150): pid=2997 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  421.584489] audit: type=1101 audit(1657316743.318:151): pid=3007 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="benjamim" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  421.584908] audit: type=1110 audit(1657316743.318:152): pid=3007 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  421.585428] audit: type=1105 audit(1657316743.318:153): pid=3007 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  421.721160] usbcore: registered new interface driver btusb
[  421.722129] audit: type=1106 audit(1657316743.454:154): pid=3007 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_close grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  421.722240] audit: type=1104 audit(1657316743.454:155): pid=3007 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  421.727510] audit: type=1130 audit(1657316743.461:156): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  424.335707] audit: type=1101 audit(1657316746.071:157): pid=3019 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="benjamim" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[... cut ...] 746.071:158): pid=3019 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
[  424.336460] audit: type=1105 audit(1657316746.071:159): pid=3019 uid=1000 auid=1000 ses=3 subj==unconfined msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
```

dev-comrade, 2022-07-10:
> i have the same problem with my ubuntu 22.04 kernel 5.15.0-40-generic
> 
> Do you have some solution for me?)
> 
> lsmod | grep bt btusb 61440 0 btrtl 24576 1 btusb btbcm 24576 1 btusb btintel 40960 1 btusb bluetooth 688128 43 btrtl,btintel,btbcm,bnep,btusb,rfcomm
> 
> [lsusb_v.txt](https://github.com/lwfinger/rtw89/files/9020924/lsusb_v.txt)
> 
> cat /sys/kernel/debug/usb/devices:
> 
> T: Bus=01 Lev=01 Prnt=01 Port=03 Cnt=02 Dev#= 3 Spd=12 MxCh= 0 D: Ver= 1.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs= 1 P: Vendor=0cb8 ProdID=c549 Rev= 0.00 S: Manufacturer=Realtek S: Product=Bluetooth Radio S: SerialNumber=00e04c000001 C:* #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=500mA I:* If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb E: Ad=81(I) Atr=03(Int.) MxPS= 16 Ivl=1ms E: Ad=02(O) Atr=02(Bulk) MxPS= 64 Ivl=0ms E: Ad=82(I) Atr=02(Bulk) MxPS= 64 Ivl=0ms I:* If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb E: Ad=03(O) Atr=01(Isoc) MxPS= 0 Ivl=1ms E: Ad=83(I) Atr=01(Isoc) MxPS= 0 Ivl=1ms I: If#= 1 Alt= 1 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb E: Ad=03(O) Atr=01(Isoc) MxPS= 9 Ivl=1ms E: Ad=83(I) Atr=01(Isoc) MxPS= 9 Ivl=1ms I: If#= 1 Alt= 2 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb E: Ad=03(O) Atr=01(Isoc) MxPS= 17 Ivl=1ms E: Ad=83(I) Atr=01(Isoc) MxPS= 17 Ivl=1ms I: If#= 1 Alt= 3 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb E: Ad=03(O) Atr=01(Isoc) MxPS= 25 Ivl=1ms E: Ad=83(I) Atr=01(Isoc) MxPS= 25 Ivl=1ms I: If#= 1 Alt= 4 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb E: Ad=03(O) Atr=01(Isoc) MxPS= 33 Ivl=1ms E: Ad=83(I) Atr=01(Isoc) MxPS= 33 Ivl=1ms I: If#= 1 Alt= 5 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb E: Ad=03(O) Atr=01(Isoc) MxPS= 49 Ivl=1ms E: Ad=83(I) Atr=01(Isoc) MxPS= 49 Ivl=1ms

i solve my problem.

just install this driver

https://github.com/lwfinger/rtw89-BT

lwfinger, 2022-07-10:
Repo rtw899-BT has the USB device IDs for a number of new devices,unfortunately it does not fix everything.

0versun, 2022-07-10:
I try rtw899-BT too but its not solve a search BT devices issue

lwfinger, 2022-07-10:
Please go to that repo and open a new issue. In that issue, post the section of 'sudo /sys/kernel/debug/usb/devices' for the Realtek bluetooth device.

0versun, 2022-07-10:
I think my problem is in scope of this issue no necessary to open a new one? Or maybe i not clear understand.

lwfinger, 2022-07-10:
I want your details in the other repo. This one is actually for wifi, not BT.

closed 2022-09-24 by lwfinger