lwfinger/rtw89 #155: Is this driver included in the linux firmware package or the kernel?

Trqhxrd, 2022-09-02:
I've been having issues lately with my bluetooth gamepad. I was wondering if there is another driver installed that also manages my RTL8852AE chip/module/whatever. When I ran `sudo modprobe -r rtw89_8852ae` my bluetooth still worked. I tried uninstalling the package but both bluetooth and wifi kept working. Does uninstalling the package not remove the driver files?

CharString, 2022-09-06:
The names of the modules that are shipped with the kernel all start with 'rtw89_'. I have loaded the drivers from this repo and blacklisted the ones shipped with 5.18.0 kernel shipped from the debian bullseye backports channel.
```console
# lsmod | grep rtw
rtw_8852ae             16384  0
rtw_8852a             679936  1 rtw_8852ae
rtw89pci               49152  1 rtw_8852ae
rtw89core             348160  2 rtw89pci,rtw_8852a
mac80211             1097728  2 rtw89pci,rtw89core
cfg80211             1073152  3 rtw_8852a,rtw89core,mac80211
# cat /etc/modprobe.d/rtw89-kernel-version-blacklist.conf 
blacklist rtw89_8852ae
blacklist rtw89_8852a
blacklist rtw89_pci
blacklist rtw89_core
```

lwfinger, 2022-09-06:
@CharString: Please send me a private E-mail. I have an off-topic topic.

Trqhxrd, 2022-09-11:
Even with the package from the AUR installed I only have the modules starting with rtw89...

```
# lsmod | grep rtw
rtw89_8852ae           16384  0
rtw89_8852a           684032  1 rtw89_8852ae
rtw89_pci              81920  1 rtw89_8852ae
rtw89_core            438272  2 rtw89_pci,rtw89_8852a
mac80211             1249280  2 rtw89_core,rtw89_pci
cfg80211             1089536  3 rtw89_core,mac80211,rtw89_8852a
```

I'm using the `rtw89-dkms-git` from the AUR.
Any idea where the the new modules are? I'd like to test using the non-kernel version because the one shipped with the kernel produces strange errors with my gamepad and headphones.

Trqhxrd, 2022-09-11:
nevermind. I just did not read the readme. However I think the AUR package is broken. The drivers only showed up after I used the installation instructions for all distros.

lwfinger, 2022-09-11:
You need to blacklist the drivers from the kernel.

closed 2022-09-11 by lwfinger