lwfinger/rtw89 #158: Bluetooth not working after waking up from sleep.

Trqhxrd, 2022-09-11:
Sorry for bothering you again.
I'm currently trying to get my bluetooth to work after sleep or hibernation.
However when I resume my bluetooth adapter is not found.
I tried setting the two module options from the readme and also disabled the kernel versions of the modules.
Any idea how to fix this?

```
[   33.616842] Bluetooth: hci0: command 0x0c01 tx timeout
[   33.616880] Bluetooth: hci0: Opcode 0x c01 failed: -110
[   35.453815] audit: type=1131 audit(1662930900.643:139): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   35.750155] Bluetooth: hci0: Opcode 0x c1a failed: -110
[   35.750168] Bluetooth: hci0: command 0x0c1a tx timeout
[   35.750332] PM: suspend exit
[   35.763686] rtw89_8852ae 0000:03:00.0: Firmware version 0.13.36.0, cmd version 0, type 1
[   35.763692] rtw89_8852ae 0000:03:00.0: Firmware version 0.13.36.0, cmd version 0, type 3
[   35.781824] rtw89_8852ae 0000:03:00.0: chip rfe_type is 1
[   35.784805] rtw89_8852ae 0000:03:00.0 wlo1: renamed from wlan0
[   35.811753] audit: type=1130 audit(1662930900.999:140): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-suspend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   35.811762] audit: type=1131 audit(1662930900.999:141): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-suspend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   35.817475] audit: type=1111 audit(1662930901.006:142): pid=466 uid=0 auid=4294967295 ses=4294967295 msg='op=statistics interface="wlan0" ifindex=5 args=500 pid=1220 uid=1000 result=success exe="/usr/bin/NetworkManager" hostname=? addr=? terminal=? res=success'
[   35.846920] Generic FE-GE Realtek PHY r8169-0-200:00: attached PHY driver (mii_bus:phy_addr=r8169-0-200:00, irq=MAC)
[   36.037019] r8169 0000:02:00.0 eno1: Link is Down
[   36.386297] audit: type=1111 audit(1662930901.573:143): pid=466 uid=0 auid=4294967295 ses=4294967295 msg='op=statistics interface="wlo1" ifindex=5 args=500 pid=1220 uid=1000 result=success exe="/usr/bin/NetworkManager" hostname=? addr=? terminal=? res=success'
[   36.583240] rtw89_8852ae 0000:03:00.0: mac c2h class 1 func 2 not support
[   36.583281] rtw89_8852ae 0000:03:00.0: mac c2h class 1 func 2 not support
[   38.711295] wlo1: authenticate with 34:31:c4:76:a1:ab
[   38.711303] wlo1: Invalid HE elem, Disable HE
[   38.932157] wlo1: send auth to 34:31:c4:76:a1:ab (try 1/3)
[   38.933931] wlo1: authenticated
[   38.937558] wlo1: associate with 34:31:c4:76:a1:ab (try 1/3)
[   38.940680] wlo1: RX AssocResp from 34:31:c4:76:a1:ab (capab=0x1511 status=0 aid=4)
[   39.053429] wlo1: associated
[   39.106774] wlo1: Limiting TX power to 27 (30 - 3) dBm as advertised by 34:31:c4:76:a1:ab
[   39.106844] IPv6: ADDRCONF(NETDEV_CHANGE): wlo1: link becomes ready
```

```
#  cat /etc/modprobe.d/rtw89.conf 
blacklist rtw89_8852ae
blacklist rtw89_8852a
blacklist rtw89_pci
blacklist rtw89_core
options rtw89pci disable_aspm_l1=true
options rtw89pci disable_aspm_l1ss=true
```

lwfinger, 2022-09-11:
I provide the BT drivers as a convenience, but I do not do bluetooth.

closed 2022-09-11 by lwfinger

Trqhxrd, 2022-09-13:
Who maintains the BT drivers? I'll post an issue over there.