lwfinger/rtw89 #160: Issue with build on Linux Mint Kernel 5.4.0-126-generic

gudgeong, 2022-09-30:
Just wondering if there is something I am doing wrong or if there are some incompatibilities with the modules and new Kernel versions?

I have attached the output of the make command but alas my programming skills are not good enough to diagnose the issue.

Previously i have been making updates every time the OS is upgraded without issues but maybe there are now dependencies that are not met ?

If I have missed something requirements or new dependencies apologies in advance

[make.log
[image]
](https://github.com/lwfinger/rtw89/files/9679849/make.log)

lwfinger, 2022-10-06:
You need to learn how to use git.

If you do a 'git checkout main' and a 'git pull', do you still have a problem? I thought I fixed the issue.

lwfinger, 2022-10-06:
It is a PITA to backport beyond kernel 5.8. Yes, you with kernels 5.4 will likely need to use the earlier commit as described above.

gudgeong, 2022-10-06:
ok so I have moved my installation to Kernel Version 5.15.  It appears my Wireless adapter has a module built in this kernel version already so appears I don't need to update from this repo.

thanks for the advice on all this

closed 2022-10-06 by gudgeong

lwfinger, 2022-10-06:
The version of the driver in 5.15 is missing some of the improvements in this repo, but if it works for you - then use the in-kernel version.