lwfinger/rtw89 #161: update rtw89-r210.9e91c9f on Oct 4 2022 fails to install with error after running make on Manjaro

salimp2009, 2022-10-04:
Hi
I just installed new update on Manjaro but the wireless disappeared after reboot.
I tried to install using make (this typically happens on most of the updates)
when I try sudo make, it gives bunch of errors; here are the errors. 

`make -C /lib/modules/5.15.65-1-MANJARO/build M=/usr/src/rtw89-r210.9e91c9f modules
make[1]: Entering directory '/usr/lib/modules/5.15.65-1-MANJARO/build'
  CC [M]  /usr/src/rtw89-r210.9e91c9f/mac.o
/usr/src/rtw89-r210.9e91c9f/mac.c:1260:10: error: ‘const struct rtw89_mac_size_set’ has no member named ‘wde_size6’; did you mean ‘wde_size0’?
 1260 |         .wde_size6 = {RTW89_WDE_PG_64, 512, 0,},
      |          ^~~~~~~~~
      |          wde_size0
/usr/src/rtw89-r210.9e91c9f/mac.c:1262:10: error: ‘const struct rtw89_mac_size_set’ has no member named ‘wde_size9’; did you mean ‘wde_size0’?
 1262 |         .wde_size9 = {RTW89_WDE_PG_64, 0, 1024,},
      |          ^~~~~~~~~
      |          wde_size0
/usr/src/rtw89-r210.9e91c9f/mac.c:1272:10: error: ‘const struct rtw89_mac_size_set’ has no member named ‘ple_size6’; did you mean ‘ple_size0’?
 1272 |         .ple_size6 = {RTW89_PLE_PG_128, 496, 16,},
      |          ^~~~~~~~~
      |          ple_size0
/usr/src/rtw89-r210.9e91c9f/mac.c:1274:10: error: ‘const struct rtw89_mac_size_set’ has no member named ‘ple_size8’; did you mean ‘ple_size0’?
 1274 |         .ple_size8 = {RTW89_PLE_PG_128, 64, 960,},
      |          ^~~~~~~~~
      |          ple_size0
/usr/src/rtw89-r210.9e91c9f/mac.c:1284:10: error: ‘const struct rtw89_mac_size_set’ has no member named ‘wde_qt6’; did you mean ‘wde_qt0’?
 1284 |         .wde_qt6 = {448, 48, 0, 16,},
      |          ^~~~~~~
      |          wde_qt0
/usr/src/rtw89-r210.9e91c9f/mac.c:1296:10: error: ‘const struct rtw89_mac_size_set’ has no member named ‘ple_qt18’; did you mean ‘ple_qt13’?
 1296 |         .ple_qt18 = {147, 0, 16, 20, 17, 13, 89, 0, 32, 14, 8, 0,},
      |          ^~~~~~~~
      |          ple_qt13
/usr/src/rtw89-r210.9e91c9f/mac.c:1306:10: error: ‘const struct rtw89_mac_size_set’ has no member named ‘ple_qt58’; did you mean ‘ple_qt5’?
 1306 |         .ple_qt58 = {147, 0, 16, 20, 157, 13, 229, 0, 172, 14, 24, 0,},
      |          ^~~~~~~~
      |          ple_qt5
/usr/src/rtw89-r210.9e91c9f/mac.c:1306:21: error: extra brace group at end of initializer
 1306 |         .ple_qt58 = {147, 0, 16, 20, 157, 13, 229, 0, 172, 14, 24, 0,},
      |                     ^
/usr/src/rtw89-r210.9e91c9f/mac.c:1306:21: note: (near initialization for ‘rtw89_mac_size’)
/usr/src/rtw89-r210.9e91c9f/mac.c:1306:21: warning: excess elements in struct initializer
/usr/src/rtw89-r210.9e91c9f/mac.c:1306:21: note: (near initialization for ‘rtw89_mac_size’)
/usr/src/rtw89-r210.9e91c9f/mac.c: In function ‘rtw89_mac_set_csi_para_reg’:
/usr/src/rtw89-r210.9e91c9f/mac.c:4455:26: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4455 |         u8 *phy_cap = sta->deflink.he_cap.he_cap_elem.phy_cap_info;
      |                          ^~
/usr/src/rtw89-r210.9e91c9f/mac.c:4472:17: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4472 |         if ((sta->deflink.vht_cap.cap & IEEE80211_VHT_CAP_MU_BEAMFORMER_CAPABLE) ||
      |                 ^~
/usr/src/rtw89-r210.9e91c9f/mac.c:4473:17: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4473 |             (sta->deflink.vht_cap.cap & IEEE80211_VHT_CAP_SU_BEAMFORMER_CAPABLE)) {
      |                 ^~
/usr/src/rtw89-r210.9e91c9f/mac.c:4474:34: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4474 |                 ldpc_en &= !!(sta->deflink.vht_cap.cap & IEEE80211_VHT_CAP_RXLDPC);
      |                                  ^~
/usr/src/rtw89-r210.9e91c9f/mac.c:4475:34: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4475 |                 stbc_en &= !!(sta->deflink.vht_cap.cap & IEEE80211_VHT_CAP_RXSTBC_MASK);
      |                                  ^~
In file included from <command-line>:
/usr/src/rtw89-r210.9e91c9f/mac.c:4477:34: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4477 |                               sta->deflink.vht_cap.cap);
      |                                  ^~
././include/linux/compiler_types.h:314:23: note: in definition of macro ‘__compiletime_assert’
  314 |                 if (!(condition))                                       \
      |                       ^~~~~~~~~
././include/linux/compiler_types.h:334:9: note: in expansion of macro ‘_compiletime_assert’
  334 |         _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |         ^~~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:39:37: note: in expansion of macro ‘compiletime_assert’
   39 | #define BUILD_BUG_ON_MSG(cond, msg) compiletime_assert(!(cond), msg)
      |                                     ^~~~~~~~~~~~~~~~~~
./include/linux/bitfield.h:68:17: note: in expansion of macro ‘BUILD_BUG_ON_MSG’
   68 |                 BUILD_BUG_ON_MSG(__bf_cast_unsigned(_mask, _mask) >     \
      |                 ^~~~~~~~~~~~~~~~
./include/linux/bitfield.h:58:43: note: in expansion of macro ‘__unsigned_scalar_typeof’
   58 | #define __bf_cast_unsigned(type, x)     ((__unsigned_scalar_typeof(type))(x))
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/bitfield.h:69:34: note: in expansion of macro ‘__bf_cast_unsigned’
   69 |                                  __bf_cast_unsigned(_reg, ~0ull),       \
      |                                  ^~~~~~~~~~~~~~~~~~
./include/linux/bitfield.h:125:17: note: in expansion of macro ‘__BF_FIELD_CHECK’
  125 |                 __BF_FIELD_CHECK(_mask, _reg, 0U, "FIELD_GET: ");       \
      |                 ^~~~~~~~~~~~~~~~
/usr/src/rtw89-r210.9e91c9f/mac.c:4476:21: note: in expansion of macro ‘FIELD_GET’
 4476 |                 t = FIELD_GET(IEEE80211_VHT_CAP_SOUNDING_DIMENSIONS_MASK,
      |                     ^~~~~~~~~
In file included from /usr/src/rtw89-r210.9e91c9f/core.h:9,
                 from /usr/src/rtw89-r210.9e91c9f/cam.h:8,
                 from /usr/src/rtw89-r210.9e91c9f/mac.c:5:
/usr/src/rtw89-r210.9e91c9f/mac.c:4477:34: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4477 |                               sta->deflink.vht_cap.cap);
      |                                  ^~
./include/linux/bitfield.h:126:35: note: in definition of macro ‘FIELD_GET’
  126 |                 (typeof(_mask))(((_reg) & (_mask)) >> __bf_shf(_mask)); \
      |                                   ^~~~
/usr/src/rtw89-r210.9e91c9f/mac.c: In function ‘rtw89_mac_csi_rrsc’:
/usr/src/rtw89-r210.9e91c9f/mac.c:4518:16: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4518 |         if (sta->deflink.he_cap.has_he) {
      |                ^~
/usr/src/rtw89-r210.9e91c9f/mac.c:4523:16: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4523 |         if (sta->deflink.vht_cap.vht_supported) {
      |                ^~
/usr/src/rtw89-r210.9e91c9f/mac.c:4528:16: error: ‘struct ieee80211_sta’ has no member named ‘deflink’
 4528 |         if (sta->deflink.ht_cap.ht_supported) {
      |                ^~
make[2]: *** [scripts/Makefile.build:289: /usr/src/rtw89-r210.9e91c9f/mac.o] Error 1
make[1]: *** [Makefile:1896: /usr/src/rtw89-r210.9e91c9f] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.15.65-1-MANJARO/build'
make: *** [Makefile:76: all] Error 2`

salimp2009, 2022-10-04:
i added the error msg as text file ;
[rtw89-r219.9e91c9f_InstallationError.txt](https://github.com/lwfinger/rtw89/files/9709634/rtw89-r219.9e91c9f_InstallationError.txt)

rafaelbernard, 2022-10-04:
I have the same kernel and I experience the same issue: the wireless is not working with the latest commit.

I do have it working with https://github.com/lwfinger/rtw89/commit/e3ef10e3a6b3a002af800d658efa904e0d30f1a2. Any commit after that does not work with 5.15.65-1, as far as I can see.

Do you manage it working?

salimp2009, 2022-10-04:
> I have the same kernel and I experience the same issue: the wireless is not working with the latest commit.
> 
> I do have it working with [e3ef10e](https://github.com/lwfinger/rtw89/commit/e3ef10e3a6b3a002af800d658efa904e0d30f1a2). Any commit after that does not work with 5.15.65-1, as far as I can see.
> 
> Do you manage it working?
no i could not. i tried the go back to previous working 0e69f6f but directory is empty, I used the Manjaro package manager .
I also looked at the repo how i can go back to that version. Shall i just clone the repo and how will i switch that change because it is not tagged or it is not a branch. is there a way to switch to that version

salimp2009, 2022-10-04:
i closed it by mistake sorry when i was attaching the file so it is still open :)

salimp2009, 2022-10-04:
> 

do I have to use ; git reset --hard 0e69f6f
will this work ?

rafaelbernard, 2022-10-04:
If you just want to try with the same version as I have it working on my machine:

```shell
git checkout e3ef10e3a6b3a002af800d658efa904e0d30f1a2
make
sudo make install
```

Then reboot and check if you have it working.

gudgeong, 2022-10-05:
> I have the same kernel and I experience the same issue: the wireless is not working with the latest commit.
>
> I do have it working with [...]

Out of interest could you explain what was done here just so I can understand the changes that were required....

Cheers

salimp2009, 2022-10-06:
i just got the update to Manjaro which changed  the kernel to 5.15.71 and also got the new update for rtw89;  a7e89c2  and it works. thank you :)

closed 2022-10-06 by salimp2009

rafaelbernard, 2022-10-08:
> Out of interest could you explain what was done here just so I can understand the changes that were required....
>
> Cheers

In my case, I just got a previous commit from the history (one that I knew it was working) and compiled the drivers with it. Does this answer your question?

salimp2009, 2022-10-08:
> In my case, I just got a previous commit from the history (one that I knew it was working) and compiled the drivers with it. Does this answer your question?

the latest update now should be working with kernel  5.15.65 I believe. 
I updated Manjaro and it switched to kernel 5.15.71 and it works with that kernel