lwfinger/rtw89 #162: Driver no longer works after update from kernel 5.15.65 to 5.15.71 on manjaro

KatanaFluorescent, 2022-10-06:
Hi, my wifi card is a Realtek RTL8852AE (computer is a Lenovo Legion 15AC6H). My computer is running manjaro and I recently made a system upgrade so kernel got updated.
Everything worked fine with kernel 5.15.65 but now, even though I ran, as suggested in the README : 
`cd rtw89
git pull
make clean
make
sudo make install`

which ran without any error, the wifi doesn't work under kernel 5.15.71.

Oddly enough, when I inspect `/lib/modules/5.15.65-1-MANJARO/kernel/drivers/net/wireless/realtek/rtw89` I get : 
`$ sudo ls /lib/modules/5.15.65-1-MANJARO/kernel/drivers/net/wireless/realtek/rtw89
rtw_8852ae.ko  rtw_8852a.ko  rtw89core.ko  rtw89pci.ko
`

So driver 8852ae is here (the one I presume I need ?)

But when I inspect `/lib/modules/5.15.71-1-MANJARO/kernel/drivers/net/wireless/realtek/rtw89` I get : 
`$ sudo ls /lib/modules/5.15.71-1-MANJARO/kernel/drivers/net/wireless/realtek/rtw89
[sudo] password for turbobato: 
rtw_8852a.ko   rtw_8852b.ko   rtw_8852c.ko  rtw89pci.ko
rtw_8852be.ko  rtw_8852ce.ko  rtw89core.ko`

So everything seems to be here, except for the 8852ae which is the one I presume I need. 

If anyone has an idea, I would be glad to hear from them. Thanks !

marshal-shi, 2022-10-06:
I did the follow updates at line 54 of `Makefile` and it worked:

```
# obj-m += rtw_8852be.o
# rtw_8852ae-y := rtw8852be.o
obj-m += rtw_8852ae.o
rtw_8852ae-y := rtw8852ae.o
```

KatanaFluorescent, 2022-10-06:
> I did the follow updates at line 54 of `Makefile` and it worked:
> 
> ```
> # obj-m += rtw_8852be.o
> # rtw_8852ae-y := rtw8852be.o
> obj-m += rtw_8852ae.o
> rtw_8852ae-y := rtw8852ae.o
> ```

Doing this worked, thanks alot !

closed 2022-10-06 by KatanaFluorescent

lwfinger, 2022-10-06:
Yes, that type is now fixed.