lwfinger/rtw89 #163: Module fail to initialize

nicball, 2022-10-10:
I have install the modules and firmwares accordingly on NixOS:
<https://github.com/nicball/nicpkgs/blob/e1cc11d9f7ba893ac9d3f3950179606100535867/flake.nix#L202>

Here is the error dmesg:

> [    2.946818] rtw89_8852be 0000:03:00.0: failed to early request firmware: -22
[    2.947344] rtw89_8852be 0000:03:00.0: failed to async firmware request
[    2.947812] rtw89_8852be 0000:03:00.0: no firmware loaded
[    2.947812] rtw89_8852be 0000:03:00.0: failed to initialise core
[    2.948262] rtw89_8852be: probe of 0000:03:00.0 failed with error -12

Just to mention, the bluetooth function is loading and working perfectly.

lwfinger, 2022-10-10:
The RTW8852BE driver in this repo is incomplete. It was included because I did a 'git push' when I meant 'quilt push'. Later today, I will push the "final" version that has been running on my system for about 18 hours with no drops or any diagnostic messages in the log. The requisite firmware will be part of this push to the repo. Check for new material with 'git pull' in about 4 hours. Once you get an update, it should work.

nicball, 2022-10-10:
Thanks for the quick reply! However it's midnight at my timezone. I will definitely try it tomorrow!

nicball, 2022-10-11:
I have installed the latest commit. So far it worked perfectly! Speed is good and ping is steady.