lwfinger/rtw89 #166: Failed to load firmware on Arch Linux 6.0

whao, 2022-10-14:
System Info
```
Linux ThinkBook-14-ARA 6.0.1-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 13 Oct 2022 18:58:49 +0000 x86_64 GNU/Linux
```
Building Info
```
➜  rtw89 git:(main) make 
make -C /lib/modules/6.0.1-arch2-1/build M=/home/whao/rtw89 modules
make[1]: Entering directory '/usr/lib/modules/6.0.1-arch2-1/build'
  CC [M]  /home/whao/rtw89/core.o
  CC [M]  /home/whao/rtw89/chan.o
  CC [M]  /home/whao/rtw89/mac80211.o
  CC [M]  /home/whao/rtw89/mac.o
  CC [M]  /home/whao/rtw89/phy.o
  CC [M]  /home/whao/rtw89/fw.o
  CC [M]  /home/whao/rtw89/cam.o
  CC [M]  /home/whao/rtw89/efuse.o
  CC [M]  /home/whao/rtw89/regd.o
  CC [M]  /home/whao/rtw89/sar.o
  CC [M]  /home/whao/rtw89/coex.o
  CC [M]  /home/whao/rtw89/ps.o
  CC [M]  /home/whao/rtw89/debug.o
  CC [M]  /home/whao/rtw89/ser.o
  LD [M]  /home/whao/rtw89/rtw89core.o
  CC [M]  /home/whao/rtw89/rtw8852a.o
  CC [M]  /home/whao/rtw89/rtw8852a_table.o
  CC [M]  /home/whao/rtw89/rtw8852a_rfk.o
  CC [M]  /home/whao/rtw89/rtw8852a_rfk_table.o
  LD [M]  /home/whao/rtw89/rtw_8852a.o
  CC [M]  /home/whao/rtw89/rtw8852ae.o
  LD [M]  /home/whao/rtw89/rtw_8852ae.o
  CC [M]  /home/whao/rtw89/rtw8852b.o
  CC [M]  /home/whao/rtw89/rtw8852b_table.o
  CC [M]  /home/whao/rtw89/rtw8852b_rfk.o
  CC [M]  /home/whao/rtw89/rtw8852b_rfk_table.o
  LD [M]  /home/whao/rtw89/rtw_8852b.o
  CC [M]  /home/whao/rtw89/rtw8852be.o
  LD [M]  /home/whao/rtw89/rtw_8852be.o
  CC [M]  /home/whao/rtw89/rtw8852c.o
  CC [M]  /home/whao/rtw89/rtw8852c_table.o
  CC [M]  /home/whao/rtw89/rtw8852c_rfk.o
  CC [M]  /home/whao/rtw89/rtw8852c_rfk_table.o
  LD [M]  /home/whao/rtw89/rtw_8852c.o
  CC [M]  /home/whao/rtw89/rtw8852ce.o
  LD [M]  /home/whao/rtw89/rtw_8852ce.o
  CC [M]  /home/whao/rtw89/pci.o
  LD [M]  /home/whao/rtw89/rtw89pci.o
  MODPOST /home/whao/rtw89/Module.symvers
  CC [M]  /home/whao/rtw89/rtw89core.mod.o
  LD [M]  /home/whao/rtw89/rtw89core.ko
  BTF [M] /home/whao/rtw89/rtw89core.ko
  CC [M]  /home/whao/rtw89/rtw89pci.mod.o
  LD [M]  /home/whao/rtw89/rtw89pci.ko
  BTF [M] /home/whao/rtw89/rtw89pci.ko
  CC [M]  /home/whao/rtw89/rtw_8852a.mod.o
  LD [M]  /home/whao/rtw89/rtw_8852a.ko
  BTF [M] /home/whao/rtw89/rtw_8852a.ko
  CC [M]  /home/whao/rtw89/rtw_8852ae.mod.o
  LD [M]  /home/whao/rtw89/rtw_8852ae.ko
  BTF [M] /home/whao/rtw89/rtw_8852ae.ko
  CC [M]  /home/whao/rtw89/rtw_8852b.mod.o
  LD [M]  /home/whao/rtw89/rtw_8852b.ko
  BTF [M] /home/whao/rtw89/rtw_8852b.ko
  CC [M]  /home/whao/rtw89/rtw_8852be.mod.o
  LD [M]  /home/whao/rtw89/rtw_8852be.ko
  BTF [M] /home/whao/rtw89/rtw_8852be.ko
  CC [M]  /home/whao/rtw89/rtw_8852c.mod.o
  LD [M]  /home/whao/rtw89/rtw_8852c.ko
  BTF [M] /home/whao/rtw89/rtw_8852c.ko
  CC [M]  /home/whao/rtw89/rtw_8852ce.mod.o
  LD [M]  /home/whao/rtw89/rtw_8852ce.ko
  BTF [M] /home/whao/rtw89/rtw_8852ce.ko
make[1]: Leaving directory '/usr/lib/modules/6.0.1-arch2-1/build'
➜  rtw89 git:(main) sudo make install
make -C /lib/modules/6.0.1-arch2-1/build M=/home/whao/rtw89 modules
make[1]: Entering directory '/usr/lib/modules/6.0.1-arch2-1/build'
make[1]: Leaving directory '/usr/lib/modules/6.0.1-arch2-1/build'
Install rtw89 SUCCESS
```
Loading Firmware
```
➜  rtw89 git:(main) sudo modprobe -v rtw_8852be 
insmod /lib/modules/6.0.1-arch2-1/kernel/net/mac80211/mac80211.ko.zst 
insmod /lib/modules/6.0.1-arch2-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko 
insmod /lib/modules/6.0.1-arch2-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko 
insmod /lib/modules/6.0.1-arch2-1/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/6.0.1-arch2-1/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko 
```
Dmesg Errors
```
➜  rtw89 git:(main) sudo dmesg | grep firmware
[  801.528154] rtw89_8852be 0000:03:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[  801.528161] rtw89_8852be 0000:03:00.0: failed to early request firmware: -2
[  801.528261] rtw89_8852be 0000:03:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[  801.528263] rtw89_8852be 0000:03:00.0: failed to request firmware
[  801.529545] rtw89_8852be 0000:03:00.0: failed to wait firmware completion
```
[8852be repo](https://github.com/lwfinger/rtw8852be) works fine tho.

GraxCode, 2022-10-16:
The exact same problem on my side, on Manjaro 6.0.2-2. Same error messages.

dleidert, 2022-10-27:
The firmware file is not installed by the Makefile. For now, you'll have to copy it manually. Then it should work. I think this and #172 are duplicates.

lwfinger, 2022-10-27:
Yes they are duplicates. The fix is pushed.

closed 2022-10-27 by lwfinger

eastoncrafter, 2023-02-13:
I am having this issue, even though the fix was pushed. 

dmesg | grep rtw

[    5.893163] rtw89core: loading out-of-tree module taints kernel.
[    5.893496] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[    5.986329] rtw89_8852be 0000:03:00.0: firmware: failed to load rtw89/rtw8852b_fw.bin (-2)
[    5.986381] rtw89_8852be 0000:03:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    5.986385] rtw89_8852be 0000:03:00.0: failed to early request firmware: -2
[    5.988155] rtw89_8852be 0000:03:00.0: firmware: failed to load rtw89/rtw8852b_fw.bin (-2)
[    5.988206] rtw89_8852be 0000:03:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[    5.988211] rtw89_8852be 0000:03:00.0: failed to request firmware
[    5.989575] rtw89_8852be 0000:03:00.0: failed to wait firmware completion
[    5.989631] rtw89_8852be 0000:03:00.0: failed to setup chip information
[    5.990275] rtw89_8852be: probe of 0000:03:00.0 failed with error -22

Running on debian 11. Any ideas?

eastoncrafter, 2023-02-13:
> Read file README.md!!!!

Oh man. I just re-read the entire thing and noticed the warning at the bottom. Thank you so much, everything works perfectly now!