lwfinger/rtw89 #168: does not work on rtw8852be

saikatm, 2022-10-15:
distro: fedora 36
kernel: 5.19.15.201

Issue: wifi missing 

1. I was using [rtw8852be](https://github.com/lwfinger/rtw8852be) repo
2. updated the kernel to latest
3. rebooted wifi gone
4. compiled this rtw89 repo
5. did make sudo install
6. did cold reboot

wifi is still missing!
I dont know how to provide logs and other stuff. let me know how I can provide more details. thank you so much.

AJRepo, 2022-10-27:
This may be a duplicate of #172 , an issue which has been solved. 

See that ticket for the logs and commands executed for diagnosis.

henk717, 2022-11-30:
Having this on my HP Elitebook 865 G9 with latest firmware.
[  197.111228] rtw89_8852be 0000:01:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[  197.111234] rtw89_8852be 0000:01:00.0: failed to early request firmware: -2
[  197.111346] rtw89_8852be 0000:01:00.0: enabling device (0000 -> 0003)
[  197.111403] rtw89_8852be 0000:01:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
[  197.111407] rtw89_8852be 0000:01:00.0: failed to request firmware
[  197.113388] rtw89_8852be 0000:01:00.0: failed to wait firmware completion
[  197.113393] rtw89_8852be 0000:01:00.0: failed to setup chip information

lwfinger, 2022-11-30:
'sudo apt install *firmware-realtek' should fix you up.  If that does not work, download https://lwfinger.com/download/rtw8852b_fw.bin and copy the bin file (as root) to directory /lib/firmware/rtw89/.

Guite, 2022-12-09:
Copying the bin file fixed this issue for me. Thanks for this hint and your entire efforts @lwfinger

lwfinger, 2022-12-09:
I knew that would do it. You should also post on your distro's site asking them to update their firmware package.

closed 2022-12-09 by lwfinger