lwfinger/rtw89 #169: Kernel oops: RIP: 0010:rtw89_fw_download+0xac/0x430 [rtw89core]

dleidert, 2022-10-23:
I just updated [my DKMS package](https://salsa.debian.org/dleidert/realtek-rtw89) for Ubuntu/Debian package to ship the latest Git contents. After installing it to a Ubuntu 22.04 system (kernel 5.15), loading the module crashes:

```
rtw89core: loading out-of-tree module taints kernel.
rtw89_8852be 0000:03:00.0: enabling device (0000 -> 0003)
rtw89_8852be 0000:03:00.0: Firmware version 10.10.10.60, cmd version 110, type 1
BUG: unable to handle page fault for address: ffffb8fd40fdc010
#PF: supervisor write access in kernel mode
#PF: error_code(0x0002) - not-present page
PGD 100000067 P4D 100000067 PUD 1001dd067 PMD 1115ce067 PTE 0
Oops: 0002 [#1] SMP NOPTI
CPU: 0 PID: 432 Comm: systemd-udevd Tainted: G           O      5.15.0-50-generic #56-Ubuntu
Hardware name: LENOVO 82UD/LNVNB161216, BIOS JECN09WW(V1.02) 01/17/2022
RIP: 0010:rtw89_fw_download+0xac/0x430 [rtw89core]
Code: 0f 84 83 03 00 00 45 31 c0 48 8d 95 e0 fe ff ff 8b 46 04 89 c1 81 e1 ff ff ff 00 a9 00 00 00 10 74 03   
83 c1 08 25 00 00 00 20 <89> 4a 10 41 83 c0 01 48 83 c6 10 48 c1 e8 1d 48 89 7a 08 48 01 cf
[..]
Modules linked in: i2c_algo_bit(+) fb_sys_fops rtw_8852be(O+) snd_rn_pci_acp3x(+) syscopyarea rtw_8852b(O) s
ysfillrect ucsi_acpi(+) snd_pci_acp3x soundcore sysimgblt ccp typec_ucsi typec ideapad_laptop(+) sparse_keymap platform_profile mac_hid soc_button_array
amd_pmc(+) sch_fq_codel rtw89pci(O) rtw89core(O) mac80211 cfg80211 libarc4 ipmi_devintf ipmi_msghandler msr parport_pc ppdev lp parport drm ramoops ree
d_solomon pstore_blk pstore_zone efi_pstore ip_tables x_tables autofs4 hid_sensor_custom wacom hid_sensor_hub usbhid hid_generic ahci crc32_pclmul libah
ci sdhci_pci i2c_hid_acpi cqhci nvme i2c_hid xhci_pci i2c_piix4 sdhci nvme_core xhci_pci_renesas wmi video hid
CR2: ffffb8fd40fdc010
---[ end trace 6007c6c24d59a360 ]---
RIP: 0010:rtw89_fw_download+0xac/0x430 [rtw89core]
```

I did some research and found <https://patchwork.kernel.org/project/linux-wireless/patch/20211105071725.31539-1-tiwai@suse.de/>. This patch has not been applied in your git repository. However, after applying it, the crash disappears, but the firmware fails to load with:

```
rtw89_8852be 0000:03:00.0: enabling device (0000 -> 0003)
rtw89_8852be 0000:03:00.0: Firmware version 10.10.10.60, cmd version 110, type 1
rtw89_8852be 0000:03:00.0: fw type 1 isn't recognized
rtw89_8852be 0000:03:00.0: failed to setup chip information
rtw89_8852be: probe of 0000:03:00.0 failed with error -2
```

which is an error message thrown by `rtw89_fw_download()`, the very function that caused the crash. So, I guess that the patch to fix the crash might need some adjustments. Would you be so kind to look into the crash?

Hardware is a Lenovo Yoga 6 with Ubuntu 22.04 LTS.

lwfinger, 2022-10-23:
As far as I know, that patch has never been applied to the wireless-next repo - at least I could not find it, thus it would not be in my repo. At least, we know why.

You should be able to avoid compressed firmware if you use my repo to install the driver and firmware, which will be non-compressed in that case and should load correctly. Alternatively, you could 'sudo unxz /usr/lib/firmware/rtw89/*.xz'.

dleidert, 2022-10-23:
@lwfinger It is your firmware file. I downloaded it and put it into `/lib/firmware/rtw89/`. There is no compression involved.

lwfinger, 2022-10-23:
If there is no .xz compressed firmware files in /usr/lib/firmware/rtw89/, then the referenced patch could never fix the problem.

I have been running rtw_8852be from this repo for several weeks without a problem.

dleidert, 2022-10-24:
Which operating system? In Ubuntu 22.04 the `linux-firmware` firmware package [ships *some* of the firmware blobs](https://packages.ubuntu.com/jammy-updates/all/linux-firmware/filelist) already (uncompressed though). I wonder if all the firmware blobs has to be replaced by the files from your repository.

dleidert, 2022-10-27:
@lwfinger I think that was caused by Ubuntu 20.04 shipping its own `rtw8852a_fw.bin` file, which I did not replace. After copying all firmware files, the driver finally works. In [my Debian -dkms package](https://salsa.debian.org/dleidert/realtek-rtw89), I ship the binary files now and [create diversions](https://salsa.debian.org/dleidert/realtek-rtw89/-/blob/master/debian/preinst) to not conflict with files shipped by the distribution's firmware package. Are you interested in updating your `debian` directory?

lwfinger, 2022-10-27:
Not really.

closed 2022-10-27 by dleidert