lwfinger/rtw89 #172: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2

AJRepo, 2022-10-26:
Thanks for all you do. 

I am running Ubuntu 22.04.1 LTS  on an  HP Elitebook with a Realtek RTL8852BE WiFi 6801.11az PCIe chip. 

```
$ lspci | grep 02:00
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852

$ sudo lshw (with non relevant stuff removed)
  *-network UNCLAIMED
       description: Network controller
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:02:00.0
       version: 00
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress cap_list
       configuration: latency=0
       resources: ioport:2000(size=256) memory:fb500000-fb5fffff
```

I've tested this with both 
kernel = 5.15.0-52-generic  (from the install)
and
kernel = 5.18.19-051819-generic from kernel.ubuntu.com

The wireless device does not show up at all. I have tried both "make install" and "make signed-install" but always get the same messages from dmesg: 

> [    4.251504] rtw89core: loading out-of-tree module taints kernel.
> [    4.251756] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
> [    4.264537] rtw89_8852be 0000:02:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
> [    4.264540] rtw89_8852be 0000:02:00.0: failed to early request firmware: -2
> [    4.264597] rtw89_8852be 0000:02:00.0: enabling device (0000 -> 0003)
> [    4.264618] rtw89_8852be 0000:02:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
> [    4.264619] rtw89_8852be 0000:02:00.0: failed to request firmware
> [    4.267073] rtw89_8852be 0000:02:00.0: failed to wait firmware completion
> [    4.267077] rtw89_8852be 0000:02:00.0: failed to setup chip information
> [    4.267585] rtw89_8852be: probe of 0000:02:00.0 failed with error -22

I get the same message even if I remove and then use `modprobe -v rtw_8852be`

sudo lsmod | grep rtw gives 
```
rtw_8852be             16384  0
rtw_8852b             389120  1 rtw_8852be
rtw89pci               61440  1 rtw_8852be
rtw89core             430080  2 rtw89pci,rtw_8852b
mac80211             1249280  2 rtw89pci,rtw89core
cfg80211              974848  3 rtw89core,mac80211,rtw_8852b

```
Running ` ip add list ` only shows the loopback and hardwired interfaces. 

I see you said to only use Issues here if it was a module compiling or loading issue and so wondering if the "firmware load ... failed" is suitable for opening here? 

Thanks again!

lwfinger, 2022-10-26:
Your system has compressed firmware, which the driver cannot yet handle. The files in /usr/lib/firmware/rtw89/ probably have the extension "xz". Run the command
sudo unxz /usr/lib/firmware/rtw89/*.xz
After that, the driver should work.

dleidert, 2022-10-27:
@lwfinger The Makefile does not install the `rtw8852b_fw.bin` and `rtw8852c_fw.bin` firmware-files: <https://github.com/lwfinger/rtw89/blob/main/Makefile#L99-L100>. That's why the module cannot find them.

lwfinger, 2022-10-27:
I pushed a fix for this.

closed 2022-10-27 by lwfinger

Demesaikiran, 2022-10-30:
Does not work on the rtl8852be wifi 6 802.11ax PCIe Adapter

lwfinger, 2022-10-30:
I have to see more diagnostics. Please provide the appropriate section of the dmesg output. Note that if you provide the entire dmesg output, I will ignore it.

PureTryOut, 2022-10-31:
This chip works fine with this driver on a laptop of a colleague of mine. Any chance you could upstream this driver to mainline Linux so people don't have to keep searching around the internet for why their wifi doesn't work and end up in the deep dark of compiling drivers...? :sweat_smile:

lwfinger, 2022-10-31:
The driver is in the process of being submitted to the mainline kernel by Realtek, and will likely be in kernel 6.2!

As a volunteer, I make this repo available even before kernel 6.1 is released. It is also conditionalized so that it will build on kernels as old as v5.5.

If you consider a single internel search too taxing, and find compiling a driver too complicated, then there is nothing I can do, other than suggest that you not purchase leading-edge hardware, particularly from Lenovo or HP. Dell, for example, is much more likely to use older wifi hardware.

PureTryOut, 2022-10-31:
Oh 6.2, that I didn't know, awesome! Like I said it's not my hardware, I would've chosen differently as I tend to check any hardware I buy for Linux compatibility 😉

Great that it's already fixed though, we'll keep compiling this driver till 6.2 is released!