lwfinger/rtw89 #173: Getting error when running make install

edmundsj, 2022-10-27:
I am getting the following error when running make install:
```
jordan@jordan-IdeaPad-1-15ALC7:~/rtw89$ sudo make install
make -C /lib/modules/5.15.0-52-generic/build M=/home/jordan/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-52-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  CC [M]  /home/jordan/rtw89/fw.o
/home/jordan/rtw89/fw.c: In function ‘rtw89_fw_h2c_add_wow_fw_ofld’:
/home/jordan/rtw89/fw.c:902:23: error: too many arguments to function ‘ieee80211_nullfunc_get’
  902 |                 skb = ieee80211_nullfunc_get(rtwdev->hw, vif, -1, false);
      |                       ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/jordan/rtw89/core.h:13,
                 from /home/jordan/rtw89/cam.h:8,
                 from /home/jordan/rtw89/fw.c:5:
./include/net/mac80211.h:5101:17: note: declared here
 5101 | struct sk_buff *ieee80211_nullfunc_get(struct ieee80211_hw *hw,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/home/jordan/rtw89/fw.c:905:23: error: too many arguments to function ‘ieee80211_nullfunc_get’
  905 |                 skb = ieee80211_nullfunc_get(rtwdev->hw, vif, -1, true);
      |                       ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/jordan/rtw89/core.h:13,
                 from /home/jordan/rtw89/cam.h:8,
                 from /home/jordan/rtw89/fw.c:5:
./include/net/mac80211.h:5101:17: note: declared here
 5101 | struct sk_buff *ieee80211_nullfunc_get(struct ieee80211_hw *hw,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:297: /home/jordan/rtw89/fw.o] Error 1
make[1]: *** [Makefile:1900: /home/jordan/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-52-generic'
make: *** [Makefile:86: all] Error 2
```

rafaelbernard, 2022-10-27:
Hello. I got the same after `make`:

```
➜   make
make -C /lib/modules/5.15.74-3-MANJARO/build M=/home/rafael/dev/community/lwfinger/rtw89 modules
make[1]: Entering directory '/usr/lib/modules/5.15.74-3-MANJARO/build'
  CC [M]  /home/rafael/dev/community/lwfinger/rtw89/core.o
  CC [M]  /home/rafael/dev/community/lwfinger/rtw89/chan.o
  CC [M]  /home/rafael/dev/community/lwfinger/rtw89/mac80211.o
  CC [M]  /home/rafael/dev/community/lwfinger/rtw89/mac.o
  CC [M]  /home/rafael/dev/community/lwfinger/rtw89/phy.o
  CC [M]  /home/rafael/dev/community/lwfinger/rtw89/fw.o
/home/rafael/dev/community/lwfinger/rtw89/fw.c: In function ‘rtw89_fw_h2c_add_wow_fw_ofld’:
/home/rafael/dev/community/lwfinger/rtw89/fw.c:902:23: error: too many arguments to function ‘ieee80211_nullfunc_get’
  902 |                 skb = ieee80211_nullfunc_get(rtwdev->hw, vif, -1, false);
      |                       ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/rafael/dev/community/lwfinger/rtw89/core.h:13,
                 from /home/rafael/dev/community/lwfinger/rtw89/cam.h:8,
                 from /home/rafael/dev/community/lwfinger/rtw89/fw.c:5:
./include/net/mac80211.h:5090:17: note: declared here
 5090 | struct sk_buff *ieee80211_nullfunc_get(struct ieee80211_hw *hw,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/home/rafael/dev/community/lwfinger/rtw89/fw.c:905:23: error: too many arguments to function ‘ieee80211_nullfunc_get’
  905 |                 skb = ieee80211_nullfunc_get(rtwdev->hw, vif, -1, true);
      |                       ^~~~~~~~~~~~~~~~~~~~~~
./include/net/mac80211.h:5090:17: note: declared here
 5090 | struct sk_buff *ieee80211_nullfunc_get(struct ieee80211_hw *hw,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:289: /home/rafael/dev/community/lwfinger/rtw89/fw.o] Error 1
make[1]: *** [Makefile:1898: /home/rafael/dev/community/lwfinger/rtw89] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.15.74-3-MANJARO/build'
make: *** [Makefile:86: all] Error 2
```

Hash:
```
➜   git rev-parse HEAD
dbbb0a0a8e9fb884824446ca8418c8887cf45e02
```

mauricioquintela, 2022-10-27:
Same error is present when building for kernel 6.0.2 as of latest comming:
```
~/Scripts/rtw89 $ make clean && make -j16 && sudo make install
make -C /lib/modules/6.0.2-arch1-1/build M=/home/mauricioquintela/Scripts/rtw89 modules
make[1]: Entering directory '/usr/lib/modules/6.0.2-arch1-1/build'
  CC [M]  /home/mauricioquintela/Scripts/rtw89/core.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/chan.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/mac80211.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/mac.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/phy.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/fw.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/cam.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/efuse.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/regd.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/sar.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/coex.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/ps.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/debug.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/ser.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/wow.o
  CC [M]  /home/mauricioquintela/Scripts/rtw89/rtw8852a.o
/home/mauricioquintela/Scripts/rtw89/fw.c: In function ‘rtw89_fw_h2c_add_wow_fw_ofld’:
/home/mauricioquintela/Scripts/rtw89/fw.c:902:23: error: too many arguments to function ‘ieee80211_nullfunc_get’
  902 |                 skb = ieee80211_nullfunc_get(rtwdev->hw, vif, -1, false);
      |                       ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/mauricioquintela/Scripts/rtw89/core.h:13,
                 from /home/mauricioquintela/Scripts/rtw89/cam.h:8,
                 from /home/mauricioquintela/Scripts/rtw89/fw.c:5:
./include/net/mac80211.h:5305:17: note: declared here
 5305 | struct sk_buff *ieee80211_nullfunc_get(struct ieee80211_hw *hw,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
/home/mauricioquintela/Scripts/rtw89/fw.c:905:23: error: too many arguments to function ‘ieee80211_nullfunc_get’
  905 |                 skb = ieee80211_nullfunc_get(rtwdev->hw, vif, -1, true);
      |                       ^~~~~~~~~~~~~~~~~~~~~~
./include/net/mac80211.h:5305:17: note: declared here
 5305 | struct sk_buff *ieee80211_nullfunc_get(struct ieee80211_hw *hw,
      |                 ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:249: /home/mauricioquintela/Scripts/rtw89/fw.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1852: /home/mauricioquintela/Scripts/rtw89] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.0.2-arch1-1/build'
make: *** [Makefile:86: all] Error 2
make -j16  24,45s user 2,10s system 862% cpu 3,077 total
```

Commit hash:
```
~/Scripts/rtw89 $ git rev-parse HEAD
dbbb0a0a8e9fb884824446ca8418c8887cf45e02
```

lwfinger, 2022-10-27:
Fixed. Do a 'git pull'

closed 2022-10-27 by lwfinger

edmundsj, 2022-10-28:
#bless