lwfinger/rtw89 #174: The uninstall target copies the firmware into /lib/firmware/rtw89 instead of removing it

dleidert, 2022-10-27:
The `uninstall` target in `Makefile` [copies all firmware files](https://github.com/lwfinger/rtw89/blob/main/Makefile#L108) into `/lib/firmware/rtw89` instead of removing them. Please note, that in Ubuntu 20.04 and above and in Debian 11 and above the `firmware` packages ship some firmware files which the `install` target overwrites. So by removing these files, one original files won't get recovered. Instead, the user would have to reinstall the `linux-firmware`/`firmware-realtek` package to restore the deleted files.

PS: Was it intentional that you dropped the installation of `rtl8852au*.bin` in the recent fix for Makefile?

lwfinger, 2022-10-27:
I take care that my firmware exactly matches the latest from Realtek. Can you assure me that Debian does the same?  If the files differ, the original are wrong, and should not be recovered/
Yes. BT firmware should not be handled in this repo.

dleidert, 2022-10-27:
I'm sorry, but you sound hostile.

> I take care that my firmware exactly matches the latest from Realtek. Can you assure me that Debian does the same?

Neither Debian nor Ubuntu nor any other distribution does what you do. The firmware files in the `linux-firmware`and `firmware-realtek` (or however the firmware package is called in each distribution) are the ones shipped with the version of linux-firmware packaged by the distribution. The firmware files in Ubuntu Jammy, for example, seem to be newer than the ones you have here. You, on the other hand, always provide a version that is probably near to the latest version in Git and that is fitting to your needs.

That doesn't change the fact that you overwrite files which belong to other packages, and that these files are missing if you remove the files you copied. It also means, that the modules shipped by the distribution might fail to load these firmware files after you have deleted your modules via the uninstall action and the modules shipped with the distribution's kernel are loaded. Also, whenever the packages mentioned above get updated, they will overwrite your firmware files (at least those shipped by them). Or consider your decision to no longer install the bluetooth related firmware blob anymore: This file now sits on a lot of systems. You don't care, but this firmware file might actually be incompatible with the user's kernel module loaded.

One cannot simply overwrite and/or remove files also provided by system packages. That is not a problem for Debian or Ubuntu alone. That is the case for all distributions. On Debian/Ubuntu we can use diversions to handle this case. I don't know what the correct way would be for other distributions. Maybe the best course of action would be to rename the firmware files you install if possible.

> If the files differ, the original are wrong,

They are not "wrong".

> and should not be recovered

Ideally, you should leave the system as you found it after an uninstall action.

I'm a Debian developer. I actually packaged drawxtl, which I believe is your software as well. I can handle myself, and I can write a DKMS package that does not bad things and leaves the system without a trace after being uninstalled. But most of your users do not have that experience.

lwfinger, 2022-10-27:
Yes, the 8852be firmware was one version old. It has now been updated.

Yes, DRAWxtl is my software.

Write changes to the Makefile and I will consider them. I have another project that is occupying most of my time at the moment.

closed 2022-12-14 by lwfinger