lwfinger/rtw89 #179: error during make for 5.4.0

yankayy, 2022-10-30:
hello, i'm very new to....everything. i've installed ubuntu 18.04 for a course, followed the steps on the readme but im encountering errors during the make. did i do something wrong? appreciate the help

> make -C /lib/modules/5.4.0-131-generic/build M=/home/yankay/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-131-generic'
  CC [M]  /home/yankay/rtw89/core.o
In file included from /home/yankay/rtw89/cam.h:8:0,
                 from /home/yankay/rtw89/core.c:7:
/home/yankay/rtw89/core.h:4431:18: warning: ‘struct cfg80211_tid_config’ declared inside parameter list will not be visible outside of this definition or declaration
           struct cfg80211_tid_config *tid_config);
                  ^~~~~~~~~~~~~~~~~~~
/home/yankay/rtw89/core.c:2607:19: warning: ‘struct cfg80211_tid_cfg’ declared inside parameter list will not be visible outside of this definition or declaration
            struct cfg80211_tid_cfg *tid_conf)
                   ^~~~~~~~~~~~~~~~
/home/yankay/rtw89/core.c: In function ‘_rtw89_core_set_tid_config’:
/home/yankay/rtw89/core.c:2611:21: error: dereferencing pointer to incomplete type ‘struct cfg80211_tid_cfg’
  u32 mask = tid_conf->mask;
                     ^~
In file included from ./include/linux/bitops.h:5:0,
                 from ./include/linux/kernel.h:12,
                 from ./include/linux/skbuff.h:13,
                 from ./include/linux/ip.h:16,
                 from /home/yankay/rtw89/core.c:4:
/home/yankay/rtw89/core.c:2626:18: error: ‘NL80211_TID_CONFIG_ATTR_AMPDU_CTRL’ undeclared (first use in this function); did you mean ‘NL80211_MESHCONF_ATTR_MAX’?
   if (mask & BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL)) {
                  ^
./include/linux/bits.h:8:30: note: in definition of macro ‘BIT’
 #define BIT(nr)   (UL(1) << (nr))
                              ^~
/home/yankay/rtw89/core.c:2626:18: note: each undeclared identifier is reported only once for each function it appears in
   if (mask & BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL)) {
                  ^
./include/linux/bits.h:8:30: note: in definition of macro ‘BIT’
 #define BIT(nr)   (UL(1) << (nr))
                              ^~
/home/yankay/rtw89/core.c:2627:27: error: ‘NL80211_TID_CONFIG_ENABLE’ undeclared (first use in this function); did you mean ‘NL80211_CMD_CONN_FAILED’?
    if (tid_conf->ampdu == NL80211_TID_CONFIG_ENABLE) {
                           ^~~~~~~~~~~~~~~~~~~~~~~~~
                           NL80211_CMD_CONN_FAILED
/home/yankay/rtw89/core.c: At top level:
/home/yankay/rtw89/core.c:2652:18: warning: ‘struct cfg80211_tid_config’ declared inside parameter list will not be visible outside of this definition or declaration
           struct cfg80211_tid_config *tid_config)
                  ^~~~~~~~~~~~~~~~~~~
/home/yankay/rtw89/core.c:2650:6: error: conflicting types for ‘rtw89_core_set_tid_config’
 void rtw89_core_set_tid_config(struct rtw89_dev *rtwdev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/yankay/rtw89/cam.h:8:0,
                 from /home/yankay/rtw89/core.c:7:
/home/yankay/rtw89/core.h:4429:6: note: previous declaration of ‘rtw89_core_set_tid_config’ was here
 void rtw89_core_set_tid_config(struct rtw89_dev *rtwdev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/yankay/rtw89/core.c: In function ‘rtw89_core_set_tid_config’:
/home/yankay/rtw89/core.c:2656:28: error: dereferencing pointer to incomplete type ‘struct cfg80211_tid_config’
  for (i = 0; i < tid_config->n_tid_conf; i++)
                            ^~
/home/yankay/rtw89/core.c: In function ‘rtw89_core_register_hw’:
/home/yankay/rtw89/core.c:3355:13: error: ‘struct wiphy’ has no member named ‘tid_config_support’; did you mean ‘bss_select_support’?
  hw->wiphy->tid_config_support.vif |= BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL);
             ^~~~~~~~~~~~~~~~~~
             bss_select_support
In file included from ./include/linux/bitops.h:5:0,
                 from ./include/linux/kernel.h:12,
                 from ./include/linux/skbuff.h:13,
                 from ./include/linux/ip.h:16,
                 from /home/yankay/rtw89/core.c:4:
/home/yankay/rtw89/core.c:3355:43: error: ‘NL80211_TID_CONFIG_ATTR_AMPDU_CTRL’ undeclared (first use in this function); did you mean ‘NL80211_MESHCONF_ATTR_MAX’?
  hw->wiphy->tid_config_support.vif |= BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL);
                                           ^
./include/linux/bits.h:8:30: note: in definition of macro ‘BIT’
 #define BIT(nr)   (UL(1) << (nr))
                              ^~
/home/yankay/rtw89/core.c:3356:13: error: ‘struct wiphy’ has no member named ‘tid_config_support’; did you mean ‘bss_select_support’?
  hw->wiphy->tid_config_support.peer |= BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL);
             ^~~~~~~~~~~~~~~~~~
             bss_select_support
scripts/Makefile.build:270: recipe for target '/home/yankay/rtw89/core.o' failed
make[2]: *** [/home/yankay/rtw89/core.o] Error 1
Makefile:1762: recipe for target '/home/yankay/rtw89' failed
make[1]: *** [/home/yankay/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-131-generic'
Makefile:86: recipe for target 'all' failed
make: *** [all] Error 2

lwfinger, 2022-10-30:
I do not support kernel 5.4. There are too many additions of necessary features in later kernels.

You might check the backports project to see if they have code that will run on 5.4.

closed 2022-10-30 by lwfinger

yankayy, 2022-10-30:
Oh I see…alright, thanks!