lwfinger/rtw89 #181: RTL8852BE driver doesn't work if idle for a while

leiless, 2022-11-01:
My system installed the rtw89 RTL8852BE driver, and it works well.
Sometimes, if the system is put idle for 10-30 minutes(I turned off sleep mode completely), the driver won't work anymore.
As a workaround, I have to turn off and then turn on the WiFi so the Internet could work again.

FYI, I'm using Ubuntu 22.04 LTS, 5.15.0-52-generic, x86_64.
And my WiFi uses WPA2, Enterprise with Tunneled TLS via username and password.

leiless, 2022-11-01:
<details>
  <summary>Partial system log</summary>
  
```
[21529.471545] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a00f
[21529.471572] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a0f7
[21529.471583] rtw89_8852be 0000:02:00.0: --->
[21529.471599] rtw89_8852be 0000:02:00.0: R_AX_SER_DBG_INFO =0xf8000000
[21529.471603] rtw89_8852be 0000:02:00.0: R_AX_CMAC_ERR_ISR =0x00000000
[21529.471607] rtw89_8852be 0000:02:00.0: R_AX_DMAC_ERR_ISR =0x00000040
[21529.471611] rtw89_8852be 0000:02:00.0: R_AX_WDE_ERR_FLAG_CFG =0x00000000 
[21529.471615] rtw89_8852be 0000:02:00.0: R_AX_PLE_ERR_FLAG_CFG =0x83190000
[21529.471618] rtw89_8852be 0000:02:00.0: R_AX_WDE_ERR_IMR=0xffffffff 
[21529.471622] rtw89_8852be 0000:02:00.0: R_AX_WDE_ERR_ISR=0x00000000
[21529.471626] rtw89_8852be 0000:02:00.0: R_AX_PLE_ERR_IMR=0xffffffdf 
[21529.471630] rtw89_8852be 0000:02:00.0: R_AX_PLE_ERR_FLAG_ISR=0x02000000
[21529.471633] rtw89_8852be 0000:02:00.0: R_AX_WD_CPUQ_OP_0=0x00000000
[21529.471637] rtw89_8852be 0000:02:00.0: R_AX_WD_CPUQ_OP_1=0x00000000
[21529.471641] rtw89_8852be 0000:02:00.0: R_AX_WD_CPUQ_OP_2=0x00000000
[21529.471644] rtw89_8852be 0000:02:00.0: R_AX_WD_CPUQ_OP_STATUS=0x00000000
[21529.471648] rtw89_8852be 0000:02:00.0: R_AX_PL_CPUQ_OP_0=0x00000000
[21529.471652] rtw89_8852be 0000:02:00.0: R_AX_PL_CPUQ_OP_1=0x00000000
[21529.471656] rtw89_8852be 0000:02:00.0: R_AX_PL_CPUQ_OP_2=0x00000000
[21529.471659] rtw89_8852be 0000:02:00.0: R_AX_PL_CPUQ_OP_STATUS=0x00000000
[21529.471663] rtw89_8852be 0000:02:00.0: R_AX_RXDMA_PKT_INFO_0=0x800081b5
[21529.471667] rtw89_8852be 0000:02:00.0: R_AX_RXDMA_PKT_INFO_1=0x80008000
[21529.471671] rtw89_8852be 0000:02:00.0: R_AX_RXDMA_PKT_INFO_2=0x81e081b6
[21529.471674] rtw89_8852be 0000:02:00.0: R_AX_HOST_DISPATCHER_ALWAYS_IMR=0x0c000161 
[21529.471678] rtw89_8852be 0000:02:00.0: R_AX_HOST_DISPATCHER_ALWAYS_ISR=0x00000000
[21529.471682] rtw89_8852be 0000:02:00.0: R_AX_CPU_DISPATCHER_ALWAYS_IMR=0x04000062 
[21529.471686] rtw89_8852be 0000:02:00.0: R_AX_CPU_DISPATCHER_ALWAYS_ISR=0x00000000
[21529.471689] rtw89_8852be 0000:02:00.0: R_AX_OTHER_DISPATCHER_ALWAYS_IMR=0x00000000 
[21529.471693] rtw89_8852be 0000:02:00.0: R_AX_OTHER_DISPATCHER_ALWAYS_ISR=0x00000000
[21529.471697] rtw89_8852be 0000:02:00.0: R_AX_RPQ_RXBD_IDX =0x00010001
[21529.471700] rtw89_8852be 0000:02:00.0: R_AX_DBG_ERR_FLAG=0x00000000
[21529.471704] rtw89_8852be 0000:02:00.0: R_AX_LBC_WATCHDOG=0x00000081
[21529.471704] rtw89_8852be 0000:02:00.0: <---
[21529.471705] rtw89_8852be 0000:02:00.0: SER catches error: 0x1000
[21530.475579] rtw89_8852be 0000:02:00.0: c2h reg timeout
[21531.476989] rtw89_8852be 0000:02:00.0: c2h reg timeout
[21531.477370] rtw89_8852be 0000:02:00.0: FW status = 0x1001100
[21531.477376] rtw89_8852be 0000:02:00.0: FW BADADDR = 0xff
[21531.477380] rtw89_8852be 0000:02:00.0: FW EPC/RA = 0x0
[21531.477384] rtw89_8852be 0000:02:00.0: FW MISC = 0xb898828b
[21531.477388] rtw89_8852be 0000:02:00.0: R_AX_HALT_C2H = 0x1001
[21531.477392] rtw89_8852be 0000:02:00.0: R_AX_SER_DBG_INFO = 0xf8000000
[21531.477401] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a027
[21531.477445] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a0ef
[21531.477473] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a00b
[21531.477500] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a0ef
[21531.477526] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a009
[21531.477552] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a0d5
[21531.477580] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a0fd
[21531.477607] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a02f
[21531.477634] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a101
[21531.477659] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a02b
[21531.477687] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a0eb
[21531.477714] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a003
[21531.477741] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a0ef
[21531.477767] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a0d3
[21531.477793] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb898a03d
[21531.477804] rtw89_8852be 0000:02:00.0: SER catches error: 0x1001
[21531.478108] rtw89_8852be 0000:02:00.0: FW status = 0x1008100
[21531.478130] rtw89_8852be 0000:02:00.0: FW BADADDR = 0xff
[21531.478135] rtw89_8852be 0000:02:00.0: FW EPC/RA = 0x0
[21531.478139] rtw89_8852be 0000:02:00.0: FW MISC = 0xb898828b
[21531.478143] rtw89_8852be 0000:02:00.0: R_AX_HALT_C2H = 0x1002
[21531.478147] rtw89_8852be 0000:02:00.0: R_AX_SER_DBG_INFO = 0xf8000000
[21531.478155] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb893444b
[21531.478182] rtw89_8852be 0000:02:00.0: [ERR]fw PC = 0xb893471d
[... 114 lines cut ...]
[21532.724014] rtw89_8852be 0000:02:00.0: timed out to flush queues
[21532.931471] rtw89_8852be 0000:02:00.0: timed out to flush queues
[21532.932963] rtw89_8852be 0000:02:00.0: c2h class 1 func 3 not support
[21533.450592] rtw89_8852be 0000:02:00.0: timed out to flush queues
[21533.655224] rtw89_8852be 0000:02:00.0: timed out to flush queues
[21534.177300] rtw89_8852be 0000:02:00.0: timed out to flush queues
[21534.383815] rtw89_8852be 0000:02:00.0: timed out to flush queues

# After I turned off and then turned on the WiFi

[21674.821929] wlp2s0: deauthenticating from 78:8a:20:55:57:0c by local choice (Reason: 3=DEAUTH_LEAVING)
[21675.029716] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 0
[21675.237472] rtw89_8852be 0000:02:00.0: timed out to flush queues
[21675.238161] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 0
[... cut ...]
[21675.443507] rtw89_8852be 0000:02:00.0: timed out to flush pci txch: 0
[21682.322719] wlp2s0: authenticate with 78:8a:20:55:57:0c
[21682.505388] wlp2s0: send auth to 78:8a:20:55:57:0c (try 1/3)
[21682.509405] wlp2s0: authenticated
[21682.513358] wlp2s0: associate with 78:8a:20:55:57:0c (try 1/3)
[21682.520581] wlp2s0: RX AssocResp from 78:8a:20:55:57:0c (capab=0x1511 status=0 aid=50)
[21682.629365] wlp2s0: associated
[21683.209384] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
```
</details>

lwfinger, 2022-11-01:
I think the problem is that your BIOS does not set up the PCIe after resuming. The way to handle this case is to add file suspend-rtw89 as root to directory /usr/lib/systemd/system-sleep/.

The file should contain the following:
#!/bin/sh
if [ "${1}" == "pre" ]; then
  modprobe -rv rtw_8852be
elif [ "${1}" == "post" ]; then
  modprobe -v rtw_8852be
fi

After creating the file, run the command 'sudo chmmod a+x /usr/lib/systemd/system-sleep/suspend-rtw89'.

With this file in place, the driver will be unloaded upon hibernation or sleep, and be reloaded on resume.

If you look in dmesg after the resume, you should see

lwfinger, 2022-11-02:
When it drops, look at the output of the dmesg command and post the rtw8852be output. There are many reasons that could happen. Yes, the driver would still be loaded. In fact, it takes an unload command to get rid of it.

leiless, 2022-11-02:
After I added the script, this bug can still reproduced while I'm working.
dmesg log: https://pastebin.com/FM3eABgh

lwfinger, 2022-11-02:
Your PCIe interface is getting corrupted, which is usually caused by a flaky BIOS.

As root, create a file /usr/lib/modprobe.d/60-rtw_8852be.conf containing the following lines:
options rtw89pci disable_aspm_l1-y
options rtw89pci disable_aspm_l1ss=y
options rtw89pci disable_clkreq=y
options rtw89core disable_ps_mode=y

Not all of those are probably needed, but we will try them all at first.

leiless, 2022-11-02:
After I put the above `/usr/lib/modprobe.d/60-rtw_8852be.conf`, it still won't work.
Previously I'm using Windows 11, and this issue never happened as I remembered.

lwfinger, 2022-11-02:
The Windows driver is totally different, thus the comparison is totally useless other than to say that Windows works around the defect!.

Try the driver at https://github.com/lwfinger/rtw8852be.git. Be aware that that repo will be deleted on Dec. 1. If it works, guard your source as you will be unable to reclone it after that time. I am also not fixing anything. If you gett a build error, you will need to sort it out yourself.

leiless, 2022-11-03:
Is there any option to turn on verbose running/debugging info?

lwfinger, 2022-11-06:
Wrong place to ask this question. Look at the last paragraph of README.md. In that post, show the kind of logiing that makes you ask this question.

leiless, 2022-11-07:
Okay, I guess I just have to close this issue if it's a hardware issue.

closed 2022-11-07 by leiless