lwfinger/rtw89 #182: Driver not working for newer kernels after BIOS update

kovadam69, 2022-11-01:
Hi!

After BIOS Update (HP Elitebook 865 G9) the wifi driver is not working for certain kernels:

I tried: 6.0.6, 6.0.3, 5.19.5, these installed using Ubuntu Mainline kernel installer. The kernel 5.15.5x coming from Ubuntu repo is working though.

I have this error message in the dmesg, and the wifi is continously turning on and off:

`[  338.565179] usb 1-3: new full-speed USB device number 73 using xhci_hcd
[  338.636081] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  338.636084] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  339.071798] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  339.071803] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  339.185678] usb 1-3: device descriptor read/64, error -71
[  339.341715] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  339.341719] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  339.774996] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  339.775001] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  339.917913] usb 1-3: device descriptor read/64, error -71
[  340.025740] usb usb1-port3: attempt power cycle
[  340.046319] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  340.046322] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  340.306460] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  340.306464] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  340.713171] usb 1-3: new full-speed USB device number 74 using xhci_hcd
[  340.713458] usb 1-3: Device not responding to setup address.
[  340.753934] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  340.753937] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  340.921780] usb 1-3: Device not responding to setup address.
[  341.129167] usb 1-3: device not accepting address 74, error -71
[  341.192496] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  341.192501] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  341.324502] [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
[  341.324506] [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
[  342.093517] usb 1-3: new full-speed USB device number 75 using xhci_hcd
[  342.093614] usb 1-3: Device not responding to setup address.
[  342.301685] usb 1-3: Device not responding to setup address.
[  342.509660] usb 1-3: device not accepting address 75, error -71
[  342.509863] usb usb1-port3: unable to enumerate USB device
[  342.781177] usb 1-3: new full-speed USB device number 76 using xhci_hcd
[  343.401718] usb 1-3: device descriptor read/64, error -71
[  344.129391] usb 1-3: device descriptor read/64, error -71
[  344.853623] usb 1-3: new full-speed USB device number 77 using xhci_hcd
`

What's wrong? It was working with 5.19.5 and also with 6.0.3 before the BIOS update. Current BIOS is 01.02.01 Rev A. Here are the release notes what the new BIOS has, but I don't see anything Wifi hardware related: https://support.hp.com/sg-en/drivers/selfservice/hp-elitebook-865-16-inch-g9-notebook-pc/2100825627

Thanks in advance!

lwfinger, 2022-11-01:
It seems pretty clear that HP screwed up the new BIOS, the driver is unable to handle PCIe reads/writes. I have no idea what changed in the kernel, but the latest (6.1-rc3) works fine with my computer.

I hope you have the old firmware so that you can reflash the BIOS. If not, you will need to contact HP and hope that they can provide the older BIOS code.

kovadam69, 2022-11-01:
Yeah, it seems... strange that with 5.15 kernel.it is working properly,
just with 5.19.and 6.0.x seems to be broken...

lwfinger ***@***.***> ezt írta (időpont: 2022. nov. 1., Ke
19:24):

kovadam69, 2022-11-02:
I tried kernel 6.1.0-rc3 and kernel 6.1.0-rc1, but with both RC kernels, my notebook did not even boot up. Or at least it was booting, but then it was freezing when starting the GUI... This seems to be a major issue with the BIOS and/or newer kernels...

kovadam69, 2022-11-04:
I tried now kernel 6.0.7, still not working, I got however some interesting messages during boot and also the same in the dmesg:

`[   65.612981] Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000b lmp_ver=0b lmp_subver=8852
[   65.618979] Bluetooth: hci0: RTL: rom_version status=0 version=1
[   65.618984] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
[   65.619037] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin
[   65.619067] Bluetooth: hci0: RTL: cfg_sz 14, total sz 33515
[   65.631961] usb 1-3: USB disconnect, device number 41
[   65.632146] Bluetooth: hci0: RTL: download fw command failed (-19)
[   65.632162] Bluetooth: hci0: sending frame failed (-19)
[   65.632167] Bluetooth: hci0: Failed to read MSFT supported features (-19)
[   65.632174] Bluetooth: hci0: sending frame failed (-19)
[   65.632178] Bluetooth: hci0: AOSP get vendor capabilities (-19)
`
What can this mean?

lwfinger, 2022-11-04:
Please post the output of 'ls /usr/lib/firmware/rtl_bt/'. Most distros have switched to compressed firmware and btusb/btrtl may not handle the decompression correctly. If the firmware is not compressed, then who knows what is wrong. Is the device at usb 1-3 the Realtek BT device? The lsusb command will show this.

In any case, this problem is not connected with any wifi problems. The Realtek chips use a PCIe interface for the wifi, and a separate USB hub for the BT device. They share the 2.4 GHz radio, but otherwise are completely separate.

kovadam69, 2022-11-04:
This is the ls command output:

-rw-r--r-- 1 root root 38764 okt    3 13:08 rtl8192ee_fw.bin
-rw-r--r-- 1 root root 37904 okt    3 13:08 rtl8192eu_fw.bin
-rw-r--r-- 1 root root 24548 okt    3 13:08 rtl8723a_fw.bin
-rw-r--r-- 1 root root 45048 okt    3 13:08 rtl8723b_fw.bin
lrwxrwxrwx 1 root root    29 okt    3 13:08 rtl8723bs_config-OBDA0623.bin -> rtl8723bs_config-OBDA8723.bin
-rw-r--r-- 1 root root    64 okt    3 13:08 rtl8723bs_config-OBDA8723.bin
-rw-r--r-- 1 root root 52116 okt    3 13:08 rtl8723bs_fw.bin
-rw-r--r-- 1 root root    10 okt    3 13:08 rtl8723d_config.bin
-rw-r--r-- 1 root root 52080 okt    3 13:08 rtl8723d_fw.bin
-rw-r--r-- 1 root root 74488 okt    3 13:08 rtl8761a_fw.bin
-rw-r--r-- 1 root root    25 okt    3 13:08 rtl8761b_config.bin
-rw-r--r-- 1 root root 44204 okt    3 13:08 rtl8761b_fw.bin
-rw-r--r-- 1 root root     6 okt    3 13:08 rtl8761bu_config.bin
-rw-r--r-- 1 root root 42088 okt    3 13:08 rtl8761bu_fw.bin
-rw-r--r-- 1 root root 40520 okt    3 13:08 rtl8812ae_fw.bin
lrwxrwxrwx 1 root root    19 okt    3 13:08 rtl8821a_config.bin -> rtl8821c_config.bin
-rw-r--r-- 1 root root 37420 okt    3 13:08 rtl8821a_fw.bin
-rw-r--r-- 1 root root    10 okt    3 13:08 rtl8821c_config.bin
-rw-r--r-- 1 root root 51700 okt    3 13:08 rtl8821c_fw.bin
-rw-r--r-- 1 root root    14 okt    3 13:08 rtl8822b_config.bin
-rw-r--r-- 1 root root 51176 okt    3 13:08 rtl8822b_fw.bin
-rw-r--r-- 1 root root     6 okt    3 13:08 rtl8822cu_config.bin
-rw-r--r-- 1 root root 58960 okt    3 13:08 rtl8822cu_fw.bin
-rw-r--r-- 1 root root    33 okt    3 13:08 rtl8822cs_config.bin
-rw-r--r-- 1 root root 61796 okt    3 13:08 rtl8822cs_fw.bin
-rw-r--r-- 1 root root     6 nov    4 10:07 rtl8852au_config.bin
-rw-r--r-- 1 root root 49844 nov    4 10:07 rtl8852au_fw.bin
-rw-r--r-- 1 root root    14 nov    4 10:07 rtl8852bu_config.bin
-rw-r--r-- 1 root root 49828 nov    4 10:07 rtl8852bu_fw.bin
-rw-r--r-- 1 root root     6 nov    4 10:07 rtl8852cu_config.bin
-rw-r--r-- 1 root root 96348 nov    4 10:07 rtl8852cu_fw.bin

I also posted to the BT device driver issue page, there are some other useful info there. For some reason on other kernels it fails to init BT device and continously resets the USB. The wifi is turning on/off continously. May be the wifi itself would work, if the kernel would not reset the device all the time. Sometimes I even see the access points for a second, before the wifi is turned off. So it seems something is wrong with the bluetooth.

lsusb output:

Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 30c9:0040 Luxvisions Innotech Limited HP 5MP Camera
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 06cb:00f0 Synaptics, Inc. 
Bus 001 Device 002: ID 0bda:b85c Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

This is on the working kernel. I will post the output for a non-working kernel also.

lspci output:

00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14b5 (rev 01)
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Device 14b6
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14b7 (rev 01)
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14b7 (rev 01)
00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 14ba
00:02.4 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 14ba
00:03.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14b7 (rev 01)
00:03.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 14cd
00:04.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14b7 (rev 01)
00:04.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 14cd
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 14b7 (rev 01)
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 14b9 (rev 10)
00:08.3 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 14b9 (rev 10)
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 71)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 1679
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 167a
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 167b
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 167c
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 167d
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 167e
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 167f
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Device 1680
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
02:00.0 Non-Volatile memory controller: Sandisk Corp WD PC SN810 / Black SN850 NVMe SSD (rev 01)
63:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Rembrandt (rev d4)
63:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Device 1640
63:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] VanGogh PSP/CCP
[... cut ...]
63:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] Raven/Raven2/FireFlight/Renoir Audio Processor (rev 60)
63:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) HD Audio Controller
64:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] Device 161f
64:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Device 15d6
64:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Device 15d7
64:00.5 USB controller: Advanced Micro Devices, Inc. [AMD] Device 162e
64:00.6 USB controller: Advanced Micro Devices, Inc. [AMD] Device 162f

kovadam69, 2022-11-04:
lsusb for kernel 6.0.7:

Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 003: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 008 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 004: ID 2109:8888 VIA Labs, Inc. USB Billboard Device   
Bus 007 Device 007: ID 2109:0102 VIA Labs, Inc. USB 2.0 BILLBOARD             
Bus 007 Device 006: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 007 Device 005: ID 1a2c:4bbc China Resource Semico Co., Ltd USB Keyboard
Bus 007 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 007 Device 002: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 30c9:0040 Luxvisions Innotech Limited HP 5MP Camera
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 06cb:00f0 Synaptics, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The bluetooth device is not even there...

dmesg parts:

loading the wifi driver seems to be OK:
[    0.388529] ACPI: PM: Power Resource [P0NV]
[    2.854998] 8852be: loading out-of-tree module taints kernel.
[    2.904617] 8852be: module verification failed: signature and/or required key missing - tainting kernel
[    3.170976] rtw8852be driver loaded
[    3.170997] rtl8852be 0000:01:00.0: enabling device (0000 -> 0003)
[    4.206950] rtl8852be 0000:01:00.0 wlp1s0: renamed from wlan0

cannot init BT device though:
[    1.340126] usb 1-3: new full-speed USB device number 2 using xhci_hcd
[    1.511262] usb 1-3: New USB device found, idVendor=0bda, idProduct=b85c, bcdDevice= 0.00
[    1.511271] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.511275] usb 1-3: Product: Bluetooth Radio
[    1.511277] usb 1-3: Manufacturer: Realtek
[    1.511279] usb 1-3: SerialNumber: 00e04c000001
[    3.033648] usb 1-3: USB disconnect, device number 2
[    4.020444] usb 1-3: new full-speed USB device number 4 using xhci_hcd
[    4.152502] usb 1-3: Device not responding to setup address.
[    4.360393] usb 1-3: Device not responding to setup address.
[    4.572051] usb 1-3: device not accepting address 4, error -71
[    5.532454] usb 1-3: new full-speed USB device number 5 using xhci_hcd
[    6.156414] usb 1-3: device descriptor read/64, error -71
[    6.880078] usb 1-3: device descriptor read/64, error -71
[    7.468101] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[    7.511311] usb 1-3: New USB device found, idVendor=0bda, idProduct=b85c, bcdDevice= 0.00
[    7.511319] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    7.511321] usb 1-3: Product: Bluetooth Radio
[    7.511323] usb 1-3: Manufacturer: Realtek
[    7.511324] usb 1-3: SerialNumber: 00e04c000001
[    7.682491] usb 1-3: USB disconnect, device number 6
[    8.136059] usb 1-3: new full-speed USB device number 7 using xhci_hcd
[    8.264555] usb 1-3: Device not responding to setup address.
[    8.472158] usb 1-3: Device not responding to setup address.
[    8.680065] usb 1-3: device not accepting address 7, error -71
[    9.640407] usb 1-3: new full-speed USB device number 8 using xhci_hcd
[   10.264710] usb 1-3: device descriptor read/64, error -71
[   10.992312] usb 1-3: device descriptor read/64, error -71
[   13.144557] usb 1-3: new full-speed USB device number 9 using xhci_hcd
[   13.144698] usb 1-3: Device not responding to setup address.
[   13.352560] usb 1-3: Device not responding to setup address.
[   13.560478] usb 1-3: device not accepting address 9, error -71
[   14.524132] usb 1-3: new full-speed USB device number 10 using xhci_hcd
[   14.552719] usb 1-3: device descriptor read/8, error -71
[   14.687258] usb 1-3: device descriptor read/8, error -71
[   15.172482] usb 1-3: new full-speed USB device number 11 using xhci_hcd
[   16.620181] usb 1-3: Device not responding to setup address.
[   16.832104] usb 1-3: Device not responding to setup address.
[   17.040047] usb 1-3: device not accepting address 11, error -71
[   18.628446] usb 1-3: new full-speed USB device number 12 using xhci_hcd
[   19.248953] usb 1-3: device descriptor read/64, error -71
[   19.976496] usb 1-3: device descriptor read/64, error -71

And this repeates infinitely. There is also some dump from the BT driver:

[   91.962891] usb 1-3: new full-speed USB device number 52 using xhci_hcd
[   92.005952] usb 1-3: New USB device found, idVendor=0bda, idProduct=b85c, bcdDevice= 0.00
[   92.005965] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   92.005970] usb 1-3: Product: Bluetooth Radio
[   92.005973] usb 1-3: Manufacturer: Realtek
[   92.005976] usb 1-3: SerialNumber: 00e04c000001
[   92.019552] ************** Entering btusb_probe, driver_info 0x0
[   92.019561] **************** output of usb_match_id 0000000094dcd2b5
[   92.019565] ************** driver_info 0x10000
[   92.019576] ***************** configuring Realtek callbacks
[   92.189771] *************** Entering btrtl_setup_realtek
[   92.191529] Bluetooth: hci0: RTL: HCI_OP_READ_LOCAL_VERSION failed (-71)
[   92.191535] ************ btrtl_initialize failed with error -71

And every time the kernel makes a power cycle on this USB device, the Wifi turns OFF, then ON again, so basically it is useless.

On kernel 5.15.0-52 I use an older BT driver, not the most recent one. I'm going to compile and install the latest, and see if it gets there also corrupted.

kovadam69, 2022-11-04:
lsusb for kernel 6.0.7:

Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 003: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 008 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 004: ID 2109:8888 VIA Labs, Inc. USB Billboard Device   
Bus 007 Device 007: ID 2109:0102 VIA Labs, Inc. USB 2.0 BILLBOARD             
Bus 007 Device 006: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 007 Device 005: ID 1a2c:4bbc China Resource Semico Co., Ltd USB Keyboard
Bus 007 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 007 Device 002: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 30c9:0040 Luxvisions Innotech Limited HP 5MP Camera
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 06cb:00f0 Synaptics, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The bluetooth device is not even there...

dmesg parts:

loading the wifi driver seems to be OK:
[    0.388529] ACPI: PM: Power Resource [P0NV]
[    2.854998] 8852be: loading out-of-tree module taints kernel.
[    2.904617] 8852be: module verification failed: signature and/or required key missing - tainting kernel
[    3.170976] rtw8852be driver loaded
[    3.170997] rtl8852be 0000:01:00.0: enabling device (0000 -> 0003)
[    4.206950] rtl8852be 0000:01:00.0 wlp1s0: renamed from wlan0

cannot init BT device though:
[    1.340126] usb 1-3: new full-speed USB device number 2 using xhci_hcd
[    1.511262] usb 1-3: New USB device found, idVendor=0bda, idProduct=b85c, bcdDevice= 0.00
[    1.511271] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.511275] usb 1-3: Product: Bluetooth Radio
[    1.511277] usb 1-3: Manufacturer: Realtek
[    1.511279] usb 1-3: SerialNumber: 00e04c000001
[    3.033648] usb 1-3: USB disconnect, device number 2
[    4.020444] usb 1-3: new full-speed USB device number 4 using xhci_hcd
[    4.152502] usb 1-3: Device not responding to setup address.
[    4.360393] usb 1-3: Device not responding to setup address.
[    4.572051] usb 1-3: device not accepting address 4, error -71
[    5.532454] usb 1-3: new full-speed USB device number 5 using xhci_hcd
[    6.156414] usb 1-3: device descriptor read/64, error -71
[    6.880078] usb 1-3: device descriptor read/64, error -71
[    7.468101] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[    7.511311] usb 1-3: New USB device found, idVendor=0bda, idProduct=b85c, bcdDevice= 0.00
[    7.511319] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    7.511321] usb 1-3: Product: Bluetooth Radio
[    7.511323] usb 1-3: Manufacturer: Realtek
[    7.511324] usb 1-3: SerialNumber: 00e04c000001
[    7.682491] usb 1-3: USB disconnect, device number 6
[    8.136059] usb 1-3: new full-speed USB device number 7 using xhci_hcd
[    8.264555] usb 1-3: Device not responding to setup address.
[    8.472158] usb 1-3: Device not responding to setup address.
[    8.680065] usb 1-3: device not accepting address 7, error -71
[    9.640407] usb 1-3: new full-speed USB device number 8 using xhci_hcd
[   10.264710] usb 1-3: device descriptor read/64, error -71
[   10.992312] usb 1-3: device descriptor read/64, error -71
[   13.144557] usb 1-3: new full-speed USB device number 9 using xhci_hcd
[   13.144698] usb 1-3: Device not responding to setup address.
[   13.352560] usb 1-3: Device not responding to setup address.
[   13.560478] usb 1-3: device not accepting address 9, error -71
[   14.524132] usb 1-3: new full-speed USB device number 10 using xhci_hcd
[   14.552719] usb 1-3: device descriptor read/8, error -71
[   14.687258] usb 1-3: device descriptor read/8, error -71
[   15.172482] usb 1-3: new full-speed USB device number 11 using xhci_hcd
[   16.620181] usb 1-3: Device not responding to setup address.
[   16.832104] usb 1-3: Device not responding to setup address.
[   17.040047] usb 1-3: device not accepting address 11, error -71
[   18.628446] usb 1-3: new full-speed USB device number 12 using xhci_hcd
[   19.248953] usb 1-3: device descriptor read/64, error -71
[   19.976496] usb 1-3: device descriptor read/64, error -71

And this repeates infinitely. There is also some dump from the BT driver:

[   91.962891] usb 1-3: new full-speed USB device number 52 using xhci_hcd
[   92.005952] usb 1-3: New USB device found, idVendor=0bda, idProduct=b85c, bcdDevice= 0.00
[   92.005965] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   92.005970] usb 1-3: Product: Bluetooth Radio
[   92.005973] usb 1-3: Manufacturer: Realtek
[   92.005976] usb 1-3: SerialNumber: 00e04c000001
[   92.019552] ************** Entering btusb_probe, driver_info 0x0
[   92.019561] **************** output of usb_match_id 0000000094dcd2b5
[   92.019565] ************** driver_info 0x10000
[   92.019576] ***************** configuring Realtek callbacks
[   92.189771] *************** Entering btrtl_setup_realtek
[   92.191529] Bluetooth: hci0: RTL: HCI_OP_READ_LOCAL_VERSION failed (-71)
[   92.191535] ************ btrtl_initialize failed with error -71

And every time the kernel makes a power cycle on this USB device, the Wifi turns OFF, then ON again, so basically it is useless.

On kernel 5.15.0-52 I use an older BT driver, not the most recent one. I'm going to compile and install the latest, and see if it gets there also corrupted.

Checked now on 5.15.0-52 the newest driver code and it still working properly.

lwfinger, 2022-11-04:
Device USB 1-3, which is having all the problems is "Bus 001 Device 003: ID 06cb:00f0 Synaptics, Inc.". That one is not in the tables that I checked, but the Synaptics devices are all touchpads or fingerprint readers.

That laptop is really strange. Who puts their gigabit ethernet adapter on a USB bus? The only other place I have seen that was with some early Raspberry PI models, and similar SBCs.

kovadam69, 2022-11-04:
the nitebook does not have ethernet nic, just wifi, that realtek ethernet
nic is from the thunderbolt docking station. It is.indeed strange that usb
1-3 is listed as the synaptic touchpad, however in dmesg the BT device is
identified as USB 1-3 device

lwfinger ***@***.***> ezt írta (időpont: 2022. nov. 4., Pén
17:04):

lwfinger, 2022-11-04:
The working kernel definitely shows the BT device at USB 1-2 (Bus 001 Device 002: ID 0bda:b85c Realtek Semiconductor Corp. Bluetooth Radio), not USB 1-3. You can keep the BT from loading by blacklisting btusb in /usr/lib/modprobe.d/. That might give you working wifi.

I have no idea why the newer kernels do not show the BT device in the lsusb list, then install its driver on the wrong USB slot.

I take it that you do not have the old BIOS firmware available, and that the vendor is no help. If you are willing to do a bisection between v5.19 and v5.15 on the mainline source, it might be possible to identify what changed in the kernel and get it fixed, probably with a quirk specific to this machine. That will be 14 or 15 kernel builds. Until we know the exact change that leads to the problem, there is nothing to complain to the gurus. Obviously, the new BIOS fails to handle USB correctly.

kovadam69, 2022-11-04:
Both kernels show the BT device on USB 1-2, however with the newer kernel it is not in the list, cause I barely can capture an LSUSB output where the USB device is not in a reset state... the lsusb command itself takes seconds to execute instead of milliseconds like on the working kernel.

But on the working kernel it is listed in lsusb as USB 1-2 but configured according to dmesg to USB 1-3 like on newer kernels.

kovadam69, 2022-11-04:
this is lsusb & dmesg output from working kernel:

Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 003: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 008 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 004: ID 2109:8888 VIA Labs, Inc. USB Billboard Device   
Bus 007 Device 007: ID 2109:0102 VIA Labs, Inc. USB 2.0 BILLBOARD             
Bus 007 Device 006: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 007 Device 005: ID 1a2c:4bbc China Resource Semico Co., Ltd USB Keyboard
Bus 007 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 007 Device 002: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 30c9:0040 Luxvisions Innotech Limited HP 5MP Camera
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 06cb:00f0 Synaptics, Inc. 
Bus 001 Device 002: ID 0bda:b85c Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
~$ sudo dmesg | grep "usb 1-3"
[    1.341531] usb 1-3: new full-speed USB device number 2 using xhci_hcd
[    1.513210] usb 1-3: New USB device found, idVendor=0bda, idProduct=b85c, bcdDevice= 0.00
[    1.513219] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.513223] usb 1-3: Product: Bluetooth Radio
[    1.513226] usb 1-3: Manufacturer: Realtek
[    1.513228] usb 1-3: SerialNumber: 00e04c000001

As you see it's on Bus #1 Dev #2 however in demesg it seems to be enumerated to usb 1-3, the same as in the newer kernel...

kovadam69, 2022-11-04:
I have the prev BIOS, I also tried to flash, however it gave me an error,
that the current BIOS version is locked. I checked the BIOS settings, and
the "lock current BIOS version" checkbox is DEFINITELLY not checked, so it
is not locked, also DOWNGRADE is enabled, but despite this, I did not
manage to reflash the prev BIOS... don't know why. :( Can be, that this
BIOS version itself has a lock for downgrade...

Actually the new BIOS fixed thunderbolt display mirroring, since with
5.15.0-52 I did not manage to use both displays on my thunderbolt dock, now
with the new BIOS I have both displays working, so actually I do not need
newer, that the 5.15. Previously it was working with 5.19 only (or at least
that was the most recent kernel by that time). The machine is quite new at
HP, so I think it's hardver is not fully supported yet by linux, at least
not by newer builds. Kernel 6.1 RC does not even boot... it seems to start,
then hangs when starting up lightdm...

Once I have time, I try to play with 5.15-5.18 and see which version is
starting to behave wrong regarding BT USB device.

lwfinger ***@***.***> ezt írta (időpont: 2022. nov. 4., P,
17:44):

kovadam69, 2022-11-07:
So I was trying with some kernels:

5.19 -> not working
5.18 -> not working
5.17 -> not working
5.16 -> cannot boot (tried 5.16.0 yet)
5.15.77 -> driver&BT working

I managed now to boot up the notebook with 6.1-rc4, BT seems to work (at least there is no errors in the DMESG except this one line: [    3.840277] Bluetooth: hci0: HCI_REQ-0xfcf0
Otherwise seems OK:

[    1.545255] usb 1-3: Product: Bluetooth Radio
[    2.994546] Bluetooth: Core ver 2.22
[    2.994569] Bluetooth: HCI device and connection manager initialized
[    2.994576] Bluetooth: HCI socket layer initialized
[    2.994579] Bluetooth: L2CAP socket layer initialized
[    2.994584] Bluetooth: SCO socket layer initialized
[    3.209218] Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000b lmp_ver=0b lmp_subver=8852
[    3.212212] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    3.212217] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
[    3.213057] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin
[    3.213175] Bluetooth: hci0: RTL: cfg_sz 14, total sz 33515
[    3.383544] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    3.383548] Bluetooth: BNEP filters: protocol multicast
[    3.383552] Bluetooth: BNEP socket layer initialized
[    3.675238] Bluetooth: hci0: RTL: fw version 0xd9b744d8
[    3.840277] Bluetooth: hci0: HCI_REQ-0xfcf0
[    3.847247] Bluetooth: hci0: AOSP extensions version v0.96
[    3.847251] Bluetooth: hci0: AOSP quality report is not supported
[    3.847342] Bluetooth: MGMT ver 1.22
[    6.562920] Bluetooth: RFCOMM TTY layer initialized
[    6.562930] Bluetooth: RFCOMM socket layer initialized
[    6.562934] Bluetooth: RFCOMM ver 1.11

But I cannot compile the driver, it gives me an error:

rtw8852be/os_dep/osdep_service_linux.c:854:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
  854 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1: some warnings being treated as errors

With kernel 5.16 I need to try, but I have to get it working somehow... It has some issues with amdgpu and freezes when starting GUI

kovadam69, 2022-11-07:
Indeed the other driver was compiling against 6.1-rc4. And it's working, both wifi and BT, I'm writing this now on WIFI and I connected a BT mouse, so it seems there is something conflicting with this BIOS version and kernels from 5.16.0 to 6.0.x. 6.1 seems to work, however if I connect the docking station to the notebook, it wont boot up the GUI, it simply freezes. I cannot gather any logs, why. But that's another story, i think we can close this one, since it's not with the driver, but with BIOS & kernel conflict.

closed 2022-11-07 by kovadam69

kovadam69, 2022-11-08:
I fired a bug report on bugzilla.kernel.org, but seems other faced the problem also, some with the WIFI/BT subsystem, like me. Here should be the solution:

https://bugzilla.kernel.org/show_bug.cgi?id=216468

Randall-Coding, 2023-02-12:
I'm not sure if I had this exact issue, but I did see this same line in dmesg when my driver wasn't working

> 8852be: module verification failed: signature and/or required key missing - tainting kernel

I fixed it by using the signed installer and secure boot.  I'm running Ubuntu 20.04 and kernel 5.15.

lwfinger, 2023-02-12:
If you use secure boot, then you MUST sign the module. README.md covers this.

Randall-Coding, 2023-02-12:
> If you use secure boot, then you MUST sign the module. README.md covers this.

In my case I had turned off secure boot already trying other solutions for my driver issue.