lwfinger/rtw89 #184: "make" misfunction on void linux

xiaoshihou514, 2022-11-10:
## Problem
Compile errors happened during recommended installation.
OS: void linux (glibc)
## What did I try:
### Install the linux header via `sudo xbps-install linux5.19-headers-5.19.17_1`
```
[xiaoshihou@void]$ uname -r
5.19.17_1
[xiaoshihou@void]$ xbps-query -s header
[*] kernel-libc-headers-5.10.4_1 Linux API headers for userland development
[*] linux5.19-headers-5.19.17_1  Linux kernel and modules (5.19 series) - source headers for 3rd party modules
```
### Install gcc via xbps
```
[xiaoshihou@void rtw89]$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib64/gcc/x86_64-unknown-linux-gnu/10.2.1/lto-wrapper
Target: x86_64-unknown-linux-gnu
Configured with: /builddir/gcc-10.2.1_pre1/configure --build=x86_64-unknown-linux-gnu --enable-gnu-unique-object --enable-vtable-verify --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --libexecdir=/usr/lib64 --libdir=/usr/lib64 --enable-threads=posix --enable-__cxa_atexit --disable-multilib --with-system-zlib --enable-shared --enable-lto --enable-plugins --enable-linker-build-id --disable-werror --disable-nls --enable-default-pie --enable-default-ssp --enable-checking=release --disable-libstdcxx-pch --with-isl --with-linker-hash-style=gnu --disable-sjlj-exceptions --disable-target-libiberty --enable-languages=c,c++,objc,obj-c++,fortran,lto,go,ada
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.1 20201203 (GCC) 
```

### Get the recommended setup as shown in README (for all distros section)
```bash
git clone https://github.com/lwfinger/rtw89.git -b v7
cd rtw89
make
sudo make install
```

### Make output
```
[xiaoshihou@void rtw89]$ make
make -C /lib/modules/5.19.17_1/build M=/home/xiaoshihou/Documents/repository/wifi-driver/rtw89 modules
make[1]: Entering directory '/usr/src/kernel-headers-5.19.17_1'
  CC [M]  /home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.o
In file included from /home/xiaoshihou/Documents/repository/wifi-driver/rtw89/coex.h:8,
                 from /home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:6:
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.h: In function 'rtw89_sta_has_beamformer_cap':
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.h:3315:10: error: 'struct ieee80211_sta' has no member named 'vht_cap'
 3315 |  if ((sta->vht_cap.cap & IEEE80211_VHT_CAP_MU_BEAMFORMER_CAPABLE) ||
      |          ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.h:3316:10: error: 'struct ieee80211_sta' has no member named 'vht_cap'
 3316 |      (sta->vht_cap.cap & IEEE80211_VHT_CAP_SU_BEAMFORMER_CAPABLE) ||
      |          ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.h:3317:10: error: 'struct ieee80211_sta' has no member named 'he_cap'
 3317 |      (sta->he_cap.he_cap_elem.phy_cap_info[3] & IEEE80211_HE_PHY_CAP3_SU_BEAMFORMER) ||
      |          ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.h:3318:10: error: 'struct ieee80211_sta' has no member named 'he_cap'
 3318 |      (sta->he_cap.he_cap_elem.phy_cap_info[4] & IEEE80211_HE_PHY_CAP4_MU_BEAMFORMER))
      |          ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c: In function 'rtw89_core_tx_update_ampdu_info':
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:312:14: error: 'struct ieee80211_sta' has no member named 'ht_cap'
  312 |      4 << sta->ht_cap.ampdu_factor) - 1);
      |              ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:315:32: error: 'struct ieee80211_sta' has no member named 'ht_cap'
  315 |  desc_info->ampdu_density = sta->ht_cap.ampdu_density;
      |                                ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c: In function '__rtw89_core_tx_check_he_qos_htc':
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:478:18: error: 'struct ieee80211_sta' has no member named 'he_cap'
  478 |  if (!sta || !sta->he_cap.has_he)
      |                  ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c: In function 'rtw89_core_tx_btc_spec_pkt_notify':
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:600:6: error: implicit declaration of function 'ip_hdr' [-Werror=implicit-function-declaration]
  600 |      ip_hdr(skb)->protocol == IPPROTO_UDP) {
      |      ^~~~~~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:600:17: error: invalid type argument of '->' (have 'int')
  600 |      ip_hdr(skb)->protocol == IPPROTO_UDP) {
      |                 ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:601:12: error: implicit declaration of function 'udp_hdr' [-Werror=implicit-function-declaration]
  601 |   udphdr = udp_hdr(skb);
      |            ^~~~~~~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:601:10: warning: assignment to 'struct udphdr *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  601 |   udphdr = udp_hdr(skb);
      |          ^
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:602:15: error: invalid use of undefined type 'struct udphdr'
  602 |   if (((udphdr->source == htons(67) && udphdr->dest == htons(68)) ||
      |               ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:602:46: error: invalid use of undefined type 'struct udphdr'
  602 |   if (((udphdr->source == htons(67) && udphdr->dest == htons(68)) ||
      |                                              ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:603:15: error: invalid use of undefined type 'struct udphdr'
  603 |        (udphdr->source == htons(68) && udphdr->dest == htons(67))) &&
      |               ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:603:46: error: invalid use of undefined type 'struct udphdr'
  603 |        (udphdr->source == htons(68) && udphdr->dest == htons(67))) &&
      |                                              ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:611:17: error: invalid type argument of '->' (have 'int')
  611 |      ip_hdr(skb)->protocol == IPPROTO_ICMP) {
      |                 ^~
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c: In function 'rtw89_init_he_cap':
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:2147:7: error: 'IEEE80211_HE_PHY_CAP9_NOMIMAL_PKT_PADDING_16US' undeclared (first use in this function); did you mean 'IEEE80211_HE_PHY_CAP9_NOMINAL_PKT_PADDING_16US'?
 2147 |       IEEE80211_HE_PHY_CAP9_NOMIMAL_PKT_PADDING_16US;
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |       IEEE80211_HE_PHY_CAP9_NOMINAL_PKT_PADDING_16US
/home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.c:2147:7: note: each undeclared identifier is reported only once for each function it appears in
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /home/xiaoshihou/Documents/repository/wifi-driver/rtw89/core.o] Error 1
make[1]: *** [Makefile:1846: /home/xiaoshihou/Documents/repository/wifi-driver/rtw89] Error 2
make[1]: Leaving directory '/usr/src/kernel-headers-5.19.17_1'
make: *** [Makefile:54: all] Error 2
```
### Additional info
```
[xiaoshihou@void rtw89]$ ls /lib/modules/5.19.17_1/
build   modules.alias      modules.builtin            modules.builtin.bin      modules.dep      modules.devname  modules.softdep  modules.symbols.bin
kernel  modules.alias.bin  modules.builtin.alias.bin  modules.builtin.modinfo  modules.dep.bin  modules.order    modules.symbols
[xiaoshihou@void rtw89]$ sudo find / -name build
/* irrelevant info omitted */
/usr/lib/modules/5.19.17_1/build
```

Thanks a ton for creating the driver, I successfully installed it on manjaro on another disk on the same pc before, but this time I could not really figure out what is happening. Best wishes for the dev(s).

lwfinger, 2022-11-10:
Branch v7 has been long abandoned. Run a 'git checkout origin/main -b main' and all should be well. The errors you are seeing in v7 were fixed a long time ago.

closed 2022-11-10 by lwfinger

xiaoshihou514, 2022-11-10:
> Branch v7 has been long abandoned. Run a 'git checkout origin/main -b main' and all should be well. The errors you are seeing in v7 were fixed a long time ago.

Thanks for your quick answer! It works :D 👍