lwfinger/rtw89 #186: Error while building "debug.c"

ii-richie, 2022-11-10:
`
/home/richie/rtw89/debug.c:60:10: error: ‘RATE_INFO_BW_320’ undeclared here (not in a function); did you mean ‘RATE_INFO_BW_20’?
   60 |         [RATE_INFO_BW_320] = 320,
      |          ^~~~~~~~~~~~~~~~
      |          RATE_INFO_BW_20
/home/richie/rtw89/debug.c:60:10: error: array index in initializer not of integer type
/home/richie/rtw89/debug.c:60:10: note: (near initialization for ‘rtw89_rate_info_bw_to_mhz_map’)
make[2]: *** [scripts/Makefile.build:297: /home/richie/rtw89/debug.o] Error 1
make[1]: *** [Makefile:1900: /home/richie/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-52-generic'
make: *** [Makefile:86: all] Error 2
`

lwfinger, 2022-11-10:
That symbol was not defined until kernel 5.18. Do a git pull and it should build now.

ii-richie, 2022-11-10:
works now! thanks for being so quick!!

closed 2022-11-10 by ii-richie