lwfinger/rtw89 #189: Manjaro Kernel update fails because of c

salimp2009, 2022-11-15:
Hi
I am using Manjaro.
the update to Kernel 5.15.78-1 failed today giving an error due to file 
`usr/lib/firmware/rtw89/rtw8852b_fw.bin `

I have Realtek RTL8852AE 

I deleted the file and update completed. 
This was never happening before
Also after reboot the wireless was not gone and I did not have to do reinstall as usual.
I am on commit a275186 
This was not happening before .  at least 2-3 commit before
Let me know if you need any specific logs

rafaelbernard, 2022-11-15:
The same happened to me. And I took the same approach (I `rm /usr/lib/firmware/rtw89/rtw8852b_fw.bin`). The Wireless was working fine after reboot as well.

rafaelbernard, 2022-11-15:
I will also post if I recover it by a chance.

salimp2009, 2022-11-16:
the only related i could filter from the installation is that somehow it reinstalled the rtw package
I will keep looking

filtered out from pacman .log
using
`cat pacman.log | grep '2022-11-15' | grep 'rtw89' `

```
2022-11-15T14:02:32+0300] [ALPM-SCRIPTLET] ==> dkms remove --no-depmod rtw89/r296.a275186 -k 5.15.76-1-MANJARO
[2022-11-15T14:03:10+0300] [ALPM-SCRIPTLET] ==> dkms install --no-depmod rtw89/r296.a275186 -k 5.15.78-1-MANJARO
```

salimp2009, 2022-11-17:
I posted on Manjaro forum and responses were mainly regarding if a package leaves an unused file or if installation done without using a package manager then there can be files that are either in conflict or orphon.
so the error message I got was similar to this one ; 
```
error: could not prepare transaction
error: failed to commit transaction (conflicting files)
libname: /insert/file/name/here exists in filesystem
```
also there is a link from Arch wiki
[](https://wiki.archlinux.org/title/pacman#%22Failed_to_commit_transaction_(conflicting_files)%22_error)
or a similar one here
[](https://bbs.archlinux.org/viewtopic.php?id=56373)

lwfinger, 2022-11-17:
I removed all the firmware from this repo.

closed 2022-11-17 by lwfinger

salimp2009, 2022-11-17:
so does it mean we dont need to install the package anymore. since it is linux firmware ?

lwfinger, 2022-11-17:
You do not understand the difference between firmware and drivers. Firmware is the stuff that is executed in the on-board CPU of the device. The driver runs in the main CPU. The device will not work without both!!

Firmware is kernel independent, but the driver is not. If you run a 6.0.0 or newer kernel, then you do not need this package, but you do for older kernels; however, this repo has better code than 6.0 as this repo is the equivalent of the code in kernel 6.2.0, which will not be out for about 2 months. For rtw8852be, there are significant improvements.